thesofproject/sof #280: UT don't work after master merge

jajanusz, 2018-08-30:
After merge UT build fails, I'll fix it

jajanusz, 2018-08-30:
@slawblauciak 
Can you fix alloc tests after https://github.com/thesofproject/sof/pull/281 is merged?

mengdonglin, 2018-08-31:
Set the milestone to "TBD"

jajanusz, 2018-08-31:
There are only just alloc tests giving error - they don't even fail, they just crash with message:
```
1..66
*WARNING* Unhandled user exception: StoreProhibitedCause (0x0000118c)
( Xtensa ISS ) *WARNING* run exited with status 'virtual breakpoint' 
```
So I guess the problem is not with the tests and may go away when we solve memory problems.

jajanusz, 2018-10-10:
It just doesn't break CI because buffer tests are disabled, so issue cannot be closed - need to fix buffer tests for simulator. I'll look into this