thesofproject/sof #282: Host platform build fails

singalsu, 2018-08-30:
The x86 compiled libraries and testbench binary are used by component tests in SOFT/test. The host build fails after next branch merge to master.

The host build is done with
´´´
./autogen.sh 
./configure --with-arch=host --enable-library=yes --host=x86_64-unknown-linux-gnu --prefix=$PWD/../host-root/
make
´´´
The first error is
´´´
make[2]: Entering directory '/home/singalsu/work/current/sof/sof/src/ipc'
/bin/bash ../../libtool  --tag=CC   --mode=compile gcc -DHAVE_CONFIG_H -I. -I../../src/include    -O3 -Wall -Werror -Wl,-EL -Wmissing-prototypes -g -I /home/singalsu/work/current/sof/sof/src/library/include -I /home/singalsu/work/current/sof/sof/src/include -I /home/singalsu/work/current/sof/sof/src/arch/host/include -g -O2 -MT libsof_ipc_la-ipc.lo -MD -MP -MF .deps/libsof_ipc_la-ipc.Tpo -c -o libsof_ipc_la-ipc.lo `test -f 'ipc.c' || echo './'`ipc.c
libtool: compile:  gcc -DHAVE_CONFIG_H -I. -I../../src/include -O3 -Wall -Werror -Wl,-EL -Wmissing-prototypes -g -I /home/singalsu/work/current/sof/sof/src/library/include -I /home/singalsu/work/current/sof/sof/src/include -I /home/singalsu/work/current/sof/sof/src/arch/host/include -g -O2 -MT libsof_ipc_la-ipc.lo -MD -MP -MF .deps/libsof_ipc_la-ipc.Tpo -c ipc.c  -fPIC -DPIC -o .libs/libsof_ipc_la-ipc.o
In file included from ipc.c:39:0:
../../src/include/sof/alloc.h:86:31: error: ‘PLATFORM_DCACHE_ALIGN’ undeclared here (not in a function); did you mean ‘PLATFORM_DMA_TIMEOUT’?
 } __attribute__ ((__aligned__(PLATFORM_DCACHE_ALIGN)));
                               ^~~~~~~~~~~~~~~~~~~~~
                               PLATFORM_DMA_TIMEOUT
Makefile:687: recipe for target 'libsof_ipc_la-ipc.lo' failed
´´´

singalsu, 2018-08-30:
@jajanusz This may have some common with your reported issue with UT.

lgirdwood, 2018-08-30:
@singalsu easy fix, can you define this for host build as sizeof(uint32_t)

singalsu, 2018-08-30:
@lgirdwood Adding that to src/arch/host/include/arch/sof.h seems to work. I'll continue with next fixes and see if I can get this built...

singalsu, 2018-08-30:
Yep, I can compile it now. I'll clean up the work make a pull request.

singalsu, 2018-08-30:
Fix proposal is in PR #283 .

jajanusz, 2018-08-31:
Still fails for me after #283  because of 'uninitialized' char pointers warnings in gcc version 5.4.0, what versions do we support?

lgirdwood, 2018-08-31:
@jajanusz gcc 5.x is deprecated, we are all using gcc 7.x or 8.x. i.e. Ubuntu ships with 7.3 now. Can you paste the build errors.
@singalsu which version of gcc do you have, I guess you don't see this ?

jajanusz, 2018-08-31:
@lgirdwood 
Error is:
```
testbench.c: In function ‘main’:
testbench.c:98:29: error: ‘comp_token’ may be used uninitialized in this function [-Werror=maybe-uninitialized]
   lib_table[index].handle = dlopen(token1, RTLD_LAZY);
                             ^
testbench.c:70:20: note: ‘comp_token’ was declared here
  char *lib_token, *comp_token;
                    ^
cc1: all warnings being treated as errors
```
It just complains that char pointers are unintialized while it doesn't matter cos strkok_r will initialize them. I tested with gcc 7.3 and it also showed that error

lgirdwood, 2018-08-31:
@jajanusz great thanks !

closed 2018-08-31 by lgirdwood