thesofproject/sof #286: work_reschedule_default could not accept too small timeout

xiulipan, 2018-08-31:
If the timeout is too small we will get some too small delta for the queue timeout.
EG.
if we call 
`work_reschedule_default(&trace_data->dmat_work,  DMA_TRACE_RESCHEDULE_TIME);`
at 1000us.
The DMA_TRACE_RESCHEDULE_TIME is now 5us. and the `dmat_work->timeout` is 1005us.
Then in the queue_get_next_timeout
```
/* calculate next timeout */
static void queue_get_next_timeout(struct work_queue *queue)
{
        struct list_item *wlist;
        struct work *work;
        uint64_t delta = ULONG_LONG_MAX;
        uint64_t current; //suppose we run into this in 1010us
...
        ticks = current = work_get_timer(queue);

        /* find time for next work */
        list_for_item(wlist, &queue->work) {

                work = container_of(wlist, struct work, list);

                d = calc_delta_ticks(current, work->timeout); 
                /* If the work->timeout be smaller than current *
                  * will get the delta very big. Next IRQ will ignore *
                  * the timeout 1005 but to some timeout list 1050 */

                /* is work next ? */
                if (d < delta) {
                        ticks = work->timeout;
                        delta = d;
                }
        }

        queue->timeout = ticks; //queue time out is 1050us now
}
```
Then the IRQ will comes at 1050 us and we have some window in `is_work_pending`, then the dmat_work will be ignore here. And won't have time to reschedule itself again.

xiulipan, 2018-08-31:
Need to exceed DMA_TRACE_RESCHEDULE_TIME to workaround this bug now.

mengdonglin, 2018-10-10:
@xiulipan Do you need firmware engineer to provide a formation solution for this issue?

lgirdwood, 2018-10-16:
@xiulipan we could define a PLATFORM_  min limit for this value (and round up anything smaller that this).

xiulipan, 2018-10-17:
@lgirdwood 
We may need to tune this value.

Or just make fix this limitation by schedule the work to future if it missed schedule.

tlauda, 2018-10-23:
@xiulipan @lgirdwood This issue shouldn't be possible to reproduce anymore. Work queue was reimplemented to process tasks every 1 ms (configurable in the future) in order to synchronise all cores. Can you confirm?

xiulipan, 2018-10-23:
@tlauda 

As I mentioned in this issue

> Then the IRQ will comes at 1050 us and we have some window in is_work_pending, then the dmat_work will be ignore here. And won't have time to reschedule itself again.

```c
static int is_work_pending(struct work_queue *queue)
{
...
			if (work->timeout >= win_start && work->timeout <= win_end) {
// The basic problem is that, if work->timeout is not in the window, we will ignore the work run

```

Now even if we try to make queue run for every 1ms, but the re  schedule is counted with us. So there may still be some work->timeout not inside the time window if the core is too busy.

tlauda, 2018-10-23:
@xiulipan Current window for work queue is 2 ms. I assume that if work queue won't be triggered for such a long time, there is something very wrong with our firmware in general.

xiulipan, 2018-10-29:
@tlauda 
Sorry for the late reply. I think the new method should solve the problem

Let us close this issue as the whole work queue is refined.

closed 2018-10-29 by xiulipan