thesofproject/sof #291: Cmocka unit tests for all platforms

jajanusz, 2018-09-01:
I made cmocka fork that enables us to build cmocka for any compiler and platform that supports stdlib, you can read how to build here: https://github.com/jajanusz/cmocka/blob/master/README.md#compiling-for-embedded

I tested it with xtensa compiler, it works for all platforms - apl, cnl, etc
However our baytrail toolchain is missing std lib that is essential for cmocka, but if we will have richer toolchains we will be able to have cmocka for these platforms also.

@lgirdwood Should it be for example in thesofproject/{cmocka,tools,whatever} repo instead of in my repo? If so please put it wherever is suitable.

jajanusz, 2018-09-03:
@lgirdwood I'd like to update the wiki about unit tests - will you move our custom cmocka somewhere else or can I use my link eveywhere?

lgirdwood, 2018-09-03:
@jajanusz please adds the cmocka examples and docs to sof-docs. I've also added xtensa-newlib to github that we can use as our C library for tests. @xiulipan will be able to help adding newlib and cmocka testing as part of Docker CI when he is free.

jajanusz, 2019-03-26:
Closing, #736 is more specific

closed 2019-03-26 by jajanusz