thesofproject/sof #295: DSP panic when simultaneously playback and capture

ZhendanYang, 2018-09-03:
Summary：playback and capture at the same time will lead to DSP panic. This issue exists on APL, BYT and CNL, and this exists on sof master, stable 1.2 glk-003 branch.
sof : 

Reproduce steps:
aplay -D hw:0,0 -c 2 -r 48000 -f s16_le /dev/zero --period-size=192 --buffer-size=384 & arecord -D hw:0,6 -c 2 -r 48000 -f s32_le /dev/null --period-size=192 --buffer-size=384 -vv -i

Environment；
master: 9d552d532bb9f57c05a9dae0944057b2aaf0ec9b
stable 1.2: 7346243e596cd91ff4342308e680392600496b3a
glk-003: 0ead9e762b36b87cd59afe44e90f912978a79fad
kernel: 2d3478a6087dd99d926930cf7f2c91f2d23cd8e5
sof tool:d40087fdd4ecaf0329059ef63527e3d4664b2d59

Logs:
on UP^2:
[dmesg-pc.log](https://github.com/thesofproject/sof/files/2344588/dmesg-pc.log)
[rmbox-pc.log](https://github.com/thesofproject/sof/files/2344589/rmbox-pc.log)
[rmbox-t-pc.log](https://github.com/thesofproject/sof/files/2344590/rmbox-t-pc.log)

lgirdwood, 2018-09-03:
@ZhendanYang please check if this is duplicate before logging. The scheduler list corruption can cause lots of different looking bugs but they all share similarities i.e.

EXCCAUSE 0x0000000d (always true)
EXCVADDR 0x00000040 (always < 0x100)
PS       0x00070025 (last digit is always 5).

Please do not log any other issues with the above characteristics until speaking with @RanderWang and @xiulipan

ZhendanYang, 2018-09-10:
Can't reproduce on APL and CNL with stable-1.2 branch. Test on BYT is blocked by #274 .

mengdonglin, 2018-09-10:
Will create new bug to track IPC timeout with master branch on CNL. 
CNL has no DSP panic now.

ZhendanYang, 2018-09-17:
Can't reproduce on latest code base:
sof stable-1.2: f17abe644c3b583213914b3d98e27f55786ef8c7
kernel sof-dev: 50792bc
soft: d77beaf
Instead of DSP panic, simultaneously playback and capture will cause ipc error.
[  528.473892] sof-audio sof-audio: generating page table for 000000004265c04c size 0xc00 pages 1
[  528.473895] sof-audio sof-audio: pfn i 0 idx 0 pfn 343a0
[  528.473899] sof-audio sof-audio: stream_tag 0
[  528.473938] sof-audio sof-audio: ipc tx: 0x60010000
[  528.474411] sof-audio sof-audio: error: ipc error for 0x60010000 size 0x14
[  528.474432] sof-audio sof-audio: ASoC: sof-audio hw params failed: -22
[  528.474439] sof-audio sof-audio: pcm: free stream 0 dir 0
[  528.474453] sof-audio sof-audio: ipc tx: 0x60030000
[  528.474566] sof-audio sof-audio: ipc tx succeeded: 0x60030000
[  528.474577]  Passthrough: ASoC: hw_params FE failed -22

[dmesg-pc.log](https://github.com/thesofproject/sof/files/2387674/dmesg-pc.log)
[rmbox-pc.log](https://github.com/thesofproject/sof/files/2387675/rmbox-pc.log)
[rmbox-t-pc.log](https://github.com/thesofproject/sof/files/2387676/rmbox-t-pc.log)

xiulipan, 2018-09-18:
@ZhendanYang 
Please re test the case.

Should be fixed on stable-1.2

ZhendanYang, 2018-09-19:
Verified. Can't reproduce on latest code base.
Test env:
sof stable-1.2: 7dd4b1d
kernel sof-dev: bc51a6f
soft master: d77beaf

closed 2018-09-19 by ZhendanYang