thesofproject/sof #307: ipc error when playback on BYT

ZhendanYang, 2018-09-05:
Summary:
Playback can't work and ipc error is found in dmesg.

Reproduce step:
aplay -D hw:0,0 -c 2 -r 48000 -f s16_le music.wav --period-size=192 --buffer-size=384 -vv -i

Environment:
sof-stable 1.2 : 90b455e07468e4b6ce929852bad0cdedc749587e
kernel : 7889a7ade4e20a81e297e3f5a5aea3b5d7fa55ef
sof tool: d2721194f7a36a4586b670e259c9f696cf4688be
tplg : test-ssp2-mclk-0-I2S-volume-s16le-s16le-48k-19200k-codec.tplg

Log:
[dmesg-ipcerror.log](https://github.com/thesofproject/sof/files/2351585/dmesg-ipcerror.log)
[rmbox-ipcerror.log](https://github.com/thesofproject/sof/files/2351586/rmbox-ipcerror.log)
[rmbox-t-ipcerror.log](https://github.com/thesofproject/sof/files/2351587/rmbox-t-ipcerror.log)

xiulipan, 2018-09-18:
@ZhendanYang 
Retest please, should be fixed.

ZhendanYang, 2018-09-19:
Verified. Can't reproduce on latest code base.
Test env:
sof stable-1.2: 7dd4b1d
kernel sof-dev: bc51a6f
soft master: d77beaf

closed 2018-09-19 by ZhendanYang