thesofproject/sof #327: DMIC capture accepts 4ch recordin but gives corrupted audio when configured for 2ch

singalsu, 2018-09-07:
The number of channels (used PDM stereo controllers count) is defined in topology parameters. Now the capture runs but results corrupted audio. Example of this kind of violation is to use topology sof-apl-nocodec.tplg or sof-apl-da7219.m4 and run capture like:

time arecord -Dhw:0,6 -c 4 -f S32_LE -r 48000 -d10 tmp.wav --dump-hw-params

The time print shows that such capture will take 20s instead of requested 10. The 2ch audio samples will be distributed to all 4ch and recording will sound double speed and double pitch.

The proper driver behavior should be to reject capture of non-matching channels count.

lgirdwood, 2018-09-07:
@singalsu ok so dmic driver should reject channels if not supported by topology. Host driver does not have internal pipeline logic to do this rejection.

singalsu, 2018-10-12:
As update the fix is under construction in SOFT PR #90.

michalgrodzicki, 2018-10-17:
@singalsu PR #90 is closed, what is a status of this issue?

singalsu, 2018-10-18:
@michalgrodzicki @lgirdwood @juimonen No, SOFT PR #90 is not. I feel it could be useful for merge if #92 that has a wider scope takes still time to develop. Though #90 seems to need an update. What is your recommendation. This has been open for a long time with first simple fix #90 on hold.

singalsu, 2018-10-22:
This issue can be closed. I just checked that arecord command with four channels is rejected (and two channels works).

```
$ arecord -Dhw:0,6 -c 2 -f S32_LE -r 48000 -d 10 tmp.wav
Recording WAVE 'tmp.wav' : Signed 32 bit Little Endian, Rate 48000 Hz, Stereo
$ arecord -Dhw:0,6 -c 4 -f S32_LE -r 48000 -d 10 tmp.wav
Recording WAVE 'tmp.wav' : Signed 32 bit Little Endian, Rate 48000 Hz, Channels 4
arecord: set_params:1305: Channels count non available
```

closed 2018-10-22 by singalsu