thesofproject/sof #328: EQ settings fail to persist in idle due to suspend/resume

singalsu, 2018-09-07:
The kernel and firmware have been just updated with EQ feature enabling. The topology and binary tool for EQ configuration will be later up-streamed with SOFT.

The DSP suspend-resume cycle currently erases the filter coefficients configuration that has been previously sent from user space to DSP via ASoC ext bytes put feature. Without configuration the EQ switches to bypass configuration. The issue can be heard in playback as "no effect" even if successfully configured.

This issue will impact every component that uses the same bytes configuration system.

There's need to develop some system to retrieve the COMP_CMD_SET_DATA / SOF_CTRL_CMD_BINARY settings from components before suspend (use COMP_CMD_GET_DATA)  and  re-send the COMP_CMD_SET_DATA when resuming.

lgirdwood, 2018-09-07:
This needs to be cached and @ranj063 has already created infrastructure to do this. Lets add a generic cache for each binary kcontrol.

ranj063, 2018-09-07:
@lgirdwood I will fix the issue with caching the byte controls data for the EQ.

ranj063, 2018-09-09:
@singalsu I have sent you a patch for retreiving the cached byte control data after resume. Could you please try it out?

singalsu, 2018-09-10:
That's wonderful, I'm testing it now! I do see an error with ABI magic and version not set. I'll see if I can fix that.

singalsu, 2018-09-10:
@ranj063 I got it working! In the code instead there's need to check that there's a valid IPC message stored to scontrol before sending to DSP. The 1st fail will with component IPC prevent success for later runtime. I'll send a new version patch back to you for check.

mengdonglin, 2018-09-17:
@ranj063 Would you please share the PR# here when it's ready?
Take this as a enhancement for the moment.

singalsu, 2018-09-18:
@mengdonglin This bug has been fixed by PR https://github.com/thesofproject/linux/pull/121

mengdonglin, 2018-10-10:
Okay, I closed it. Sorry for the late reply.

closed 2018-10-10 by mengdonglin