thesofproject/sof #331: Noise in channels on GLK

cujomalainey, 2018-09-07:
I'm seeing a lot of noise in GLK  occasionally in channels, sometimes the noise is only for a split second at the beginning sometimes it continues. This is at the top of stable-1.2 and it doesn't appear to matter what pipeline I use to trigger it issue (i.e. headphones and speakers are both reproducible.) 

To verify I used speaker-test to generate a sine wave at 440Hz and used my phone to verify.
[image]

This is an analysis of the speaker when its producing the noise, it does not fluctuate at all, I'm still trying to hunt down the cause.
[image]

I am not seeing any under-runs in the kernel or the firmware or any errors of any other sort. Any suggestions on where to look would be awesome.

cujomalainey, 2018-09-07:
This may or may not be related, but I am seeing the firmware boot after every time I run speaker-test.

```
[ 9086.395796] sof-audio sof-audio: tag 1 not opened!
[ 9086.417647] sof-audio sof-audio: DSP core(s) enabled? 0 : core_mask 3
[ 9088.199890] sof-audio sof-audio: unstall/run core: core_mask = 1
[ 9088.199897] sof-audio sof-audio: DSP core(s) enabled? 1 : core_mask 1
[ 9088.266682] sof-audio sof-audio: ipc: DSP is ready 0x70000000 offset 0x81000
[ 9088.266762] sof-audio sof-audio:  Firmware info: version 1.1-5df6c build 84 on Sep  7 2018:17:56:23
[ 9088.266896] sof-audio sof-audio: cannot create debugfs entry.
[ 9088.266902] sof-audio sof-audio: cannot create debugfs entry.
[ 9088.266906] sof-audio sof-audio: cannot create debugfs entry.
[ 9088.266910] sof-audio sof-audio: cannot create debugfs entry.
[ 9088.266913] sof-audio sof-audio: cannot create debugfs entry.
[ 9088.266917] sof-audio sof-audio: cannot create debugfs entry.
[ 9088.266921] sof-audio sof-audio: cannot create debugfs entry.
[ 9088.266925] sof-audio sof-audio:  mailbox upstream 0x81000 - size 0x1000
[ 9088.266927] sof-audio sof-audio:  mailbox downstream 0xa0000 - size 0x2000
[ 9088.266928] sof-audio sof-audio:  stream region 0xc1000 - size 0x1000
[ 9088.266976] sof-audio sof-audio: Firmware download successful, booting...
[ 9088.266978] sof-audio sof-audio: firmware boot complete
[ 9088.285650] sof-audio sof-audio: period_bytes:0x3600
```

ranj063, 2018-09-07:
@cujomalainey Could you please give the glk-0003-drop-stable branch a try? I had similar issues with playback which has been fixed by PR#326.

ranj063, 2018-09-07:
@cujomalainey the firmware boot that you are seeing is likely because runtime_pm is enabled and the DSP is going in and out of D3

cujomalainey, 2018-09-07:
@ranj063 yep I cannot reproduce it on that branch and if I checkout before the merge I always get noise. I tried cherry-picking over the change but it wasn't merging nicely so I left it for now. 

Ah that explains it, and I verified by disabling runtime_pm. So many new features since I last time I looked at this project last year.

Thanks @ranj063

closed 2018-09-07 by cujomalainey