thesofproject/sof #337: Stress test: press space key when playback or capture will cause ipc timeout on APL and CNL

ZhendanYang, 2018-09-10:
Summary:
Stress test: press space key will cause ipc timeout on APL and CNL.

[   54.688310] sof-audio sof-audio: error: ipc timed out for 0x60070000 size 0xc
[   54.688330] sof-audio sof-audio: status: fw entered - code 00000005
[   54.688356] sof-audio sof-audio: error: unexpected fault 0x00000000 trace 0x00004000
[   54.688364] sof-audio sof-audio: error: waking up any trace sleepers
[   54.688388]  DMIC01: ASoC: trigger FE failed -110
[   54.688405] sof-audio sof-audio: pcm: trigger stream 6 dir 1 cmd 3
[   54.832432] sof-audio sof-audio: ipc rx: 0x90020000
[   54.832454] sof-audio sof-audio: ipc rx done: 0x90020000
[   55.008294] sof-audio sof-audio: error: ipc timed out for 0x60060000 size 0xc
[   55.008314] sof-audio sof-audio: status: fw entered - code 00000005
[   55.008340] sof-audio sof-audio: error: unexpected fault 0x00000000 trace 0x00004000
[   55.008348] sof-audio sof-audio: error: waking up any trace sleepers
[   55.008373]  DMIC01: ASoC: trigger FE failed -110

Reproduce step:
1. arecord -D hw:0,6 -c 2 -r 48000 -f s32_le test.wav --period-size=192 --buffer-size=384 -vv -i
2. press space key for a while

Environment:
sof-master: 7abb55
soft: d77bea
kernel: sof-dev 66dc735d45f64fcfc868c3772bb2c4c52446c000
tplg: sof-apl-nocodec.tplg

Log:
[dmesg-ipctimeout.log](https://github.com/thesofproject/sof/files/2365687/dmesg-ipctimeout.log)
[rmbox-ipctimeout.log](https://github.com/thesofproject/sof/files/2365688/rmbox-ipctimeout.log)
[rmbox-t-ipctimeout.log](https://github.com/thesofproject/sof/files/2365689/rmbox-t-ipctimeout.log)

ZhendanYang, 2018-09-10:
IPC timeout can also occur when simultaneous playback and capture on CNL.

[  136.800332] sof-audio sof-audio: error: ipc timed out for 0x60010000 size 0x60
[  136.800350] sof-audio sof-audio: status: fw entered - code 00000005
[  136.800440] sof-audio sof-audio: error: unexpected fault 0x00000000 trace 0x00004000
[  136.800447] sof-audio sof-audio: error: waking up any trace sleepers
[  136.800534] sof-audio sof-audio: error: got wrong posn offset 0xffff9d55 for PCM -110
[  136.800543] sof-audio sof-audio: ASoC: sof-audio hw params failed: -110
[  136.800555] sof-audio sof-audio: pcm: free stream 0 dir 0
[  137.120321] sof-audio sof-audio: error: ipc timed out for 0x60030000 size 0xc
[  137.120344] sof-audio sof-audio: status: fw entered - code 00000005
[  137.120395] sof-audio sof-audio: error: unexpected fault 0x00000000 trace 0x00004000
[  137.120400] sof-audio sof-audio: error: waking up any trace sleepers
[  137.120422]  Passthrough: ASoC: hw_params FE failed -110

Log:
[rmbox-t-ipctimeout.log](https://github.com/thesofproject/sof/files/2365736/rmbox-t-ipctimeout.log)
[rmbox-ipctimeout.log](https://github.com/thesofproject/sof/files/2365738/rmbox-ipctimeout.log)
[dmesg-ipctimeout.log](https://github.com/thesofproject/sof/files/2365739/dmesg-ipctimeout.log)

mengdonglin, 2018-09-10:
@lgirdwood Need your help to merge stable 1.2 branch which contains the fix

ZhendanYang, 2018-09-14:
Test with latest code base and this issue can't reproduce, so I'll close it.
Test env:
kernel: sof-dev ae5f868
sof: matster f17abe644c3b583213914b3d98e27f55786ef8c7
soft: d77beaf

closed 2018-09-14 by ZhendanYang