thesofproject/sof #344: PCM I/O write error during aplay HDMI playback on GLK

keqiaozhang, 2018-09-11:
This issue happens when doing long time playback through HDMI.

How to reproduce:
1. speaker-test -D hw:0,5 -c 2 -t wav

Error message:
Write error: -5,Input/output error
xrun_recovery failed: -5,Input/output error
Transfer failed: Input/output error

Test env:
kernel-cros-sof-v4.14-rebase: e4bc814c982d5
sof-glk-003-drop-stable: f245831ea
soft-master:d77beaf8b

[dmesg.log](https://github.com/thesofproject/sof/files/2369593/dmesg.log)
[rmbox.log](https://github.com/thesofproject/sof/files/2369595/rmbox.log)
[rmbox-t.log](https://github.com/thesofproject/sof/files/2369596/rmbox-t.log)

keqiaozhang, 2018-09-17:
HSD link: https://hsdes.intel.com/appstore/article/#/1506756720

dabekjakub, 2018-10-11:
@keqiaozhang 

Problem couldn't be reproduced on
FW branch master: 7996f1ac10b8a8c14c87a3752d3e0dbe94d09709

Please confirm if it is still occurring and the length of playback needed for this bug to occur.

keqiaozhang, 2018-10-11:
Thanks @dabekjakub .
The possibility of this issue is low and also need long time playback, I will try to reproduce it tomorrow.