thesofproject/sof #345: IPC timed our error when speaker headset and HDMI playback work at some time on GLK

keqiaozhang, 2018-09-11:
This issue happens at the second time. it's OK during the first run.
How to reproduce:
1. speaker-test -Dhw:0,0 -c 2 -t wav & speaker-test -Dhw:0,1 -c 2 -t wav & speaker-test -Dhw:0,5 -c 2 -t wav

Error log:
[  224.504065] sof-audio sof-audio: error: ipc timed out for 0x60040000 size 0xc
[  224.504078] sof-audio sof-audio: status: fw entered - code 00000005
[  224.504205] sof-audio sof-audio: error: unexpected fault 0x00000000 trace 0x00004000
[  224.504207] sof-audio sof-audio: error: waking up any trace sleepers
[  224.504213] sof-audio sof-audio: error: ipc timed out for 0x60010000 size 0x60
[  224.504219]  Headset: ASoC: trigger FE failed -110
[  224.504222] sof-audio sof-audio: status: fw entered - code 00000005
[  224.504229] sof-audio sof-audio: pcm: trigger stream 1 dir 0 cmd 0
[  224.504350] sof-audio sof-audio: error: unexpected fault 0x00000000 trace 0x00004000
[  224.504354] sof-audio sof-audio: error: waking up any trace sleepers
[  224.504364] sof-audio sof-audio: error: got wrong posn offset 0x6967b00 for PCM -110
[  224.504368] sof-audio sof-audio: ASoC: sof-audio hw params failed: -110
[  224.504372]  Speakers: ASoC: hw_params FE failed -110
[  224.504384] sof-audio sof-audio: pcm: free stream 0 dir 0
[  224.696357] sof-audio sof-audio: PCM: stream 5 dir 0 DMA position 0 DAI position 0
[  224.808345] sof-audio sof-audio: error: ipc timed out for 0x60050000 size 0xc
[  224.808366] sof-audio sof-audio: status: fw entered - code 00000005
[  224.808501] sof-audio sof-audio: error: unexpected fault 0x00000000 trace 0x00004000
[  224.808508] sof-audio sof-audio: error: waking up any trace sleepers
[  224.808531] sof-audio sof-audio: error: ipc timed out for 0x60030000 size 0xc
[  224.808539] sof-audio sof-audio: status: fw entered - code 00000005
[  224.808543]  Headset: ASoC: trigger FE failed -110
[  224.808673] sof-audio sof-audio: error: unexpected fault 0x00000000 trace 0x00004000
[  224.808676] sof-audio sof-audio: error: waking up any trace sleepers
[  224.810061] sof-audio sof-audio: pcm: close stream 0 dir 0
[  225.200159] sof-audio sof-audio: pcm: trigger stream 5 dir 0 cmd 0
[  225.312472] sof-audio sof-audio: pcm: free stream 1 dir 0
[  225.504326] sof-audio sof-audio: error: ipc timed out for 0x60050000 size 0xc
[  225.504345] sof-audio sof-audio: status: fw entered - code 00000005
[  225.504475] sof-audio sof-audio: error: unexpected fault 0x00000000 trace 0x00004000
[  225.504481] sof-audio sof-audio: error: waking up any trace sleepers
[  225.504502]  HDMI1: ASoC: trigger FE failed -110
[  225.616075] sof-audio sof-audio: error: ipc timed out for 0x60030000 size 0xc
[  225.616088] sof-audio sof-audio: status: fw entered - code 00000005
[  225.616220] sof-audio sof-audio: error: unexpected fault 0x00000000 trace 0x00004000
[  225.616223] sof-audio sof-audio: error: waking up any trace sleepers
[  225.616264] sof-audio sof-audio: pcm: free stream 5 dir 0
[  225.920324] sof-audio sof-audio: error: ipc timed out for 0x60030000 size 0xc
[  225.920342] sof-audio sof-audio: status: fw entered - code 00000005
[  225.920470] sof-audio sof-audio: error: unexpected fault 0x00000000 trace 0x00004000
[  225.920475] sof-audio sof-audio: error: waking up any trace sleepers
[  225.932496] sof-audio sof-audio: pcm: close stream 1 dir 0
[  225.933526] sof-audio sof-audio: pcm: close stream 5 dir 0
[  231.712475] sof-audio sof-audio: error: ipc timed out for 0x40010000 size 0x20
[  231.712493] sof-audio sof-audio: status: fw entered - code 00000005
[  231.712639] sof-audio sof-audio: error: unexpected fault 0x00000000 trace 0x00004000
[  231.712642] sof-audio sof-audio: error: waking up any trace sleepers
[  231.712653] sof-audio sof-audio: error: ctx_save ipc error during suspend -110
[  231.712664] sof-audio-pci 0000:00:0e.0: can't suspend (snd_sof_runtime_suspend [snd_sof] returned -110)

Test env:
kernel-cros-sof-v4.14-rebase: e4bc814c982d5
sof-glk-003-drop-stable: f245831
soft-master:d77beaf8b

[dmesg.log](https://github.com/thesofproject/sof/files/2369692/dmesg.log)
[rmbox.log](https://github.com/thesofproject/sof/files/2369693/rmbox.log)
[rmbox-t.log](https://github.com/thesofproject/sof/files/2369695/rmbox-t.log)

keqiaozhang, 2018-11-20:
This issue still exists.
Tested with:
Kernel -cros-sof-v4.14-rebase: bf94547
DSP FW-glk-004-drop-stable: 1f799207
Topology-master: 8cc1dadd

lgirdwood, 2018-11-20:
@michalgrodzicki looks like something on the stream close path is not being cleanly reset if its not working second time. @mmaka1 any ideas ?

keyonjie, 2018-12-18:
set it to P2 as this is actually critical.

mengdonglin, 2018-12-25:
This issue https://github.com/thesofproject/sof/issues/345 is a subset of  issue https://github.com/thesofproject/sof/issues/743

michalgrodzicki, 2019-03-26:
#743 is closed, what is a status here?

lgirdwood, 2019-03-26:
@keqiaozhang yeah, I think this should be closed too. Can you retest and confirm.

keqiaozhang, 2019-03-27:
Confirmed that this issue is fixed, close this bug.

closed 2019-03-27 by keqiaozhang