thesofproject/sof #346: silence issue when using aplay to check the arecord data

keqiaozhang, 2018-09-11:
Test with headset: use arecord to capture audio, then use aplay to check the recorded data, but cannot hear anything. but after hot-plug the headset,  you can hear the sound that you recorded. repetition rate is 100%.  I also tried to play with speaker, no such issue.
How to reproduce:
1. arecord -D hw:0,1 -f s32_le -c 2 -r 48000 test.wav
2. aplay -D hw:0,1 test.wav (you can hear nothing at all)
3. do headset hot-plug during the playback, then you can hear the sound.

mengdonglin, 2018-09-14:
This seems not a firmware bug but more likely to be a codec driver issue.

mengdonglin, 2018-09-21:
Close this issue since it's not firmware issue, tracking it by kernel issue: https://github.com/thesofproject/linux/issues/154

closed 2018-09-21 by mengdonglin