thesofproject/sof #347: No HDMI/DP audio output option in audio setting on GLK

keqiaozhang, 2018-09-11:
With sof, there's no HDMI/DP audio output option in audio setting(chrome OS, HDMI/DP displayer 
 is connected), like sof.PNG. The normal state should be like normal.PNG.

[image]
[image]

cujomalainey, 2018-09-11:
Does CRAS recognize the HDMI output properly?

mengdonglin, 2018-09-12:
@cujomalainey, how to check if CRAS recognize the HDMI output properly? 
aplay -l can list the devices.  We're not sure if there are any other gap on ELD detection or UCM for CRAS to expose the 'HDMI1' as an output.

cujomalainey, 2018-09-12:
Also please send the path of where you are putting the UCM files

naveen-manohar, 2018-09-13:
Try using attached ucm, copied to path /usr/share/alsa/ucm/
[sof-glkda7219max.2mic.zip](https://github.com/thesofproject/sof/files/2379146/sof-glkda7219max.2mic.zip)

.. Also ensure that the machine driver is compatible for HDMI Audio.
https://chromium-review.googlesource.com/c/chromiumos/third_party/kernel/+/1220403

mengdonglin, 2018-09-14:
@naveen-manohar Thank you!  @keqiaozhang verified it works.
@ranj063 Could you help to back port the machine driver patch to cros-4.14-rebase?