thesofproject/sof #349: metallic noise when DP/HDMI audio playabck on GLK

keqiaozhang, 2018-09-11:
This noise issue is easily triggered by pause/release.

How to reproduce:
1. aplay -Dhw:0,5 -r 48000 -c 2 -f s16_le test.wav -i 
2. press the space key to do pause/release

[dmesg.log](https://github.com/thesofproject/sof/files/2370178/dmesg.log)
[rmbox.log](https://github.com/thesofproject/sof/files/2370180/rmbox.log)
[rmbox-t.log](https://github.com/thesofproject/sof/files/2370181/rmbox-t.log)

Test env:
kernel-cros-sof-v4.14-rebase: e4bc814c982d5
sof-glk-003-drop-stable: f245831
soft-master:d77beaf8b

keqiaozhang, 2018-09-17:
HSD link: https://hsdes.intel.com/appstore/article/#/1506756707

mengdonglin, 2018-09-18:
@keqiaozhang Please verify if this issue is solved by latest cros-4.14-rebase kernel.
It seems the kernel patch has been merged to solve this issue:
https://github.com/thesofproject/linux/commit/94f143a4a975430096e6ec24f15a7ab708626b51

@ranj063  I keep you as the bug owner before Keqiao verify and close this bug.

keqiaozhang, 2018-09-18:
Checked with latest cros-4.14-rebase kernel+ sof-glk003 firmware , this issue is still exists.
Checked with latest cros-4.14-rebase kernel+ sof-master firmware, pause/release HDMI audio will cause IPC error, HSD link：https://hsdes.intel.com/appstore/article/#/1506757990

mengdonglin, 2018-10-19:
This is a generic HD-A issue, not only for GLK.

mengdonglin, 2018-10-23:
@keqiaozhang could you help to check if PR https://github.com/thesofproject/sof/pull/504 can fix this issue except for the PAUSE/RESUME case?

mengdonglin, 2018-10-24:
@keqiaozhang if you have verified that the metallic noise is gone with HDMI audio playack w/o PAUSE/RESUME, would you please update here and close this issue,  file a new issue for metallic noise at PAUSE/RESUME?

mengdonglin, 2018-10-26:
Keqiao has proved there is no longer metallic noise w/o PAUSE/RESUME on Yorp. 
Please keep this issue open until metallic noise w/ PAUSE/RESUME is also fixed. Thanks!

mengdonglin, 2018-11-09:
This is a generic issue and also reproduced on ICL.

mengdonglin, 2018-11-15:
PR https://github.com/thesofproject/sof/pull/575 (merged to glk-004-drop-stable) fixed this metallic noise on HDMI playback pause/resume. Now the reproducible rate is low.  

@keqiaozhang Could you help to close this one and file new bug to reflect the latest status?

keqiaozhang, 2018-11-15:
Confirmed that this issue is fixed, so this issue can be closed.
But we found another noise issue and track it in https://github.com/thesofproject/sof/issues/583

closed 2018-11-15 by keqiaozhang