thesofproject/sof #354: [HOST] int-to-pointer cast breaks compilation

jajanusz, 2018-09-11:
While building host like:
```
./autogen.sh
./configure --with-arch=host --enable-library=yes --host=x86_64-unknown-linux-gnu --prefix=$PWD/../host-root/
make
```
For these versions of gcc:
```
gcc (Ubuntu 7.3.0-21ubuntu1~16.04) 7.3.0
gcc-5 (Ubuntu 5.4.0-6ubuntu1~16.04.10) 5.4.0 20160609
```
I get error:
```
In file included from ../../src/include/uapi/ipc.h:43:0,
                 from ../../src/include/sof/debug.h:37,
                 from ../../src/include/sof/trace.h:39,
                 from ../../src/include/sof/interrupt.h:38,
                 from ../../src/include/sof/lock.h:42,
                 from ipc.c:36:
../../src/include/sof/io.h: In function ‘io_reg_read’:
../../src/include/sof/io.h:39:11: error: cast to pointer from integer of different size [-Werror=int-to-pointer-cast]
  return *((volatile uint32_t*)reg);
           ^
../../src/include/sof/io.h: In function ‘io_reg_write’:
../../src/include/sof/io.h:44:4: error: cast to pointer from integer of different size [-Werror=int-to-pointer-cast]
  *((volatile uint32_t*)reg) = val;
    ^
../../src/include/sof/io.h: In function ‘io_reg_read16’:
../../src/include/sof/io.h:54:11: error: cast to pointer from integer of different size [-Werror=int-to-pointer-cast]
  return *((volatile uint16_t*)reg);
           ^
../../src/include/sof/io.h: In function ‘io_reg_write16’:
../../src/include/sof/io.h:59:4: error: cast to pointer from integer of different size [-Werror=int-to-pointer-cast]
  *((volatile uint16_t*)reg) = val;
```

Surrounding function like:
```
#pragma GCC diagnostic push
#pragma GCC diagnostic ignored "-Wint-to-pointer-cast"
static inline uint32_t io_reg_read(uint32_t reg)
{
	return *((volatile uint32_t*)reg);
}
#pragma GCC diagnostic pop
```
Works for host build, however will break for xcc, so we can do something like:
```
#if !defined __XCC__
#pragma GCC diagnostic push
#pragma GCC diagnostic ignored "-Wint-to-pointer-cast"
#endif
static inline uint32_t io_reg_read(uint32_t reg)
{
	return *((volatile uint32_t*)reg);
}
#if !defined __XCC__
#pragma GCC diagnostic pop
#endif
```

We can also fix it in makefiles but it's problematic cos you would have to add it to all compilation units that use that header, or to disable it at all what is not good.

What do you think is the best way to solve it?

singalsu, 2018-09-11:
@jajanusz Thanks, I'll look this

lgirdwood, 2018-09-11:
@jajanusz probably best to define a CONFIG_HOST macro in config.h (via autotools) that can be used to exclude a lot of code that should not be included in host build anyway or emit an error if someone tries to.

mengdonglin, 2018-09-29:
@jajanusz @singalsu May I know if you're working on this issue?

lgirdwood, 2018-10-01:
@mengdonglin I think this is now fixed

jajanusz, 2018-10-01:
For me host build is still broke, When I build current master, I get error:
```
In file included from ../../src/include/sof/schedule.h:41:0,
                 from /home/admin/sof/repo/SOF_FW/src/arch/host/include/arch/task.h:41,
                 from ../../src/include/sof/task.h:34,
                 from common_test.c:42:
../../src/include/sof/wait.h: In function ‘poll_for_completion_delay’:
../../src/include/sof/wait.h:169:36: error: ‘PLATFORM_DEFAULT_CLOCK’ undeclared (first use in this function); did you mean ‘PLATFORM_DEFAULT_DELAY’?
  uint64_t tick = clock_ms_to_ticks(PLATFORM_DEFAULT_CLOCK, 1) *

```

tlauda, 2018-10-01:
@jajanusz Pull latest changes. You are missing adee39984b609142d88e0316855d03fcfb94b219.

jajanusz, 2018-10-01:
I'm not sure if it was intended to work like that but bug was caused by taking includes from installation dir ($prefix) instead of from project, so if there are changes in includes and you don't clean your installation dir it fails. Anyway it works now with clean build & clean installation dir.

closed 2018-10-01 by jajanusz