thesofproject/sof #375: [glk][br-glk] System failed to enter S3 and S0ix

keqiaozhang, 2018-09-14:
System failed to suspend.

Test step:
sudo suspend_stress_test -c 25

dmesg:
[ 1322.489381] PM: Wakeup pending, aborting suspend
[ 1322.489392] last active wakeup source: rtc_cmos
[ 1322.489401] PM: suspend of devices aborted after 0.025 msecs
[ 1322.489403] PM: Some devices failed to suspend, or early wake event detected
[ 1322.489409] net lo: Unbalanced pm_runtime_enable!
[ 1322.489413] PM: resume of devices complete after 0.006 msecs
[ 1325.495355] HDMI HDA Codec ehdaudio0D2: hdac_hdmi_present_sense: disconnect for pin:port 5:0
[ 1325.495632] HDMI HDA Codec ehdaudio0D2: hdac_hdmi_present_sense: disconnect for pin:port 6:0
[ 1325.495922] HDMI HDA Codec ehdaudio0D2: hdac_hdmi_present_sense: disconnect for pin:port 7:0
[ 1325.497567] PM: Finishing wakeup.
[ 1325.497569] OOM killer enabled.
[ 1325.497571] Restarting tasks ... done.
[ 1325.501835] thermal thermal_zone6: failed to read out thermal zone (-61)
[ 1325.503534] PM: suspend exit
[ 1325.780357] init: powerd main process (3483) killed by TERM signal
[ 1326.501291] sof-audio sof-audio: ipc tx: 0x40010000: GLB_PM_MSG: CTX_SAVE
[ 1326.501370] sof-audio sof-audio: ipc tx succeeded: 0x40010000: GLB_PM_MSG: CTX_SAVE
[ 1326.501378] sof-audio sof-audio: tag 1 not opened!
[ 1326.523392] sof-audio sof-audio: DSP core(s) enabled? 0 : core_mask 3

Test env:
Environment:
sof: glk-003-drop-stable 95b4a6d
soft: master d77beaf
kernel: sof-dev 64cf3e3 with one HDMI audio enabling patch:https://chromium-review.googlesource.com/c/chromiumos/third_party/kernel/+/1220403/1

[dmesg.log](https://github.com/thesofproject/sof/files/2382429/dmesg.log)
[rmbox-t.log](https://github.com/thesofproject/sof/files/2382432/rmbox-t.log)

ranj063, 2018-09-19:
@keqiaozhang can you please verify if the issue is fixed with https://github.com/thesofproject/linux/pull/141?
It has been merged into cros-sof-v4.14-rebase

mengdonglin, 2018-09-21:
@keqiaozhang Can the fix work for your GLK device?

mengdonglin, 2018-09-21:
It seems Ranjani has fixed this issue, so add label "fixed" and wait for validation.

mengdonglin, 2018-09-24:
@ranj063 It seems system suspend/resume can work on your GLK, could you clarify which SOF firmware branch you're using? master or glk-003-drop-stable?

keqiaozhang, 2018-09-25:
@ranj063 @mengdonglin 
Tested with latest sof-master+ kernel-cros-sof-v4.14-rebase , audio device 0000:00:0e.0 failed to suspend: error -110. 
But check with sof-glk-003-drop-stable, system suspend/resume(s2idle) is supported, I tested with suspend_stress_test for more than 10 times and the runtime_status is also correct.
So this issue is fixed in sof-glk-003-drop-stable + kernel-cros-sof-v4.14-rebase, some patches need to cherry-pick to sof-master.

mengdonglin, 2018-09-25:
@ranj063  @mmaka1 would you like to sync on the patches that need to be cherry picked from glk-003-drop-stable to master?

mmaka1, 2018-09-25:
@mengdonglin There is a single one for FW b8d7166, no issue with merging this one.

lgirdwood, 2018-09-25:
@mengdonglin @mmaka1 now cherry picked and merged https://github.com/thesofproject/sof/commit/b8d7166ff14d435de83e447ad64cbde8a2ddfc52

keqiaozhang, 2018-09-27:
@mmaka1
This issue still can be reproduced in latest sof-master. so there's still have a gap between glk003 and master branch. Please help to confirm this.
[  255.313810] PM: suspend entry (s2idle)
[  255.313818] PM: Syncing filesystems ... done.
[  255.329545] PM: Preparing system for sleep (s2idle)
[  255.334709] PM: Suspending system (s2idle)
[  256.505791] PM: suspend of devices complete after 36.473 msecs
[  256.511865] sof-audio sof-audio: ipc tx: 0x40010000: GLB_PM_MSG: CTX_SAVE
[  256.815392] PM: Device 0000:00:0e.0 failed to suspend late: error -110
[  256.846483] PM: early resume of devices complete after 31.081 msecs
[  256.846487] PM: late suspend of devices aborted after 340.691 msecs
[  256.846488] PM: late suspend of devices failed
[  257.004407] PM: Device 0000:00:0e.0 failed to resume: error -5
[  257.239399] PM: resume of devices complete after 392.906 msecs
[  260.247214] PM: Finishing wakeup.
[  260.270538] PM: suspend exit

keqiaozhang, 2018-09-27:
@mmaka1 I also found that firmware will be panic after system is booted in sof-master, this issue maybe blocked by this regression.

keqiaozhang, 2018-10-19:
This is a kernel issue and fixed in cros-sof-v4.14-rebase branch. 
Fixed by PR https://github.com/thesofproject/linux/pull/186

closed 2018-10-19 by keqiaozhang