thesofproject/sof #378: [GLK][br-glk] trace log time stamp is disorder

keqiaozhang, 2018-09-17:
Keep run rmbox -t will get different outputs and the time stamp is disorder.

First run output:
.............
0x1f50 [4.087000]       delta [0.000002]        volume trg
0x1f60 [4.087002]       delta [0.000002]        dai trg
0x1f70 [4.087003]       delta [0.000002]        dai tsp
0x1f80 [4.087042]       delta [0.000038]        host DDi
0x1f90 [4.467072]       delta [0.380030]        ipc Cmd
0x1fa0 [4.467075]       delta [0.000003]        ipc tri
0x1fb0 [4.467078]       delta [0.000004]        pipe cmd
0x1fc0 [4.467081]       delta [0.000002]        host trg
0x1fd0 [4.467083]       delta [0.000002]        volume trg
0x1fe0 [4.467084]       delta [0.000002]        dai trg
0x1ff0 [4.467087]       delta [0.000002]        host DEn
0x2000 [5.390656]       delta [0.923569]        ipc Cmd
0x2010 [5.390659]       delta [0.000003]        ipc tri
0x2020 [5.390662]       delta [0.000004]        pipe cmd
0x2030 [5.390665]       delta [0.000002]        host trg
0x2040 [5.390667]       delta [0.000002]        volume trg
0x2050 [5.390668]       delta [0.000002]        dai trg
0x2060 [5.390670]       delta [0.000002]        dai tsp

Second run output:
..........
0x2150 [8.071399]       delta [0.000002]        dai trg
0x2160 [8.071400]       delta [0.000002]        dai tsp
0x2170 [8.072238]       delta [0.000838]        host DDi
0x2180 [8.072414]       delta [0.000176]        ipc Cmd
0x2190 [8.072417]       delta [0.000003]        ipc SFr
0x21a0 [8.072421]       delta [0.000004]        pipe PRe
0x21b0 [8.072423]       delta [0.000002]        host res
0x21c0 [8.072425]       delta [0.000002]        host DDi
0x21d0 [8.072434]       delta [0.000009]        volume res
0x21e0 [7.213343]       delta [********]        ipc Cmd

[dmesg.log](https://github.com/thesofproject/sof/files/2387229/dmesg.log)
[rmbox-t-2.log](https://github.com/thesofproject/sof/files/2387232/rmbox-t-2.log)
[rmbox-t-1.log](https://github.com/thesofproject/sof/files/2387233/rmbox-t-1.log)

keqiaozhang, 2018-09-17:
HSD link: https://hsdes.intel.com/appstore/article/#/1506756845

keqiaozhang, 2018-09-25:
This issue can also be reproduced in sof master branch.
checked with sof-master:5450a2fcd8.

keqiaozhang, 2018-09-25:
This issue can easy reproduce with suspend/resume.
Please check the rmbox -t before/after system S3.
How to reproduce:
1. sudo rmbox -t 
2. sudo suspend_stress_test
3. sudo rmbox -t

keqiaozhang, 2018-10-10:
Confirmed that this issue cannot reproduced with logger and we will switch to use logger for trace, so this bug can be closed.

closed 2018-10-10 by keqiaozhang

xiulipan, 2018-10-10:
@keqiaozhang 
If GLK branch switch to use logger?

keqiaozhang, 2018-10-10:
@xiulipan 
GLK branch is still using rmbox, but this branch is freeze for release.  The next firmware release for GLK will be branch form master.

xiulipan, 2018-10-29:
@keqiaozhang 
That would be fine. Will track with logger later.