thesofproject/sof #391: DMIC capture will cause FW panic on UP2

xiulipan, 2018-09-19:
Summary: second arecord with DMIC will cause FW panic

Platform: APL UP2
SOF commit: e4e3cb131cff6047436b741d2de001071e86fa03
Reproduce steps:
run `arecord -D hw:0,6 -f S32_LE tmp.wav -vv -i ` for twice

dmesg log:
[  110.608559] sof-audio sof-audio: error: Exception Cause: LoadProhibitedCause, A load referenced a page mapped with an attribute that does not permit loads
[  110.608568] sof-audio sof-audio: EXCCAUSE 0x0000001c EXCVADDR 0xfffffffc PS       0x00070925 SAR     0x0000001c
[  110.608581] sof-audio sof-audio: EPC1     0xbe010745 EPC2     0xbe010d0b EPC3     0xbe010d0b EPC4    0x00000000
[  110.608589] sof-audio sof-audio: EPC5     0xbe010d0b EPC6     0x00000000 EPC7     0x00000000 DEPC    0x00000000
[  110.608597] sof-audio sof-audio: EPS2     0x00060720 EPS3     0x00060720 EPS4     0x00000000 EPS5    0x00060720
[  110.608605] sof-audio sof-audio: EPS6     0x00000000 EPS7     0x00000000 INTENABL 0x000101d8 INTERRU 0x00000222
[  110.608612] sof-audio sof-audio: stack dump from 0xbe07fde0
[  110.608621] sof-audio sof-audio: 0xbe07fde0: 0xbe01caf0 0x00000000 0x0000001c 0x00000000
[  110.608629] sof-audio sof-audio: 0xbe07fde4: 0xbe061100 0x00000000 0x00000000 0x00000000
[  110.608638] sof-audio sof-audio: 0xbe07fde8: 0x0000001c 0x00000000 0xbe023619 0xbe02361f
[  110.608646] sof-audio sof-audio: 0xbe07fdec: 0x00060325 0x00000000 0x00060020 0x00000000
[  110.608654] sof-audio sof-audio: 0xbe07fdf0: 0xbe014932 0x00000000 0xbe04c8c0 0x00000000
[  110.608662] sof-audio sof-audio: 0xbe07fdf4: 0x60010000 0x00000000 0xbe023619 0x00000000
[  110.608671] sof-audio sof-audio: 0xbe07fdf8: 0x00000000 0x00000000 0x00000001 0xbe02c780
[  110.608679] sof-audio sof-audio: 0xbe07fdfc: 0x90020000 0x00000000 0xbe061700 0x00000000
[  110.608686] sof-audio sof-audio: error: waking up any trace sleepers

rmbox log:
0x10 [106.849519]       delta [106.753808]      ipc Cmd
0x20 [106.849523]       delta [0.000004]        ipc SAl
0x30 [106.849548]       delta [0.000025]        pipe Par
0x40 [106.849551]       delta [0.000003]        host par
0x50 [106.849560]       delta [0.000009]        host Dgt
0x60 [106.849564]       delta [0.000003]        host Dsc
0x70 [106.849569]       delta [0.000005]        dai par
0x80 [106.849572]       delta [0.000003]        dma Dgt
0x90 [106.849582]       delta [0.000011]        pipe pre
0xa0 [106.849585]       delta [0.000003]        host pre
0xb0 [106.849588]       delta [0.000003]        dai pre
0xc0 [106.851678]       delta [0.002090]        ipc Cmd
0xd0 [106.851682]       delta [0.000004]        ipc tri
0xe0 [106.851706]       delta [0.000025]        pipe cmd
0xf0 [106.851709]       delta [0.000003]        host trg
0x100 [106.851712]      delta [0.000003]        host DEn
0x110 [106.851748]      delta [0.000036]        dai trg
0x120 [106.851750]      delta [0.000003]        dai tsa
0x130 [106.851766]      delta [0.000016]        dmic tri
0x140 [106.851769]      delta [0.000003]        dmic sta
0x150 [106.851771]      delta [0.000002]        dmic ffa
0x160 [106.851774]      delta [0.000003]        dmic mfn
0x170 [106.851777]      delta [0.000002]        value 0x0000000000000001
0x180 [106.851779]      delta [0.000002]        value 0x0000000000000001
0x190 [106.851781]      delta [0.000002]        value 0x0000000000000001
0x1a0 [106.851783]      delta [0.000002]        value 0x0000000000000000
0x1b0 [106.851787]      delta [0.000004]        dmic mfn
0x1c0 [106.851789]      delta [0.000002]        value 0x0000000000000000
0x1d0 [106.851792]      delta [0.000002]        value 0x0000000000000000
0x1e0 [106.851794]      delta [0.000002]        value 0x0000000000000000
0x1f0 [106.851796]      delta [0.000002]        value 0x0000000000000000
0x200 [106.851799]      delta [0.000003]        dmic unr
0x210 [106.851807]      delta [0.000008]        dmic run

xiulipan, 2018-09-20:
Have some test and find that the bug is related to memory alloc.

```diff
diff --git a/src/lib/interrupt.c b/src/lib/interrupt.c
index b40761b..da2447c 100644
--- a/src/lib/interrupt.c
+++ b/src/lib/interrupt.c
@@ -58,6 +58,9 @@ static int irq_register_child(struct irq_desc *parent, int irq,
        /* init child */
        child = rzalloc(RZONE_RUNTIME, SOF_MEM_CAPS_RAM,
                        sizeof(struct irq_desc));
+
+       trace_error_value(0xabbbbeef);
+       trace_error_value((uint32_t)child);
        if (!child) {
                ret = -ENOMEM;
                goto finish;

diff --git a/src/audio/pipeline.c b/src/audio/pipeline.c
index 0c5779f..69ae209 100644
--- a/src/audio/pipeline.c
+++ b/src/audio/pipeline.c
@@ -728,6 +740,10 @@ static int host_copy_int(struct comp_dev *dev, bool preload_run)
                        return 0;
                }
        } else {
+               trace_host_error("cp4");
+               trace_error_value((uint32_t)hd);
+               trace_error_value((uint32_t)hd->chan);
+               trace_error_value((uint32_t)hd->dma_buffer);

                if (hd->dma_buffer->avail < local_elem->size) {
                        /* buffer is enough empty, just return. */

```

The log now with the panic is:

> 0xf10 [50.966408]       delta [0.000493]        host cp4 //normal copy
> 0xf20 [50.966410]       delta [0.000002]        value 0x00000000be061300 // hd is 0xbe061300
> 0xf30 [50.966412]       delta [0.000002]        value 0x0000000000000001
> 0xf40 [50.966414]       delta [0.000002]        value 0x00000000be061500
> 0xf50 [50.966908]       delta [0.000493]        host cp4
> 0xf60 [50.966910]       delta [0.000002]        value 0x00000000be061300
> 0xf70 [50.966912]       delta [0.000002]        value 0x0000000000000001
> 0xf80 [50.966914]       delta [0.000002]        value 0x00000000be061500
> 0xf90 [50.967408]       delta [0.000494]        host cp4
> 0xfa0 [50.967410]       delta [0.000002]        value 0x00000000be061300
> 0xfb0 [50.967412]       delta [0.000002]        value 0x0000000000000001
> 0xfc0 [50.967415]       delta [0.000002]        value 0x00000000be061500
> ...
> 0x12a0 [70.729725]      delta [0.000003]        dma DEn
> 0x12b0 [70.729737]      delta [0.000013]        value 0x00000000abbbbeef
> 0x12c0 [70.729739]      delta [0.000002]        value 0x00000000be061300 // **now alloc child is 0xbe061300 too**
> 
>

xiulipan, 2018-09-20:
@lgirdwood @mmaka1 
This seems to be that we have some error with memory alloc. Need to check that what is going wrong with the mem alloc.

xiulipan, 2018-09-20:
@ZhendanYang Test with newest master commit d7404c5
The issue is gone.
@mmaka1 
This issue seems to be fixed by #383. I see that you have reorg the whole mem alloc and maybe this case will not be triggered. But if this is a potential risk?

xiulipan, 2018-09-20:
@ZhendanYang 
Enable **DEBUG_BLOCK_FREE** and can be reproduce on master d7404c5 with every aplay or arecord second run.

```diff
diff --git a/src/lib/alloc.c b/src/lib/alloc.c
index 6ce20dd..1c41a2c 100644
--- a/src/lib/alloc.c
+++ b/src/lib/alloc.c
@@ -44,7 +44,7 @@
 #define DEBUG_BLOCK_ALLOC_VALUE                0x6b6b6b6b

 /* debug to set memory value on every free TODO: not working atm */
-#define DEBUG_BLOCK_FREE               0
+#define DEBUG_BLOCK_FREE               10
 #define DEBUG_BLOCK_FREE_VALUE         0x5a5a5a5a
```

@mmaka1 @lgirdwood 
Try to enable **DEBUG_BLOCK_ALLOC** too. Buf failed at boot up time.
It seems we have some wrong use of mem alloc here. Need to debug later.

closed 2018-09-20 by lgirdwood

mmaka1, 2018-09-21:
> This issue seems to be fixed by #383. I see that you have reorg the whole mem alloc and maybe this case will not be triggered. But if this is a potential risk?
@xiulipan Moved back all system allocations to the sys zone once the heap was partitioned for SMP to enable 'free' on a core power down. Sys aligns mem buffers and ensures the allocation is successful. Btw there is no need for NULL ptr testing there anymore.

xiulipan, 2018-09-25:
@mmaka1 
Will have a full test with alloc and free later to remove all potential risk.
Thanks for the explanation

PR #397 (2018-09-20): host: do not free the list header in host_reset
The CONFIG_DMA_GW only enable with hda_dma, as the create_local_elems is
refined to create elem at every start. Did not need to keep the first
elem.

Use list_for_item_safe to delete all item in list. The old way will
delete and free the list header with belongs to the host_data. Free
the list header will directly free host_data and make the second run
fail if the memory is alloced again.

Fixed #391