thesofproject/sof #394: Stress test: Keeping pause/resume or XRUN injection can causes DSP panic when playback or capture

ZhendanYang, 2018-09-19:
Summary:
Keeping pause/resume or XRUN injection can causes DSP panic when playback or capture.

[   66.816385] sof-audio sof-audio: error : DSP panic!
[   66.816397] sof-audio sof-audio: panic: dsp_oops_offset 788480 offset 788480
[   66.816417] sof-audio sof-audio: status: fw entered - code 00000005
[   66.816485] sof-audio sof-audio: error: out of memory
[   66.816492] sof-audio sof-audio: error: trace point 00004000
[   66.816499] sof-audio sof-audio: error: DSP Firmware Oops
[   66.816546] sof-audio sof-audio: error: Exception Cause: Level1InterruptCause, Level-1 interrupt as indicated by set level-1 bits in the INTERRUPT register
[   66.816555] sof-audio sof-audio: EXCCAUSE 0x00000004 EXCVADDR 0x00000000 PS       0x00060525 SAR     0x00000020
[   66.816564] sof-audio sof-audio: EPC1     0xbe051c9d EPC2     0xbe040d22 EPC3     0xbe040d22 EPC4    0x00000000
[   66.816572] sof-audio sof-audio: EPC5     0xbe040d22 EPC6     0x00000000 EPC7     0x00000000 DEPC    0x00000000
[   66.816580] sof-audio sof-audio: EPS2     0x00060720 EPS3     0x00060720 EPS4     0x00000000 EPS5    0x00060720
[   66.816588] sof-audio sof-audio: EPS6     0x00000000 EPS7     0x00000000 INTENABL 0x000101d8 INTERRU 0x00000222

Error log:
using 19.20MHz timestamp clock
0x0 [174.957896]	delta [174.957896]	mem eM1
0x10 [174.845712]	delta [********]	ipc Cmd
0x20 [174.845715]	delta [0.000004]	ipc tri
0x30 [174.845720]	delta [0.000005]	pipe cmd
0x40 [174.845723]	delta [0.000003]	host trg
0x50 [174.845727]	delta [0.000003]	volume trg
0x60 [174.845730]	delta [0.000003]	dai trg
0x70 [174.845737]	delta [0.000007]	dma Dpr
0x80 [174.845746]	delta [0.000009]	ssp tri
0x90 [174.845749]	delta [0.000003]	ssp sta
0xa0 [174.949818]	delta [0.104069]	ipc Cmd
0xb0 [174.949822]	delta [0.000004]	ipc tri
0xc0 [174.949827]	delta [0.000005]	pipe cmd
0xd0 [174.949830]	delta [0.000003]	host trg
0xe0 [174.949834]	delta [0.000003]	volume trg
0xf0 [174.949837]	delta [0.000003]	dai trg
0x100 [174.949840]	delta [0.000003]	dai tsp
0x110 [174.949843]	delta [0.000003]	dma DDi
0x120 [174.949847]	delta [0.000005]	ssp tri
0x130 [174.950011]	delta [0.000163]	ssp Ss0
0x140 [174.950013]	delta [0.000003]	ssp Ss2
0x150 [174.957866]	delta [0.007852]	ipc Cmd
0x160 [174.957869]	delta [0.000004]	ipc tri
0x170 [174.957874]	delta [0.000005]	pipe cmd
0x180 [174.957877]	delta [0.000003]	host trg
0x190 [174.957881]	delta [0.000003]	volume trg
0x1a0 [174.957884]	delta [0.000003]	dai trg
0x1b0 [174.957891]	delta [0.000007]	dma Dpr
0x1c0 [174.957896]	delta [0.000005]	mem eM1

Reproduce step:
1.aplay -D hw:0,0 -c 2 -r 48000 -f s16_le music.wav -vv -i
2.XRUN injection or keep pressing space key

Environment:
on APL and CNL
sof master: 009447234cf01cbc31ae28841d7a9f702afbc2ff
kernel sof-dev: bc51a6f
soft master: d77beaf

Log:
[rmbox-t-apl.log](https://github.com/thesofproject/sof/files/2396560/rmbox-t-apl.log)
[rmbox-apl.log](https://github.com/thesofproject/sof/files/2396561/rmbox-apl.log)
[dmesg-apl.log](https://github.com/thesofproject/sof/files/2396562/dmesg-apl.log)
[dmesg-cnl.log](https://github.com/thesofproject/sof/files/2396564/dmesg-cnl.log)
[rmbox-t-cnl.log](https://github.com/thesofproject/sof/files/2396565/rmbox-t-cnl.log)
[rmbox-cnl.log](https://github.com/thesofproject/sof/files/2396566/rmbox-cnl.log)

xiulipan, 2018-09-20:
@lgirdwood 
Now the IPC process task is lower into interrupt LV1.
I think this maybe the cause.
I will debug to see if we can use other level interrupt to avoid this issues.

lgirdwood, 2018-09-20:
@xiulipan the error is out of memory according to the log. Changing the schedule priority is a hack

xiulipan, 2018-09-21:
@lgirdwood 
Will check with the DEBUG_BLOCK_FREE/ALLOC with https://github.com/thesofproject/sof/pull/398
I found that we may have some issues with mem alloc.

ZhendanYang, 2018-09-25:
Test with latest code base on APL and CNL, the rmbox shows like this:

0x0 [851251515568.947144]	delta [********]	value 0x2000130c00000000
0x10 [851251531898.770874]	delta [********]	value 0x2000126c00000000
0x20 [851251545767.936035]	delta [********]	value 0x2000433400000000
0x30 [851251558966.012695]	delta [********]	value 0x2000405400000000
0x40 [851251573058.874146]	delta [********]	value 0x20002f4c00000000
0x50 [851251585585.862061]	delta [********]	value 0x2000392800000000
0x60 [851251596770.672607]	delta [********]	value 0x200038bc00000000
0x70 [851251608179.179565]	delta [********]	value 0x20005b9000000000
0x80 [851251627864.446289]	delta [********]	value 0x2000618800000000
0x90 [851252185091.713745]	delta [********]	value 0x200061d000000000
0xa0 [851252196500.220703]	delta [********]	value 0x200061ac00000000
0xb0 [851278147497.929443]	delta [********]	value 0x2000130c00000000
0xc0 [851278163380.360596]	delta [********]	value 0x2000126c00000000
0xd0 [851278177025.829590]	delta [********]	value 0x2000433400000000
0xe0 [851278189552.817505]	delta [********]	value 0x2000405400000000
0xf0 [851278202079.805420]	delta [********]	value 0x20002f4c00000000
0x100 [851278214830.489624]	delta [********]	value 0x2000392800000000
0x110 [851278232502.490479]	delta [********]	value 0x20005be000000000
0x120 [851278271649.327881]	delta [********]	value 0x2000618800000000
0x130 [851278286189.581665]	delta [********]	value 0x2000621800000000
0x140 [851634178364.538940]	delta [********]	value 0x2000130c00000000
0x150 [851634194023.273804]	delta [********]	value 0x2000126c00000000
0x160 [851634207892.439087]	delta [********]	value 0x2000433400000000
0x170 [851634221090.515747]	delta [********]	value 0x2000405400000000
0x180 [851634235183.377197]	delta [********]	value 0x20002f4c00000000
0x190 [851634247710.365112]	delta [********]	value 0x2000392800000000
0x1a0 [851634258895.175659]	delta [********]	value 0x200038bc00000000
0x1b0 [851634270303.682617]	delta [********]	value 0x20005b9000000000

Test env:
sof: master 5450a2fcd8b4092bb329afa16205f82c0b5057ca
soft: 15030d4
kernel: sof-dev b5cde88

xiulipan, 2018-09-25:
@ZhendanYang 
Any error in terminal or dmesg?
This seems to be some rmbox garbage.

ZhendanYang, 2018-10-19:
This issue can't reproduce now. Should be fixed by #495 .

closed 2018-10-19 by ZhendanYang

mengdonglin, 2018-10-20:
@keqiaozhang It seems you can reproduce this issue on GLK on Oct 18. C. If you can still reproduce on GLK, please reopen it and remove APL/CNL label.