thesofproject/sof #408: cmocka: missing tests for _trace_event_mbox_atomic

lgirdwood, 2018-09-21:
```
12:32:01,454 INFO  - mv -f $depbase.Tpo $depbase.Po
12:32:01,454 INFO  - depbase=`echo src/math/numbers/norm_int32.o | sed 's|[^/]*$|.deps/&|;s|\.o$||'`;\
12:32:01,454 INFO  - xt-xcc -DHAVE_CONFIG_H -I. -I../../src/include -fno-inline-functions -mlongcalls -I./include -I/localdisk/processteam/qba1/workspace/24733/cmocka/include -I../../src/arch/xtensa/include -I /quickbuild/workspace1/24733/SOF_FW/src/arch/xtensa/include -I /quickbuild/workspace1/24733/SOF_FW/src/arch/xtensa/up/include -I /quickbuild/workspace1/24733/SOF_FW/src/arch/xtensa/up/xtos -I /quickbuild/workspace1/24733/SOF_FW/src/platform/apollolake/include -I /quickbuild/workspace1/24733/SOF_FW/src/platform/intel/include -I /localdisk/tools/xtensa/RG-2017.8-linux/X4H3I16w2D48w3a_2017_8/xtensa-elf/arch/include -I../../src/audio -I /quickbuild/workspace1/24733/SOF_FW/src/include -I /localdisk/tools/xtensa/RG-2017.8-linux/X4H3I16w2D48w3a_2017_8/xtensa-elf/include -O2 -g -Wall -Werror -Wl,-EL -Wmissing-prototypes -MT src/math/numbers/norm_int32.o -MD -MP -MF $depbase.Tpo -c -o src/math/numbers/norm_int32.o src/math/numbers/norm_int32.c &&\
12:32:01,454 INFO  - mv -f $depbase.Tpo $depbase.Po
12:32:02,607 INFO  - cc1: warnings being treated as errors
12:32:02,608 INFO  - src/audio/component/mock.c:38: warning: no previous prototype for ‘_trace_event_mbox_atomic’
12:32:02,610 INFO  - Makefile:1200: recipe for target 'src/audio/component/mock.o' failed
12:32:02,610 INFO  - make[3]: *** [src/audio/component/mock.o] Error 2
12:32:02,610 INFO  - make[3]: *** Waiting for unfinished jobs....
12:32:02,680 INFO  - cc1: warnings being treated as errors
12:32:02,680 INFO  - src/audio/buffer/mock.c:37: warning: no previous prototype for ‘_trace_event’
12:32:02,680 INFO  - src/audio/buffer/mock.c:42: warning: no previous prototype for ‘_trace_event_mbox_atomic’
12:32:02,681 INFO  - Makefile:1200: recipe for target 'src/audio/buffer/mock.o' failed
12:32:02,681 INFO  - make[3]: *** [src/audio/buffer/mock.o] Error 2
12:32:02,711 INFO  - cc1: warnings being treated as errors
12:32:02,712 INFO  - src/audio/pipeline/pipeline_mocks.c:83: warning: no previous prototype for ‘_trace_event’
12:32:02,712 INFO  - src/audio/pipeline/pipeline_mocks.c:88: warning: no previous prototype for ‘_trace_event_mbox_atomic’
12:32:02,713 INFO  - Makefile:1200: recipe for target 'src/audio/pipeline/pipeline_mocks.o' failed
```

jajanusz, 2018-09-24:
@akloniex 
caused by https://github.com/thesofproject/sof/pull/390
please check UTs before making PR next time

mengdonglin, 2018-09-29:
@jajanusz @lrgirdwo Is this issue fixed by #410? Can we close it.

closed 2018-10-01 by lgirdwood