thesofproject/sof #412: DMA trace shows garbage

ZhendanYang, 2018-09-25:
Summary:
DMA trace shows garbage after system booted.

0x0 [370751867.016533]	delta [******** ]	value 0x200060e800000000
0x10 [370966167.988907]	delta [******** ]	value 0x2000602000000000
0x20 [371134387.541333]	delta [******** ]	value 0x200060c000000000
0x30 [385518725.147307]	delta [******** ]	value 0x2000130c00000000
0x40 [385535949.755733]	delta [******** ]	value 0x200010dc00000000
0x50 [385556082.414933]	delta [******** ]	value 0x2000403000000000
0x60 [386168338.950827]	delta [******** ]	value 0x2000130c00000000
0x70 [386185116.166827]	delta [******** ]	value 0x200010dc00000000
0x80 [386201445.990400]	delta [******** ]	value 0x2000321400000000
0x90 [386618415.732053]	delta [******** ]	value 0x2000130c00000000
0xa0 [386635192.948053]	delta [******** ]	value 0x200010b400000000
0xb0 [386651299.075413]	delta [******** ]	value 0x20004d0c00000000
0xc0 [387390391.364267]	delta [******** ]	value 0x2000130c00000000
0xd0 [387406050.099200]	delta [******** ]	value 0x200010b400000000
0xe0 [387421485.137920]	delta [******** ]	value 0x20004d0c00000000
0xf0 [387959250.834773]	delta [******** ]	value 0x2000130c00000000

Reproduce steps:
1. boot
2. sudo rmbox -t

Test env:
sof: master 5450a2fcd8b4092bb329afa16205f82c0b5057ca
soft: 15030d4
kernel: sof-dev b5cde88

I test with commit f79d45c95e7fc8b79848193b802145f1d100c36d and there is no such issue. But commit 4a61c9566f8e4c62d84ffdae327665b91a757b3f has this issue. So it's probably caused by #390 

Log:
[rmbox-t-garbage.log](https://github.com/thesofproject/sof/files/2414055/rmbox-t-garbage.log)

lgirdwood, 2018-09-25:
@bardliao is working on update for tracing that will also fix ABI update

xiulipan, 2018-09-26:
@lgirdwood 
I asked @akloniex that we may missing some rmbox update to support the new format.
close this issue
move to SOFT https://github.com/thesofproject/soft/issues/75

closed 2018-09-26 by xiulipan