thesofproject/sof #425: Firmware build fail

ZhendanYang, 2018-09-27:
Summary:
Firmware build failed for BYT and CNL.

work.c: In function 'work_new_queue':
work.c:501:19: error: 'PLATFORM_WORKQ_DEFAULT_TIMEOUT' undeclared (first use in this function)
  queue->timeout = PLATFORM_WORKQ_DEFAULT_TIMEOUT;
                   ^
work.c:501:19: note: each undeclared identifier is reported only once for each function it appears in
Makefile:486: recipe for target 'libcore_a-work.o' failed
make[2]: *** [libcore_a-work.o] Error 1
make[2]: Leaving directory '/home/yangzhendan/workspace/reef/sof/src/lib'
Makefile:383: recipe for target 'all-recursive' failed
make[1]: *** [all-recursive] Error 1
make[1]: Leaving directory '/home/yangzhendan/workspace/reef/sof/src'
Makefile:436: recipe for target 'all-recursive' failed
make: *** [all-recursive] Error 1

Test env:
sof master: debe638c97b462fc45ff2a93cd6efcc00145f746
I test commit f9d11688453cb89476ec73c44922bc0021ebbe67, it's a good point.
But commit 119273d6ef39bb54a59fd346e027cb8f976c62b3 is a bad point.
So it should be caused by PR #421

xiulipan, 2018-09-27:
#421 CI build Result.

> SOF Building Result
> Platform: host FAILED
> 
> Error message:
> topology.c:196:33: error: ‘SND_SOC_TPLG_DAPM_SCHEDULER’ undeclared (first use in this function)
> temp_comp_list[i].type == SND_SOC_TPLG_DAPM_SCHEDULER)
> ...More
> Platform: baytrail FAILED
> 
> Error message:
> work.c:501:19: error: 'PLATFORM_WORKQ_DEFAULT_TIMEOUT' undeclared (first use in this function)
> queue->timeout = PLATFORM_WORKQ_DEFAULT_TIMEOUT;
> ...More
> Platform: cannonlake FAILED
> 
> Error message:
> work.c:501:19: error: 'PLATFORM_WORKQ_DEFAULT_TIMEOUT' undeclared (first use in this function)
> queue->timeout = PLATFORM_WORKQ_DEFAULT_TIMEOUT;
> ...More
> Platform: apollolake PASSED

xiulipan, 2018-09-27:
close vi e3108d653fc1c9a9d19283e4ef9328ccd5efefab

closed 2018-09-27 by xiulipan