thesofproject/sof #432: Playback and capture can't work when pulseaudio is enabled

ZhendanYang, 2018-09-28:
Summary:
On BYT, when enabled pulseaudio, playback and capture can't work. There are a lot of ipc error in dmesg log.

[  129.111820] sof-audio sof-audio: error: ipc error for 0x60010000 size 0x14
[  129.111836] sof-audio sof-audio: ASoC: sof-audio hw params failed: -22
[  129.111844] sof-audio sof-audio: pcm: free stream 0 dir 0
[  129.111858] sof-audio sof-audio: ipc tx: 0x60030000
[  129.111913] sof-audio sof-audio: ipc tx succeeded: 0x60030000
[  129.111925]  Passthrough: ASoC: hw_params FE failed -22

Reproduce step:
zql@zql-MB-T:~$ aplay -D hw:0,0 -c 2 -r 48000 -f s16_le Music/48California_Dreaming.wav -vv -i
Playing WAVE 'Music/48California_Dreaming.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Stereo
aplay: set_params:1363: Unable to install hw params:
ACCESS:  RW_INTERLEAVED
FORMAT:  S16_LE
SUBFORMAT:  STD
SAMPLE_BITS: 16
FRAME_BITS: 32
CHANNELS: 2
RATE: 48000
PERIOD_TIME: 84000
PERIOD_SIZE: 4032
PERIOD_BYTES: 16128
PERIODS: (4 5)
BUFFER_TIME: 340000
BUFFER_SIZE: 16320
BUFFER_BYTES: 65280
TICK_TIME: 0

Test env:
sof master: 299ccbce583bc839e21b8c396bd6f067ea26e364
sof tool: 60c3937
kernel: sof-dev 51ebd5d
tplg: test-ssp2-mclk-0-I2S-volume-s16le-s16le-48k-19200k-codec.tplg

Log:
[dmesg.log](https://github.com/thesofproject/sof/files/2427011/dmesg.log)
rmbox is empty and rmbox -t is filled with garbage.

xiulipan, 2018-09-28:
blocked by trace issues https://github.com/thesofproject/soft/issues/75
No debug tools now

mengdonglin, 2018-09-29:
@ZhendanYang Could you use the new logger with master branch from now on?
If it's working, we can remove the blocker.

mengdonglin, 2018-10-19:
@ZhendanYang Is the vailidation blocked by https://github.com/thesofproject/sof/issues/443 ? I think the new logger is more health now.

mengdonglin, 2018-11-22:
The test topology files are moved to: https://github.com/thesofproject/soft/tree/master/test/topology

markyang, 2018-11-23:
**Summary:**
  This issue can be reproduced on BYT with ALC5651.

**Test env:**
sof master: e8ef1f21
sof tool:   ad3f3e4f
kernel sof-dev: 4a3aa5f3
tplg: test-ssp2-mclk-0-I2S-volume-s16le-s16le-48k-19200k-codec.tplg-ad3f3e4f

**Log:**
[logger-byt.log](https://github.com/thesofproject/sof/files/2609844/logger-byt.log)
[logger-byt-t.log](https://github.com/thesofproject/sof/files/2609847/logger-byt-t.log)

mengdonglin, 2018-11-23:
@markyang Is Pulseaudio (PA) enabled in your test? Could you help to disable PA and check if this issue can be reproduced?

libinyang, 2018-11-26:
The bug is caused that pulseaudio will open/close pcm so frequently and the channels are not released, which will cause the channel is not enough.

markyang, 2018-11-26:
**Summary:**
  This issue cannot be reproduced on BYT w/ ALC5651.

Test env:
sof master: 6e14892
sof tool: ad3f3e4
kernel sof-dev: aff80d74
tplg: test-ssp2-mclk-0-I2S-volume-s16le-s16le-48k-19200k-codec.tplg-ad3f3e4 and 
        test-ssp2-mclk-0-I2S-volume-s24le-s24le-48k-19200k-codec.tplg-ad3f3e4

libinyang, 2018-11-27:
patch is submitted:
https://github.com/thesofproject/sof/pull/633

libinyang, 2018-11-28:
@markyang 
The patch is merged.

markyang, 2018-11-28:
Summary:
This issue cannot be reproduced on BYT w/ ALC5651 anymore.
fixed by PR #633

closed 2018-11-28 by markyang