thesofproject/sof #433: HDA analog capture with all 0s

keyonjie, 2018-09-28:
I tested with glk-003-drop-stable branch(commit 22fa01d059589) on my LeafHill, arecord can run with HDA analog codec device, but all captured PCM data are 0s.

@mmaka1 can you help on this?

mengdonglin, 2018-09-28:
@RanderWang Could you help to share if capture can work on WHL after you enables HD-A on WHL? Thanks

RanderWang, 2018-09-29:
@mengdonglin  yes, all zero. Thank you for your info and I don't have to debug kernel

RanderWang, 2018-10-16:
@lgirdwood 

Update status:
Janusz tested the FW on windows with loop-back mode,  found some noise in capture data.
I tested it on Linux, only zero data is gotten in FW and HOST.  I checked the output of codec
with an analyzer, the data is sent by codec to DSP. So I think it is very possible that Host
driver sets some HDA register wrongly

lgirdwood, 2018-10-16:
@RanderWang ok to confirm 

1) Codec sends PCM data to host on HDA bus validated by logic analyser.
2) Host only reads back PCM data as 0s.

So can you check 

3) FW link DMA buffer contents, just add some trace to dump first samples in period on every 10th copy. This will show if FW is reading back 0s or not.

4) If FW reads 0s, please check pinmux settings for HDA Rx pin.

@mmaka1 any ideas at your end if FW reads 0s in link DMA buffer ?

RanderWang, 2018-10-17:
> @RanderWang ok to confirm
> 
>     1. Codec sends PCM data to host on HDA bus validated by logic analyser.
> 
>     2. Host only reads back PCM data as 0s.
> 
> 
> So can you check
> 
>     1. FW link DMA buffer contents, just add some trace to dump first samples in period on every 10th copy. This will show if FW is reading back 0s or not.
> 
>     2. If FW reads 0s, please check pinmux settings for HDA Rx pin.
> 
> 
> @mmaka1 and @tlauda any ideas at your end if FW reads 0s in link DMA buffer ?

I confirm 1 & 2. I also dump data in FW, all zero.

RanderWang, 2018-10-17:
@lgirdwood 

      I also tried legacy HDA, both playback & capture work.  So I think pin setting is correct

RanderWang, 2018-10-17:
I also  checked PPLC registers with Janusz, all are correct. Now I ask Janusz for help to dump other registers on Windows.

mengdonglin, 2018-10-19:
PR https://github.com/thesofproject/linux/pull/194 is to fix this issue.
@RanderWang Maybe you can use the magic words "fixes #433" in your PR to automatically close this issue when your PR is merged?

mengdonglin, 2018-11-09:
@RanderWang Is the issue fixed or need any other PR?

Jocelyn-Li, 2018-11-24:
https://github.com/thesofproject/linux/pull/291 is to fix this issue.

Jocelyn-Li, 2018-12-07:
@RanderWang can this be closed as thesofproject/linux#291 was merged?

mengdonglin, 2018-12-17:
Closing this issue since it's fixed by thesofproject/linux#291.

mengdonglin, 2018-12-17:
Will revisit HDA capture on WHL when #700 is fixed.

RanderWang, 2019-03-05:
fixed

closed 2019-03-05 by RanderWang