thesofproject/sof #443: DSP panic after system booted

ZhendanYang, 2018-09-29:
Summary:
DSP panic happened after system booted.

[    8.142381] sof-audio sof-audio: error : DSP panic!
[    8.142451] sof-audio sof-audio: panic: dsp_oops_offset 1329152 offset 1329152
[    8.142484] sof-audio sof-audio: error: runtime exception
[    8.142544] sof-audio sof-audio: error: trace point 90010000
[    8.142605] sof-audio sof-audio: error: panic happen at :0
[    8.142665] sof-audio sof-audio: error: DSP Firmware Oops
[    8.142724] sof-audio sof-audio: error: Exception Cause: IllegalInstructionCause, Illegal instruction
[    8.142821] sof-audio sof-audio: EXCCAUSE 0x00000000 EXCVADDR 0x00000000 PS       0x00070520 SAR     0x0000001d
[    8.142926] sof-audio sof-audio: EPC1     0xff2c099b EPC2     0x00000000 EPC3     0xff2c0edf EPC4    0xff2c0edf
[    8.143030] sof-audio sof-audio: EPC5     0x00000000 EPC6     0x00000000 EPC7     0x00000000 DEPC    0x00000000
[    8.143134] sof-audio sof-audio: EPS2     0x00000000 EPS3     0x00060720 EPS4     0x00060720 EPS5    0x00000000
[    8.143238] sof-audio sof-audio: EPS6     0x00000000 EPS7     0x00000000 INTENABL 0x00078f40 INTERRU 0x000000a2
[    8.143342] sof-audio sof-audio: stack dump from 0xff327bc0
[    8.143403] sof-audio sof-audio: 0xff327bc0: 0x02480002 0x00060130 0x0000001d 0xff2c0f69
[    8.143488] sof-audio sof-audio: 0xff327bc4: 0x00000000 0xff309180 0xff306a8c 0x02480000
[    8.143574] sof-audio sof-audio: 0xff327bc8: 0x00000000 0x00000000 0xff2d0832 0xff2d0848
[    8.143660] sof-audio sof-audio: 0xff327bcc: 0x00000000 0x00000000 0x00000000 0x00000000
[    8.143745] sof-audio sof-audio: 0xff327bd0: 0xff319200 0xff319284 0xff327d40 0xff319284
[    8.143831] sof-audio sof-audio: 0xff327bd4: 0xbf2c1f91 0xff327c60 0xff327c80 0x00000010
[    8.143916] sof-audio sof-audio: 0xff327bd8: 0xbf2c408d 0xff327c70 0x00000002 0x00000001
[    8.144002] sof-audio sof-audio: 0xff327bdc: 0xbf2cb89c 0xff327c80 0x20003e48 0x00000318
[    8.417462] intel_soc_dts_thermal: request_threaded_irq ret -22

Test env:
sof master: a97cd4f
sof tool: 918e401
kernel sof-dev: 1f6c9cf
tplg: test-ssp2-mclk-0-I2S-volume-s16le-s16le-48k-19200k-codec.tplg

Log:
[dmesg-panic.log](https://github.com/thesofproject/sof/files/2430366/dmesg-panic.log)

ZhendanYang, 2018-10-08:
Summary:
DSP panic happened after system booted.

[    5.736743] sof-audio sof-audio: error : DSP panic!
[    5.736752] sof-audio sof-audio: panic: dsp_oops_offset 788480 offset 788480
[    5.736756] sof-audio sof-audio: status: fw entered - code 00000005
[    5.736778] sof-audio sof-audio: error: runtime exception
[    5.736780] sof-audio sof-audio: error: trace point 00004000
[    5.736782] sof-audio sof-audio: error: panic happen at :0
[    5.736785] sof-audio sof-audio: error: DSP Firmware Oops
[    5.736788] sof-audio sof-audio: error: Exception Cause: LoadProhibitedCause, A load referenced a page mapped with an attribute that does not permit loads
[    5.736791] sof-audio sof-audio: EXCCAUSE 0x0000001c EXCVADDR 0xfffffff8 PS       0x00060020 SAR     0x0000001d
[    5.736794] sof-audio sof-audio: EPC1     0xbe015a93 EPC2     0xbe0118f6 EPC3     0xbe013a48 EPC4    0x00000000
[    5.736796] sof-audio sof-audio: EPC5     0x00000000 EPC6     0x00000000 EPC7     0x00000000 DEPC    0x00000000
[    5.736799] sof-audio sof-audio: EPS2     0x00060720 EPS3     0x00060722 EPS4     0x00000000 EPS5    0x00000000
[    5.736802] sof-audio sof-audio: EPS6     0x00000000 EPS7     0x00000000 INTENABL 0x000101d0 INTERRU 0x00000222
[    5.736804] sof-audio sof-audio: stack dump from 0xbe07fd20

be015a93:       7c1852                         l16ui   a5, a8, 248
                        case COMP_TYPE_COMPONENT:
                                   if (icd->cd->comp.id == id)

Test env:
sof-master: aeb18eefd9a4a7b03b6159bca3427031e254db01
sof tool: 2d7a933
kernel sof-dev: 93853cb
tplg: sof-apl-nocodec.tplg

Log:
rmbox and rmbox -t are both empty.
[dmesg-apl.log](https://github.com/thesofproject/sof/files/2454907/dmesg-apl.log)

lgirdwood, 2018-10-08:
@ZhendanYang adding a little more context shows it's related to trace init.
```
[    8.140565] sof-audio sof-audio: booting DSP firmware
[    8.142027] sof-audio sof-audio: ipc rx: 0x70000000
[    8.142032] sof-audio sof-audio: ipc: DSP is ready 0x70000000 offset 0x144000
[    8.142043] sof-audio sof-audio:  Firmware info: version 1:1-a97cd build 32 on Sep 28 2018:21:57:53
[    8.142144] sof-audio sof-audio: found ext header type 1 size 0x88
[    8.142180] sof-audio sof-audio:  mailbox upstream 0x144000 - size 0x400
[    8.142183] sof-audio sof-audio:  mailbox downstream 0x144400 - size 0x400
[    8.142186] sof-audio sof-audio:  stream region 0x144a00 - size 0x200
[    8.142188] sof-audio sof-audio: booting DSP firmware completed
[    8.142197] sof-audio sof-audio: ipc rx done: 0x70000000
[    8.142209] sof-audio sof-audio: firmware boot complete
[    8.142266] sof-audio sof-audio: generating page table for (____ptrval____) size 0x10000 pages 16
[    8.142269] sof-audio sof-audio: pfn i 0 idx 0 pfn 6fa50
[    8.142272] sof-audio sof-audio: pfn i 1 idx 2 pfn 6fa51
[    8.142275] sof-audio sof-audio: pfn i 2 idx 5 pfn 6fa52
[    8.142277] sof-audio sof-audio: pfn i 3 idx 7 pfn 6fa53
[    8.142280] sof-audio sof-audio: pfn i 4 idx 10 pfn 6fa54
[    8.142283] sof-audio sof-audio: pfn i 5 idx 12 pfn 6fa55
[    8.142286] sof-audio sof-audio: pfn i 6 idx 15 pfn 6fa56
[    8.142288] sof-audio sof-audio: pfn i 7 idx 17 pfn 6fa57
[    8.142291] sof-audio sof-audio: pfn i 8 idx 20 pfn 6fa58
[    8.142294] sof-audio sof-audio: pfn i 9 idx 22 pfn 6fa59
[    8.142297] sof-audio sof-audio: pfn i 10 idx 25 pfn 6fa5a
[    8.142299] sof-audio sof-audio: pfn i 11 idx 27 pfn 6fa5b
[    8.142302] sof-audio sof-audio: pfn i 12 idx 30 pfn 6fa5c
[    8.142305] sof-audio sof-audio: pfn i 13 idx 32 pfn 6fa5d
[    8.142308] sof-audio sof-audio: pfn i 14 idx 35 pfn 6fa5e
[    8.142310] sof-audio sof-audio: pfn i 15 idx 37 pfn 6fa5f
[    8.142313] sof-audio sof-audio: dma_trace_pages: 16
[    8.142321] sof-audio sof-audio: stream_tag: -30560
[    8.142342] sof-audio sof-audio: ipc tx: 0x90010000
[    8.142381] sof-audio sof-audio: error : DSP panic!
[    8.142451] sof-audio sof-audio: panic: dsp_oops_offset 1329152 offset 1329152
[    8.142484] sof-audio sof-audio: error: runtime exception
[    8.142544] sof-audio sof-audio: error: trace point 90010000
[    8.142605] sof-audio sof-audio: error: panic happen at :0
[    8.142665] sof-audio sof-audio: error: DSP Firmware Oops
[    8.142724] sof-audio sof-audio: error: Exception Cause: IllegalInstructionCause, Illegal instruction
[    8.142821] sof-audio sof-audio: EXCCAUSE 0x00000000 EXCVADDR 0x00000000 PS       0x00070520 SAR     0x0000001d
[    8.142926] sof-audio sof-audio: EPC1     0xff2c099b EPC2     0x00000000 EPC3     0xff2c0edf EPC4    0xff2c0edf
[    8.143030] sof-audio sof-audio: EPC5     0x00000000 EPC6     0x00000000 EPC7     0x00000000 DEPC    0x00000000
[    8.143134] sof-audio sof-audio: EPS2     0x00000000 EPS3     0x00060720 EPS4     0x00060720 EPS5    0x00000000
[    8.143238] sof-audio sof-audio: EPS6     0x00000000 EPS7     0x00000000 INTENABL 0x00078f40 INTERRU 0x000000a2
[    8.143342] sof-audio sof-audio: stack dump from 0xff327bc0
[    8.143403] sof-audio sof-audio: 0xff327bc0: 0x02480002 0x00060130 0x0000001d 0xff2c0f69
[    8.143488] sof-audio sof-audio: 0xff327bc4: 0x00000000 0xff309180 0xff306a8c 0x02480000
[    8.143574] sof-audio sof-audio: 0xff327bc8: 0x00000000 0x00000000 0xff2d0832 0xff2d0848
[    8.143660] sof-audio sof-audio: 0xff327bcc: 0x00000000 0x00000000 0x00000000 0x00000000
[    8.143745] sof-audio sof-audio: 0xff327bd0: 0xff319200 0xff319284 0xff327d40 0xff319284
[    8.143831] sof-audio sof-audio: 0xff327bd4: 0xbf2c1f91 0xff327c60 0xff327c80 0x00000010
[    8.143916] sof-audio sof-audio: 0xff327bd8: 0xbf2c408d 0xff327c70 0x00000002 0x00000001
[    8.144002] sof-audio sof-audio: 0xff327bdc: 0xbf2cb89c 0xff327c80 0x20003e48 0x00000318
[    8.417462] intel_soc_dts_thermal: request_threaded_irq ret -22
[    8.440611] intel_rapl: Found RAPL domain package
[    8.440615] intel_rapl: Found RAPL domain core
[    8.448103] sof-audio sof-audio: error: ipc timed out for 0x90010000 size
```

lgirdwood, 2018-10-08:
and on APL it is a topology loading issue 
```
[    5.736618] sof-audio sof-audio: tplg: pipeline id 8 comp 47 scheduling comp id 46
[    5.736622] sof-audio sof-audio: pipeline PIPELINE.8.SSP3.IN: deadline 1000 pri 0 mips 5000 core 0 frames 48
[    5.736637] sof-audio sof-audio: ipc tx: 0x30100000
[    5.736743] sof-audio sof-audio: error : DSP panic!
[    5.736752] sof-audio sof-audio: panic: dsp_oops_offset 788480 offset 788480
[    5.736756] sof-audio sof-audio: status: fw entered - code 00000005
[    5.736778] sof-audio sof-audio: error: runtime exception
[    5.736780] sof-audio sof-audio: error: trace point 00004000
[    5.736782] sof-audio sof-audio: error: panic happen at :0
[    5.736785] sof-audio sof-audio: error: DSP Firmware Oops
[    5.736788] sof-audio sof-audio: error: Exception Cause: LoadProhibitedCause, A load referenced a page mapped with an attribute that does not permit loads
[    5.736791] sof-audio sof-audio: EXCCAUSE 0x0000001c EXCVADDR 0xfffffff8 PS       0x00060020 SAR     0x0000001d
[    5.736794] sof-audio sof-audio: EPC1     0xbe015a93 EPC2     0xbe0118f6 EPC3     0xbe013a48 EPC4    0x00000000
[    5.736796] sof-audio sof-audio: EPC5     0x00000000 EPC6     0x00000000 EPC7     0x00000000 DEPC    0x00000000
[    5.736799] sof-audio sof-audio: EPS2     0x00060720 EPS3     0x00060722 EPS4     0x00000000 EPS5    0x00000000
[    5.736802] sof-audio sof-audio: EPS6     0x00000000 EPS7     0x00000000 INTENABL 0x000101d0 INTERRU 0x00000222
[    5.736804] sof-audio sof-audio: stack dump from 0xbe07fd20
[    5.736807] sof-audio sof-audio: 0xbe07fd20: 0xbe015a93 0x00060d30 0x0000001d 0x00000180
[    5.736810] sof-audio sof-audio: 0xbe07fd24: 0xbe04db80 0x0000002f 0x9e050368 0x9e050368
[    5.736812] sof-audio sof-audio: 0xbe07fd28: 0x0000001c 0xbe015f70 0xbe015a90 0xbe015aae
[    5.736815] sof-audio sof-audio: 0xbe07fd2c: 0x00000002 0x00060d20 0xbe042048 0x20004eb8
[    5.736818] sof-audio sof-audio: 0xbe07fd30: 0x00000000 0x001d72fc 0x00000000 0xbe05b400
[    5.736821] sof-audio sof-audio: 0xbe07fd34: 0x00000008 0xbe07fdc0 0x07530000 0x00000000
[    5.736824] sof-audio sof-audio: 0xbe07fd38: 0xffffffff 0xbe014117 0xbe0141ac 0xbe0141b9
[    5.736826] sof-audio sof-audio: 0xbe07fd3c: 0x00000000 0x001d72fc 0x00000000 0x2000126c
[    5.736829] sof-audio sof-audio: error: waking up any trace sleepers
[    5.778036] random: crng init done
[    5.778041] random: 7 urandom warning(s) missed due to ratelimiting
[    6.048106] sof-audio sof-audio: error: ipc timed out for 0x30100000 size 0x30
```
So this issue is really 2 issues, one for trace init on BYT and the other for topology on APL.

ZhendanYang, 2018-10-09:
Add BYT DSP panic info: 

  942 static inline uint32_t arch_interrupt_global_disable(void)
  943 {
  944         uint32_t flags;
  945 
  946         asm volatile("rsil      %0, 5"
  947 **ff2c099b**:       006590                  rsil    a9, 5
  948 {
  949         struct sof_ipc_dsp_oops_xtensa *x =
  950                 (struct sof_ipc_dsp_oops_xtensa *) mailbox_get_exception_base();
  951 
  952         /* Exception Vector number - 0x0 */
  953         __asm__ __volatile__ ("rsr %0, EXCCAUSE" : "=a" (x->exccause) : : "memory");
  954 ff2c099e:       03e8a0                  rsr.exccause    a10
  955 ff2c09a1:       fff681                  l32r    a8, ff2c097c <_irq_task+0x8c>
  956 ff2c09a4:       0068a2                  s32i    a10, a8, 0

ZhendanYang, 2018-10-15:
Another Exception Cause on UP2:

[    6.132460] sof-audio sof-audio: ipc tx: 0x80010000
[    6.132735] sof-audio sof-audio: error : DSP panic!
[    6.132742] sof-audio sof-audio: panic: dsp_oops_offset 788480 offset 788480
[    6.132745] sof-audio sof-audio: status: fw entered - code 00000005
[    6.132766] sof-audio sof-audio: error: runtime exception
[    6.132769] sof-audio sof-audio: error: trace point 00004000
[    6.132771] sof-audio sof-audio: error: panic happen at :0
[    6.132774] sof-audio sof-audio: error: DSP Firmware Oops
[    6.132776] sof-audio sof-audio: error: Exception Cause: LoadStorePIFDataErrorCause, Synchronous PIF data error during LoadStore access
[    6.132779] sof-audio sof-audio: EXCCAUSE 0x0000000d EXCVADDR 0x0000039c PS       0x00060020 SAR     0x00000001
[    6.132782] sof-audio sof-audio: EPC1     0xbe0166d7 EPC2     0xbe014f4e EPC3     0xbe013d48 EPC4    0x00000000
[    6.132785] sof-audio sof-audio: EPC5     0x00000000 EPC6     0x00000000 EPC7     0x00000000 DEPC    0x00000000
[    6.132787] sof-audio sof-audio: EPS2     0x00060d20 EPS3     0x00060722 EPS4     0x00000000 EPS5    0x00000000
[    6.132790] sof-audio sof-audio: EPS6     0x00000000 EPS7     0x00000000 INTENABL 0x000101d0 INTERRU 0x00000222

                         /* make sure we only config DAI comps */
                         switch (icd->cd->comp.type) {
 be0166d4:       3f29a2                  l32i    a10, a9, 252
 **be0166d7**:       272a92                  l32i    a9, a10, 156
 be0166da:       022926                  beqi    a9, 2, be0166e0 <ipc_comp_dai_config+0x44>
 be0166dd:       db4966                  bnei    a9, 4, be0166bc <ipc_comp_dai_config+0x20>
 
                                 /*
                                  * set config if comp dai_index matches
                                  * config dai_index.
                                  */
                                 if (dai->dai_index == config->dai_index &&

Test env:
sof-master: 1024e94
soft-master: c0a881a
kernel sof-dev: 93853c
tplg: sof-apl-nocodec.tplg

mengdonglin, 2018-10-17:
@keqiaozhang Could you share more analysis info why this is a kernel driver issue?

RanderWang, 2018-10-19:
I tested latest master FW and kernel with had topology, no such issue

RanderWang, 2018-10-19:
sorry, with PR:https://github.com/thesofproject/linux/pull/194 and https://github.com/thesofproject/linux/pull/192

ZhendanYang, 2018-10-19:
I test latest master FW and kernel with PR 194 and 192 on up2 and minnowboard. This issue still exist.

lgirdwood, 2018-10-19:
@RanderWang yep, I see it on up^2 too - looks like something with the topology loading.

mengdonglin, 2018-10-19:
@ranj063 Would you please share your finding on kernel side?
If you think this is a kernel issue, we can follow up here or better move it to Linux repo.

mengdonglin, 2018-10-31:
we need to make logger work on DSP panic to debug this issue.  Please track logger fix on https://github.com/thesofproject/linux/issues/233

mengdonglin, 2018-11-01:
@libinyang please check if https://github.com/thesofproject/sof/pull/518 improve the logger for your debugging.

mengdonglin, 2018-11-01:
@libinyang https://github.com/thesofproject/linux/pull/237 is to fix the logger failure

libinyang, 2018-11-01:
The bug can still be found:
sof-master: 43da169
sof tool: 918e401
kernel sof-dev: 2af9599a + thesofproject/linux#237
tplg: sof-apl-nocodec.tplg

[sof-logger.log](https://github.com/thesofproject/sof/files/2537551/sof-logger.log)

[sof-logger-t.log](https://github.com/thesofproject/sof/files/2537557/sof-logger-t.log)

[dmesg.log](https://github.com/thesofproject/sof/files/2537558/dmesg.log)

markyang, 2018-11-01:
**Summary:**
DSP panic also exists on UP^2 and on BYT.
both _sof-logger_ and _sof-logger -t_ are empty.
**dmesg:**
[   19.337364] sof-audio sof-audio: error : DSP panic!
[   19.337430] sof-audio sof-audio: panic: dsp_oops_offset 1329152 offset 1329152
[   19.337463] sof-audio sof-audio: error: runtime exception
[   19.337522] sof-audio sof-audio: error: trace point 90010000
[   19.337583] sof-audio sof-audio: error: panic happen at :0
[   19.337642] sof-audio sof-audio: error: DSP Firmware Oops
[   19.337700] sof-audio sof-audio: error: Exception Cause: IllegalInstructionCause, Illegal instruction
[   19.337795] sof-audio sof-audio: EXCCAUSE 0x00000000 EXCVADDR 0x00000000 PS       0x00070520 SAR     0x0000001d
[   19.337897] sof-audio sof-audio: EPC1     0xff2c0983 EPC2     0x00000000 EPC3     0xff2c0f03 EPC4    0xff2c0f03
[   19.337999] sof-audio sof-audio: EPC5     0xff2c57e4 EPC6     0x00000000 EPC7     0x00000000 DEPC    0x00000000
[   19.338101] sof-audio sof-audio: EPS2     0x00000000 EPS3     0x00060720 EPS4     0x00060720 EPS5    0x00070024
[   19.338203] sof-audio sof-audio: EPS6     0x00000000 EPS7     0x00000000 INTENABL 0x00078f40 INTERRU 0x000000a2
[   19.338304] sof-audio sof-audio: stack dump from 0xff327bc0
[   19.338365] sof-audio sof-audio: 0xff327bc0: 0x5000b84c 0x00060130 0x0000001d 0xff306900
[   19.338448] sof-audio sof-audio: 0xff327bc4: 0x5000b848 0xff309188 0xff306a94 0x00000000
[   19.338532] sof-audio sof-audio: 0xff327bc8: 0x00000000 0x00000000 0xff2d0fee 0xff2d1004
[   19.338615] sof-audio sof-audio: 0xff327bcc: 0x00000000 0x00000000 0xff30d800 0x00000000
[   19.338699] sof-audio sof-audio: 0xff327bd0: 0xbf2c1939 0xff327c50 0xff327c70 0x00000010
[   19.338783] sof-audio sof-audio: 0xff327bd4: 0xff302944 0xff306800 0x000000bc 0x00060525
[   19.338867] sof-audio sof-audio: 0xff327bd8: 0xbf2c1939 0xff327c70 0xff327c90 0x00000010
[   19.338950] sof-audio sof-audio: 0xff327bdc: 0xff302944 0xff306800 0x000000bc 0x00000002

**Reproduce step:**
1: check dmesg immediately after system booted.

**Test env:**
sof master: 43da1696
sof tool: 4ac9248
kernel **debug/v4.19-backport**: 925637e3
tplg: sof-byt-rt5651.tplg-4ac9248 on BYT
      sof-apl-nocodec.tplg-4ac9248 on APL

[dmesg-byt.txt](https://github.com/thesofproject/sof/files/2537589/dmesg-byt.txt)

lgirdwood, 2018-11-01:
@markyang please also make sure you include the context prior to the crash

```
37079] sof-audio sof-audio:  Firmware info: version 1:1-43da1 build 44 on Nov  1 2018:18:00:04
[   19.337181] sof-audio sof-audio: found ext header type 1 size 0x88
[   19.337216] sof-audio sof-audio:  mailbox upstream 0x144000 - size 0x400
[   19.337219] sof-audio sof-audio:  mailbox downstream 0x144400 - size 0x400
[   19.337221] sof-audio sof-audio:  stream region 0x144a00 - size 0x200
[   19.337223] sof-audio sof-audio: booting DSP firmware completed
[   19.337231] sof-audio sof-audio: ipc rx done: 0x70000000: FW_READY
[   19.337237] sof-audio sof-audio: firmware boot complete
[   19.337287] sof-audio sof-audio: generating page table for 000000002ca7d7d5 size 0x10000 pages 16
[   19.337290] sof-audio sof-audio: dma_trace_pages: 16
[   19.337301] sof-audio sof-audio: stream_tag: -25879
[   19.337322] sof-audio sof-audio: ipc tx: 0x90010000: GLB_TRACE_MSG
[   19.337364] sof-audio sof-audio: error : DSP panic!
[   19.337430] sof-audio sof-audio: panic: dsp_oops_offset 1329152 offset 1329152
```

This shows the stream tag is wrong for DMA trace which probably causes the panic

Jiangxinx, 2018-11-02:
Summary:
Still can be reproduced on BYT.
dmesg:
[    7.815679] sof-audio sof-audio: booting DSP firmware
[    7.817377] sof-audio sof-audio: ipc rx: 0x70000000: FW_READY
[    7.817382] sof-audio sof-audio: ipc: DSP is ready 0x70000000 offset 0x144000
[    7.817394] sof-audio sof-audio:  Firmware info: version 1:1-43da1 build 44 on Nov  2 2018:18:01:34
[    7.817495] sof-audio sof-audio: found ext header type 1 size 0x88
[    7.817550] sof-audio sof-audio:  mailbox upstream 0x144000 - size 0x400
[    7.817554] sof-audio sof-audio:  mailbox downstream 0x144400 - size 0x400
[    7.817556] sof-audio sof-audio:  stream region 0x144a00 - size 0x200
[    7.817559] sof-audio sof-audio: booting DSP firmware completed
[    7.817568] sof-audio sof-audio: ipc rx done: 0x70000000: FW_READY
[    7.817576] sof-audio sof-audio: firmware boot complete
[    7.817648] sof-audio sof-audio: generating page table for 00000000d02a4412 size 0x10000 pages 16
[    7.817652] sof-audio sof-audio: dma_trace_pages: 16
[    7.817777] sof-audio sof-audio: stream_tag: -25548
[    7.817796] sof-audio sof-audio: ipc tx: 0x90010000: GLB_TRACE_MSG
[    7.817839] sof-audio sof-audio: error : DSP panic!
[    7.817909] sof-audio sof-audio: panic: dsp_oops_offset 1329152 offset 1329152
[    7.817942] sof-audio sof-audio: error: runtime exception
[    7.818001] sof-audio sof-audio: error: trace point 90010000
[    7.818062] sof-audio sof-audio: error: panic happen at :0
[    7.818121] sof-audio sof-audio: error: DSP Firmware Oops
[    7.818179] sof-audio sof-audio: error: Exception Cause: IllegalInstructionCause, Illegal instruction
[    7.818274] sof-audio sof-audio: EXCCAUSE 0x00000000 EXCVADDR 0x00000000 PS       0x00070520 SAR     0x0000001d
[    7.818377] sof-audio sof-audio: EPC1     0xff2c0983 EPC2     0x00000000 EPC3     0xff2c0f03 EPC4    0xff2c0f03
[    7.818479] sof-audio sof-audio: EPC5     0xff2c0f03 EPC6     0x00000000 EPC7     0x00000000 DEPC    0x00000000
[    7.818597] sof-audio sof-audio: EPS2     0x00000000 EPS3     0x00060720 EPS4     0x00060720 EPS5    0x00060720
[    7.818708] sof-audio sof-audio: EPS6     0x00000000 EPS7     0x00000000 INTENABL 0x00078f40 INTERRU 0x000000a2
[    7.818812] sof-audio sof-audio: stack dump from 0xff327bc0
[    7.818872] sof-audio sof-audio: 0xff327bc0: 0x5000b848 0x00060130 0x0000001d 0xc37cb6cf
[    7.818957] sof-audio sof-audio: 0xff327bc4: 0x5000b848 0xff309188 0xff306a94 0x00000000
[    7.819041] sof-audio sof-audio: 0xff327bc8: 0x00000000 0x00000000 0xff2d0fee 0xff2d1004
[    7.819125] sof-audio sof-audio: 0xff327bcc: 0x00000000 0x00000000 0x00060120 0x00000000
[    7.819209] sof-audio sof-audio: 0xff327bd0: 0xa77bdbcf 0x00000005 0xff2c12f4 0xff2c131f
[    7.819294] sof-audio sof-audio: 0xff327bd4: 0x00000000 0x00000000 0x00000000 0x00000001
[    7.819378] sof-audio sof-audio: 0xff327bd8: 0xeb5ffff7 0x00000005 0xff2c12f4 0xff2c131f
[    7.819462] sof-audio sof-audio: 0xff327bdc: 0x00000000 0x00000000 0x26047049 0x484c0c59

Reproduce step:
1: check dmesg immediately after system booted.

Test env:
sof master: 43da1696
sof tool: 937032f
kernel topic/sof-dev: a5ddc3d7
tplg: sof-byt-rt5651.tplg-937032f on BYT

Log:
[dmesg.log](https://github.com/thesofproject/sof/files/2540895/dmesg.log)

lgirdwood, 2018-11-02:
@Jiangxinx please use markdown when pasting logs - see my above changes.

ZhendanYang, 2018-11-05:
Move topology issue on up2 to #539 .

mengdonglin, 2018-11-05:
#443 is splitted into 2 bugs: one is #443 for stream_tag, another is described in #539

libinyang, 2018-11-16:
It seems this bug is assigned to me. I will debug this issue after I finish some other issues on UP2.

markyang, 2018-11-22:
**Summary:**
This issue can be reproduced on BYT with the following 3 tplgs:
test-ssp2-mclk-0-I2S-volume-s24le-s24le-48k-19200k-codec.tplg 
test-ssp2-mclk-0-I2S-volume-s16le-s16le-48k-19200k-codec.tplg  
and sof-byt-rt5651.tplg(input s32le output s24le).

**Test env:**
sof master: 7a793433
sof tool: 6da07a91
kernel sof-dev: b846ee56

mengdonglin, 2018-11-23:
@markyang could you verify the solution can solve this issue? We expect BYT can boot with the merged PR #613?

markyang, 2018-11-23:
**Summary:**
This issue cannot be reproduced on BYT with the following 3 tplgs:
test-ssp2-mclk-0-I2S-volume-s24le-s24le-48k-19200k-codec.tplg
test-ssp2-mclk-0-I2S-volume-s16le-s16le-48k-19200k-codec.tplg
and sof-byt-rt5651.tplg (input s32le output s24le).

**Test env:**
sof master: e8ef1f2
sof tool: ad3f3e4f
kernel sof-dev: 4a3aa5f3