thesofproject/sof #465: There is build error for the archieved tar ball

ewanglong, 2018-10-08:
1) download https://github.com/thesofproject/sof/archive/v1.2.tar.gz;
2)unzip it in local and follow the building steps:
$autogen.sh
$./configure --enable-rimage
$make
will meet below error:
man_apl.c:64:30: error: expected expression before ‘,’ token
    .major_version = SOF_MAJOR,
                              ^

It was due to there is no git tag by directly tar ball building. May refer to below script.
# get version from git tag
GIT_TAG=`git describe --abbrev=4 2>/dev/null`

So need specify or get the correct version under tar ball building environment.

xiulipan, 2018-10-08:
@ewanglong 
Yes, this bug is related to the `git tag`.
We will have some fix for it. We may not rely on the git tag anymore to avoid this problem.

xiulipan, 2018-10-09:
@lgirdwood 
I will check that. Maybe our script will try to regenerate the version and overwrite it?

zhuyingjiang, 2018-11-20:
PR 597 https://github.com/thesofproject/sof/pull/597 is submit to solve this problem, we use the .tarball-version file to get the version number if the .git folder can't be found. So every time anyone who make a tag should also change the version number in the .tarball-version file. Now the .tarball-version file should be upstreamed.

zhuyingjiang, 2018-11-21:
PR597 https://github.com/thesofproject/sof/pull/597 solved this problem, so close this issue.

jajanusz, 2018-11-21:
#605 reverted it, I described there what should be done.
One more thing tho:
I made tarball with `./autogen.sh && ./configure --enable-rimage && make dist` and it had .tarball-version inside it. However https://github.com/thesofproject/sof/archive/v1.2.tar.gz doesn't have it, you should look into how published tarballs are made because they look wrong

zhuyingjiang, 2018-11-22:
My PR is submit to master branch and only the release after that will take effect. v1.2.tar.gz is from another branch surely will not have .tarball-version  in it.

zhuyingjiang, 2018-11-22:
@lgirdwood this is a issue of " build error for the archived tar ball" or a issue for "make dist", I think they are 2 issues. My PR #597 resolved the  "build error for the archived tar ball". while make dist is another issue while relating to this

zhuyingjiang, 2018-11-22:
@jajanusz @lgirdwood 
A little more describe on this: we get version number from git tag, and compile need this version number for compiling. but if we release tarball with .git folder not included, the compile will fail for less the version number. My fix #597 is to resolve this.  
As for why  [ https://github.com/thesofproject/sof/archive/v1.2.tar.gz]( https://github.com/thesofproject/sof/archive/v1.2.tar.gz) can't work. I think:
1: the release have not include my patch for they are not in the same branch. How should it work?
2: The [ https://github.com/thesofproject/sof/archive/v1.2.tar.gz]( https://github.com/thesofproject/sof/archive/v1.2.tar.gz)  can't work is because the step for generate the tar ball is error, the "make dist" have excluded all the scripts even without "autogen.sh". the tarball in the github is not the same with the output of "make dist", someone else must have changed on that.

zhuyingjiang, 2018-11-22:
@jajanusz @lgirdwood 
It seems we have wrong git tag format after v1.1, so "git descibe" always get  v1.1 even we have tagged v1.1-rc and v1.2 , a new issue?
===================git show v1.1:===>
tag v1.1
Tagger: Liam Girdwood <liam.r.girdwood@linux.intel.com>
Date:   Wed Apr 4 14:58:51 2018 +0100

v1.1

commit 082c86f9c55ccaca307fedc0341f7304278c5aaf
Author: Pierre-Louis Bossart <pierre-louis.bossart@linux.intel.com>
Date:   Tue Apr 3 08:57:18 2018 -0500
====================git show v1.2:===>
commit 7dd4b1da6ea40bf8c5c404c0e606e48cd3d33e62
Merge: 0875f0c 380c067
Author: Liam Girdwood <lgirdwood@gmail.com>
Date:   Tue Sep 18 12:53:02 2018 +0100

lgirdwood, 2018-11-22:
@jajanusz @zhuyingjiang version 1.2 tag is missing from master tagged during the long stabilisation freeze in the stable branch whilst master churned. i.e. we could not merge stable back into master due to too many conflicts. This has been resolved for 1.3 since we are all working on master now.

zhuyingjiang, 2018-11-30:
@jajanusz I see in current "make dist", the autogen.sh and the scripts/ folder are removed from the tarball, are you still working on this? if yes, remove these files is intended or we need fix as the previous version tarball release do ?  What will be the compile steps after uncompressed the tarball? do ./configure --enable-rimage first?

zhuyingjiang, 2018-11-30:
@lgirdwood Since the title and depict of this issue is about "fix the build error", but in fact it is the "make dist" feature can't work. Shall we close this issue and new a "make dist" for more efficiency track? Or at least  change the title of this issue? I see changes between old tarball release and the current one, and the current tarball release exclude autogen.sh and the scripts/ folder, is this intended to do and someone still working on this?

zhuyingjiang, 2018-12-05:
Close this issue, #670 to support the feature of "make dist" to do the tarball release.

closed 2018-12-05 by zhuyingjiang

mengdonglin, 2018-12-11:
Mark this issue as "won't fix". We won't fix this for SOF 1.2 but will fix it for new upstream releases (feature #670)