thesofproject/sof #473: Verbose tracing breaks new logger

cujomalainey, 2018-10-08:
Enabling TRACEV in the firmware causes the logger to output absolutely nothing. Firmware appears to operating just fine according to dmesg otherwise.

Observed at:
Kernel: 1 commit off 93853cbf8f8252ec04dd8b70328cb004d1d4566a to patch glk dialog max codec for 4.19
SOF: ccaff5c87e5bb2ba29fc804abde0f47bd93cebea
SOFT: 2d7a933aa8655e99d95d7bafec53a14b410494af (dirty, but only with regards to some m4 files)

cujomalainey, 2018-10-09:
I think this might be related to a larger issue in the system due to something like memory corruption as I'm still experiencing trace issues with a dirty build. But I can fix them by adding a trace in weird spot in the code (host_copy_int) (tested 5X to be sure adding and removing the trace was fixing the logging.) Without the additional trace I only get the first 3 traces out of the dsp even though according to dmesg there is plenty of communication going on. If I add the trace I get plenty of pages worth of traces.

If you want to get an idea of where I am, I am checked out on master in my own sof repo (inital new component in the branch) https://github.com/cujomalainey/sof/commit/71d75ae0c229376749af85d5467c854cfabf789c
As you can see all I have is the skeleton on a new component, not really much going on. Not sure what could cause this behaviour in my code. The TRACEV bug still exists though on a clean repo.

mengdonglin, 2018-10-10:
@keqiaozhang Could you try to reproduce this issue?

keqiaozhang, 2018-10-10:
@mengdonglin 
We can reproduce this issue with logger. Even IPC error, there's still nothing in memory window.
Only when firmware panic, logger will shows some trace log.

cujomalainey, 2018-10-10:
With respect to my second comment, I no longer am seeing tracing fail just from my own code with the code I pulled in from last night, but if its a memory corruption issue it could have been the memory map readjusting. The TRACEV bug is still reproducible.

akloniex, 2018-10-11:
@cujomalainey  @mengdonglin  @keqiaozhang 
Enabling TRACEV and rebuilding FW causes .ldc file to regenerate. Please make sure that it's the dictionary file that you feed logger with, through -l paramter.

Also make sure that input file is correctly specified. 
With no additional parameters, logger defaults to /sys/kernel/debug/sof/etrace, 
with -p logger reads from stdin, (e.g. piping from other process),
with -t it's /sys/kernel/debug/sof/trace, 
with -i you may specify file to read explicitly (e.g. for offline purposes).

keqiaozhang, 2018-10-11:
@akloniex @mengdonglin 
Confirmed that logger works well with -t parameter.
Command: sudo logger -l sof-apl.ldc  -t

cujomalainey, 2018-10-11:
@akloniex I am using the new ldc on tracev enabling and I forgot to mention all of these trace failures are with the -t parameter

keqiaozhang, 2018-10-15:
I re-tested it with top sof-master and enabled tracev, logger works well with -t parameter.
Here are my test steps:
1. Enable tracev(trace.h):
-#define TRACEV 0
+#define TRACEV 1
2. make sof and copy the new sof-apl.ldc and sof-apl.ri to Yorp
3. make soft and copy the new topology and logger to Yorp
3. reboot Yorp
4. run audio playback: aplay -Dhw:0,5 audio.wav
5. sudo logger -l sof-apl.ldc -t
BTW, without step 4, there's no trace output.

kernel-cros-sof-v4.14-rebase: 274bfc6 
sof-master: 1024e94
soft-master: c0a881a

lgirdwood, 2018-10-16:
@keqiaozhang I applied some logger updates yesterday so could this be closed now ?

cujomalainey, 2018-10-16:
I will test again either today or tomorrow and close if it passes

cujomalainey, 2018-10-16:
I was unable to verify as logging appears to be broken on master and sof-dev for me

```
[  121.363911] sof-audio sof-audio: dma_trace_pages: 16
[  121.364545] sof-audio sof-audio: error: cannot create debugfs entry for trace
[  121.364633] sof-audio sof-audio: warning: failed to init trace after resume -19
```
I updated the kernel, firmware, logger tool, and ldc file

lgirdwood, 2018-10-17:
@cujomalainey oh, I think @ranj063 or someone else mentioned this yesterday. @ranj063 anything missing from the kernel ?

cujomalainey, 2018-10-18:
TRACEV is now working for me on topic/sof-dev and master

keyonjie, 2018-10-19:
@cujomalainey good to know it works now.
Are you run topic/sof-dev directly on chrome? or Ubuntu?

cujomalainey, 2018-10-19:
@keyonjie I'm running it on chrome, but I have the bxt_da7219_max####a driver patched for glk. You can see the modification here 60000680bd4f01f2dd37b3a73991149913ed9166

keyonjie, 2018-10-19:
@cujomalainey thanks, I am thinking if it is possible to run thesofproject/linux topic/sof-dev directly on chrome book(thesofproject/linux topic/cros-sof-v4.14-rebase works for that).

cujomalainey, 2018-10-19:
We just need that driver patched and we can run sof-dev no problem. I can send a pull request if we want to do that. It is slightly buggy though. Left channel on speaker only works for about 500ms before cutting out but it appears to be a UCM issue that is setting the left volume to 0 and headset isnt working currently.

If I could work off sof-dev without constantly rebasing patches that would make debugging and my filter development go much faster.

cujomalainey, 2018-10-19:
@lgirdwood sounds good. I'll clean up my patch I have locally and send it off soon. The 4.18 kernel on yorp and is debugable, it just takes the UI forever to get going. But if you are using ssh it's completely functional. Its how I have been working on building my filter additions since the effect topology commits are not in the cros-rebase branch last time I check

Jiangxinx, 2018-10-25:
Not reproduce on sof-logger while using UP2.

    0      2     HOST  56308262.447917         2.395833               host.c:650        cpy
    0      2     HOST  56308265.312500         2.864583            hda-dma.c:229        GwU
    0      2     HOST  56308268.802083         3.489583               host.c:145        irq
    0      2   BUFFER  56308272.291667         3.489583             buffer.c:145        pro
    0      2  unknown  56308274.843750         2.552083             buffer.c:146        value 25165824
    0      2  unknown  56308277.291667         2.447917             buffer.c:147        value 131456
    0      2  unknown  56308279.635417         2.343750             buffer.c:148        value 12583104
    0      2    POWER  56308282.552083         2.916667         pm_runtime.c:78         put
    0      2     PIPE  56308317.135417        34.583332           pipeline.c:914        CD+
    0      2   VOLUME  56308319.739583         2.604167             volume.c:445        cpy
    0      2   BUFFER  56308350.781250        31.041666             buffer.c:145        pro
    0      2  unknown  56308353.281250         2.500000             buffer.c:146        value 50331648
    0      2  unknown  56308355.729167         2.447917             buffer.c:147        value 197376
    0      2  unknown  56308358.072917         2.343750             buffer.c:148        value 25166208

STEP:
1.Enable tracev(trace.h):
     -#define TRACEV 0
     +#define TRACEV 1
2.re-make sof、soft、logger and replace them on UP2
3.reboot
4.aplay -Dhw:0,0 audio.wav
5.sudo sof-logger -l sof-apl.ldc -t -o logger.log

Test env:
hardware version: UP2
sof-master: b793f0c
soft-master: c9a4c87
kernel sof-dev: fe8404a
tplg: sof-apl-pcm512x.tplg

mengdonglin, 2018-10-25:
Please follow up in kernel repo https://github.com/thesofproject/linux/issues/214

closed 2018-10-25 by mengdonglin