thesofproject/sof #477: Error trace shows nothing in memory window

xiulipan, 2018-10-10:
Now trace_error will show nothing in memory window. Either rmbox or logger will show nothing in error trace. Even cat the debugfs will show empty.

Error trace may be broken with new logging

akloniex, 2018-10-16:
PR addessing this issue was merged. Please verify and update this issue accordingly,

ZhendanYang, 2018-10-17:
On up2 with nocodec, rmbox and logger are still empty, while rmbox -t and logger -t are not.

cujomalainey, 2018-10-17:
Still seeing empty logs as well on glk

cujomalainey, 2018-10-17:
@akloniex why are there two logger binaries? Which one am I supposed to be using? We should delete the other one otherwise this is confusing.

akloniex, 2018-10-23:
@cujomalainey Only 1 binary of sof-logger is built in the make process, You probably have some leftovers from previous runs. Please cleanup your workspace, e.g. git clean -f
@ZhendanYang Please verify if the issue occurs on revisions following PR https://github.com/thesofproject/sof/pull/490 and update this issue accordingly. Thank you.

cujomalainey, 2018-10-23:
@akloniex yep that was the issue. Sorry, I should have checked that before. I was thrown by the simultaneous name change and location move.

ZhendanYang, 2018-10-24:
@akloniex 
I did some tests on up2 with codec. I introduced XRUN injection when playback. and sof-logger -t showed some errors in red color words:

    0      1   VOLUME  92935105.520833        50.208332             volume.c:463 	xro
    0      2     COMP  92935108.072917         2.552083 ../../src/include/sof/audio/component.h:414 	Xov
    0      2  unknown  92935110.572917         2.500000 ../../src/include/sof/audio/component.h:415 	value 65536
    0      2  unknown  92935112.968750         2.395833 ../../src/include/sof/audio/component.h:416 	value 384
    0      1     PIPE  92935119.583333         6.614583           pipeline.c:903 	ePU
    0      1  unknown  92935121.822917         2.239583           pipeline.c:904 	value 4
    0      1     PIPE  92935124.635417         2.812500           pipeline.c:1198	pxr

But sof-logger without the parameter -t still showed nothing:

yzd@up-03:~/test-10.23$ sudo ./sof-logger -l sof-apl.ldc-master-gcc
 CORE  LEVEL  COMP_ID        TIMESTAMP            DELTA                FILE_NAME	CONTENT

Jiangxinx, 2018-10-25:
sof-logger without the parameter -t can show error trace.

    0      1   VOLUME  31120390.833333   1004787.812500             volume.c:463        xro
    0      1     PIPE  31120404.739583        13.906250           pipeline.c:903        ePU
    0      1  unknown  31120406.927083         2.187500           pipeline.c:904        value 4
    0      1     PIPE  31120410.000000         3.072917           pipeline.c:1198       pxr
    0      1   VOLUME  32124928.281250   1004518.250000             volume.c:463        xro
    0      1     PIPE  32124941.927083        13.645833           pipeline.c:903        ePU
    0      1  unknown  32124944.270833         2.343750           pipeline.c:904        value 4
    0      1     PIPE  32124947.187500         2.916667           pipeline.c:1198       pxr
    0      1   VOLUME  33129776.875000   1004829.687500             volume.c:463        xro
    0      1     PIPE  33129790.885417        14.010417           pipeline.c:903        ePU
    0      1  unknown  33129793.072917         2.187500           pipeline.c:904        value 4
    0      1     PIPE  33129795.989583         2.916667           pipeline.c:1198       pxr

STEP:
1.aplay -Dhw:0,1 audio.wav
2.sudo ./xrun_injection.sh
3.sudo sof-logger -l sof-apl.ldc -o logger.log

Test env:
hardware version: UP2
sof-master: b793f0c1
soft-master: c9a4c87
kernel sof-dev: fe8404a
tplg: sof-apl-pcm512x.tplg

Log:
[dmesg.log](https://github.com/thesofproject/sof/files/2513718/dmesg.log)
[logger.log](https://github.com/thesofproject/sof/files/2513719/logger.log)

Jiangxinx, 2018-10-30:
Not reproduced with kernel branch debug/v4.19-backport, on APL UP2 platfrom.

STEP:
1.aplay -Dhw:0,1 audio.wav
2.sudo ./xrun_injection.sh
3.sudo sof-logger -l sof-apl.ldc -o logger.log

Test env:
hardware version: UP2
sof-master: 04b1ce48
soft-master: 4ac9248
kernel debug/v4.19-backport: e7da1c4
tplg: sof-apl-pcm512x.tplg

Jiangxinx, 2018-11-01:
Not reproduced with kernel branch debug/v4.19-backport, on APL UP2 platfrom.

STEP:
1.aplay -Dhw:0,0 audio.wav
2.sudo ./xrun_injection.sh
3.sudo sof-logger -l sof-apl.ldc -o logger.log

Test env:
hardware version: UP2
sof-master: 43da1696
soft-master: 4ac9248
kernel debug/v4.19-backport: 925637e3
tplg: sof-apl-pcm512x.tplg