thesofproject/sof #479: DMIC channels swapping LSB sequence is wrong

keqiaozhang, 2018-10-10:
When doing the 4ch DMIC capture,  the LSB bits of 32 bit PCM samples should have this side information:
[image]
The LSB bits (0:1) will count 0, 1, 2, 3, 0, 1, 2, 3, … from beginning to end of capture.
But check the captured raw data in binary form, the samples would be from channel 0, and only some from channel 1, with 2 and 3 missing totally. Please refer to below:
00000000: 00000000 00000000 00000000 00000000 00000001 00000000  ......
00000006: 00000000 00000000 00000000 00000000 00000000 00000000  ......
0000000c: 00000001 00000000 00000000 00000000 00000000 00000000  ......
00000012: 00000000 00000000 00000001 00000000 00000000 00000000  ......
00000018: 00000000 00000000 00000000 00000000 00000001 00000000  ......
0000001e: 00000000 00000000 00000000 00000000 00000000 00000000  ......
00000024: 00000001 00000000 00000000 00000000 00000000 00000000  ......
0000002a: 00000000 00000000 00000001 00000000 00000000 00000000  ......
00000030: 00000000 00000000 00000000 00000000 00000001 00000000  ......
00000036: 00000000 00000000 00000000 00000000 00000000 00000000  ......
0000003c: 00000001 00000000 00000000 00000000 00000000 00000000  ......
00000042: 00000000 00000000 00000001 00000000 00000000 00000000  ......
00000048: 00000000 00000000 00000000 00000000 00000001 00000000  ......
0000004e: 00000000 00000000 00000000 00000000 00000000 00000000  ......
00000054: 00000001 00000000 00000000 00000000 00000000 00000000  ......
0000005a: 00000000 00000000 00000001 00000000 00000000 00000000  ......
00000060: 00000000 00000000 00000000 00000000 00000001 00000000  ......
00000066: 00000000 00000000 00000000 00000000 00000000 00000000  ......
0000006c: 00000001 00000000 00000000 00000000 00000000 00000000  ......
00000072: 00000000 00000000 00000001 00000000 00000000 00000000  ......
00000078: 00000000 00000000 00000000 00000000 00000001 00000000  ......
0000007e: 00000000 00000000 00000000 00000000 00000000 00000000  ......
00000084: 00000001 00000000 00000000 00000000 00000000 00000000  ......
0000008a: 00000000 00000000 00000001 00000000 00000000 00000000  ......
00000090: 00000000 00000000 00000000 00000000 00000001 00000000  ......
00000096: 00000000 00000000 00000000 00000000 00000000 00000000  ......
0000009c: 00000001 00000000 00000000 00000000 00000000 00000000  ......
000000a2: 00000000 00000000 00000001 00000000 00000000 00000000  ......
000000a8: 00000000 00000000 00000000 00000000 00000001 00000000  ......
000000ae: 00000000 00000000 00000000 00000000 00000000 00000000  ......
000000b4: 00000001 00000000 00000000 00000000 00000000 00000000  ......
000000ba: 00000000 00000000 00000001 00000000 00000000 00000000  ......
000000c0: 00000000 00000000 00000000 00000000 00000001 00000000  ......

How to reproduce:
1. arecord -Dhw:0,6 -c 4 -f S32_LE -r 48000 -t raw -d 1 capture.raw
2. check the capture.raw in binary form.

Test ENV:
Platform: APL
sof-1.2-stable: 7dd4b1da6e
kernel-sof-dev: 93853cbf8f
soft-master: e37aa3f4
tplg: sof-apl-pcm512x.tplg

singalsu, 2018-10-12:
I'm going to check first that the gain ramp that the driver does is not impacting this.

singalsu, 2018-10-12:
The channel info bits are FIFO packer feature so any DMIC configuration cannot overwrite the bits, good. I did a test here:

```
$ arecord -Dhw:0,6 -c 4 -f S32_LE -r 48000 -t raw -d 10 capture.raw
$ hexdump -n 128 -e '/4 "%08X\n"' capture.raw
00000000
00000001
00000002
00000003
00000000
00000001
00000002
00000003
00000000
00000001
00000002
00000003
00000000
00000001
00000002
00000003
00000000
00000001
00000002
00000003
00000000
00000001
00000002
00000003
```

keqiaozhang, 2018-10-15:
@singalsu 
For the topology, I used default sof-apl-nocodec.tplg, not sof-apl-pcm512.tplg, just a typo. 
Since there's one DSP panic issue(sof/issues/443) in sof-master. So I used sof-stable-1.2 branch for test.
So please help to provide branch/commit you used for test, I can re-test this issue with your test configure by tomorrow.

singalsu, 2018-10-15:
I'll make tomorrow two PRs to help testing this (the topology that I emailed to you plus a pass-trough setting for EQ, the sof-apl-eq-dmic topology that is also suitable for this is merged now).

singalsu, 2018-10-15:
I've here completed test of 1000x 1s long captures with no channel swap faults observed.