thesofproject/sof #497: set stream tag for host dma and link dma individually

RanderWang, 2018-10-19:
Now host dma and link dma share the same stream tag. It is not fit for some cases that the stream tags
are different with link dma. So we need to set them individually

lgirdwood, 2018-10-19:
@RanderWang can you confirm who allocates both tags, my understanding is that your change means host tag is allocated by driver and link tag by the FW.

mmaka1, 2018-10-19:
@lgirdwood Both are done by the driver, so we'd need two separate tags coming down to the FW.

keyonjie, 2018-10-19:
@lgirdwood yes, currently we are using the same tag for both host dma and link dma, this introducing limitation and issues for hda/hdmi capture where we need the exact correct tag(to align with HDA SDI pin).

RanderWang, 2019-03-05:
fixed

closed 2019-03-05 by RanderWang