thesofproject/sof #509: IPC error occurs after the system booted

markyang, 2018-10-26:
**Summary:**
 IPC error occurs after the system booted on APL UP2 and Cannonlake. 
 It will be OK if sof is rolled-back to b793f0c1 on master branch.
 the result is empty when run 'sudo ./sof-logger -l sof-apl.ldc-master-gcc-3f3d7c04'
**dmesg**:
[    8.965431] sof-audio sof-audio: ipc tx: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
[    8.965553] sof-audio sof-audio: error: ipc error for 0x60010000 size 0x14
[    8.965564] sof-audio sof-audio: ASoC: sof-audio hw params failed: -22
[    8.965568] sof-audio sof-audio: pcm: free stream 0 dir 0
...
the logger results of 'sudo ./sof-logger -l sof-cnl.ldc':
    0      1      IPC   1432817.760417      1239.375000            handler.c:186        eAS
    0      1      IPC   1433971.875000      1154.114624            handler.c:186        eAS
    0      1      IPC   1435518.802083      1546.927124            handler.c:186        eAS
    0      1      IPC   1435930.520833       411.718750            cnl-ipc.c:82         Pen
    0      1      IPC   1436856.510417       925.989563            handler.c:186        eAS
    0      1      IPC   1438003.333333      1146.822876            handler.c:186        eAS

**Reproduce step**:
1: check dmesg immediately.
2: 'aplay -l' is OK.
3: speaker-test -Dhw:0,0 -r 48000 -f s16_le -c2 -t wav
speaker-test 1.1.3
Playback device is hw:0,0
Stream parameters are 48000Hz, S16_LE, 2 channels
WAV file(s)
Rate set to 48000Hz (requested 48000Hz)
Buffer size range from 384 to 16320
Period size range from 192 to 4032
Using max buffer size 16320
Periods = 4
Unable to set hw params for playback: Invalid argument
Setting of hwparams failed: Invalid argument

**Test env**:
sof master: 3f3d7c04
sof tool: c9a4c87
kernel sof-dev: fe8404a3
tplg: sof-apl-pcm512x.tplg-c9a4c87 on up2
    test-ssp0-mclk-0-I2S-volume-s16le-s16le-48k-24000k-nocodec.tplg on cnl

**Log**:
[logger-t-apl.txt](https://github.com/thesofproject/sof/files/2518065/logger-t-apl.txt)
[dmesg-apl.txt](https://github.com/thesofproject/sof/files/2518066/dmesg-apl.txt)
[logger-t-cnl.txt](https://github.com/thesofproject/sof/files/2518067/logger-t-cnl.txt)
[logger-cnl.txt](https://github.com/thesofproject/sof/files/2518068/logger-cnl.txt)
[dmesg-cnl.log](https://github.com/thesofproject/sof/files/2518070/dmesg-cnl.log)

mengdonglin, 2018-10-26:
@RanderWang It seems your PR https://github.com/thesofproject/sof/pull/505 caused the regression. Could you comment and share to the kernel PR which is also needed?

lgirdwood, 2018-10-26:
@RanderWang reverted, it seems like the kernel PR you sent does not align with #505

mengdonglin, 2018-10-27:
@DarrenxWang Could you help to check if this issue is gone?  The PR #505 which brings this regression was reverted.

DarrenxWang, 2018-10-29:
@mengdonglin The issue is not reproduced after revert the PR #505. This week we will monitor the PR updates. Thanks!

closed 2018-10-29 by DarrenxWang

DarrenxWang, 2018-10-29:
The PR #513 revert the PR #505.