thesofproject/sof #510: Tplg component load failed

Jiangxinx, 2018-10-26:
Summary:
Tplg component load failed on APL UP2 while using sof-stable1.2 7dd4b1da.
Sof-master b793f0c is OK.

dmesg:
[    5.891558] sof-audio sof-audio: ipc tx: 0x30010000: GLB_TPLG_MSG: COMP_NEW
[    5.891627] sof-audio sof-audio: error: ipc error for 0x30010000 size 0x14
[    5.891637] sof-audio sof-audio: error: DSP failed to add widget id 0 type 27 name : HDA0.OUT stream iDisp1 reply -22
[    5.891642] sof-audio sof-audio: ASoC: failed to load widget HDA0.OUT
[    5.891644] sof-audio sof-audio: error: tplg component load failed -22
[    5.891660] sof-audio sof-audio: error: failed to load DSP topology -22
[    5.891663] sof-audio sof-audio: ASoC: failed to probe component -22
[    5.891694] bxt-pcm512x bxt-pcm512x: ASoC: failed to instantiate card -22
[    5.891748] bxt-pcm512x bxt-pcm512x: snd_soc_register_card failed -22
[    5.891763] bxt-pcm512x: probe of bxt-pcm512x failed with error -22
...
STEP:
1: 'aplay -l' show nothing.
2: check dmesg immediately.

Test env:
hardware version: UP2
sof-stable1.2: 7dd4b1da
soft-master: c9a4c87
kernel sof-dev: fe8404a
tplg: sof-apl-pcm512x.tplg

Log:
[dmesg.log](https://github.com/thesofproject/sof/files/2518146/dmesg.log)

lgirdwood, 2018-10-29:
@xiulipan driver should complain here before we send to FW. Do we check that a valid BE DAI link exists when we parse each BE def in the topology ?

mengdonglin, 2018-10-29:
This seems to be a backward compatibility issue:
SOF v1.2 cannot support HD-A backends. But latest UP2 topology includes HD-A backends.

mengdonglin, 2018-10-29:
@lgirdwood  @xiulipan Can driver query the firmware capabilities on HD-A support, then decide whether to load the HD-A pipelines in FW?  Or maybe it simpler to make SND_SOC_SOF_HDA_COMMON configurable for driver on APL?

https://github.com/thesofproject/linux/blob/topic/sof-dev/sound/soc/sof/intel/Kconfig shows that HD-A support (SND_SOC_SOF_HDA_COMMON) is always enabled for ApolloLake (UP2). And the UP2 machine driver https://github.com/thesofproject/linux/blob/topic/sof-dev/sound/soc/intel/boards/bxt_pcm512x.c has HD-A backend links.  This means driver assume HD-A is always supported on APL.

lgirdwood, 2018-10-29:
@mengdonglin easier to just say HDA not supported in 1.2. In this case the FW did the right thing.

closed 2018-10-29 by lgirdwood