thesofproject/sof #515: Pulseaudio playback has sticky noise issue

ZhendanYang, 2018-10-29:
Summary:
on BYT minnowboard and APL up2, pulseaudio playback is not smooth and has sticky noise. There is no error log in dmesg or sof-logger.

Reproduce step:
1. paplay music.wav or watch a video on youtube

Test env:
up2 and minnowboard both with codec
sof-master: e101263
sof-stable1.2: 7dd4b
soft-master: b5eb15
kernel sof-dev: 03cadf8
tplg: sof-apl-pcm512x.tplg on up2 and test-ssp2-mclk-0-I2S-volume-s16le-s16le-48k-24000k-codec.tplg

Log:
sof-logger without parameter -t are both empty.
sof-logger with -t on BYT is empty.
[logger-t-apl.txt](https://github.com/thesofproject/sof/files/2524284/logger-t-apl.txt)
[dmesg-apl.log](https://github.com/thesofproject/sof/files/2524285/dmesg-apl.log)
[dmesg-byt.log](https://github.com/thesofproject/sof/files/2524286/dmesg-byt.log)

lgirdwood, 2018-10-29:
@ZhendanYang please turn on pulseaudio debug and paste the logs here. Must be something pulseuadio is complaining about.

Jiangxinx, 2018-11-01:
Still can be reproduced with kernel branch debug/v4.19-backport, on APL UP2 platfrom.

STEP:
1.pulseaudio --kill
2.pulseaudio --start --log-level=debug --log-target=newfile:/home/up2/pulse.log
3.paplay audio.wav

Test env:
hardware version: UP2
sof-master: 43da169
soft-master: 4ac9248
kernel debug/v4.19-backport: 925637e3
tplg: sof-apl-pcm512x.tplg

Pulseaudio log:
[pulse.log](https://github.com/thesofproject/sof/files/2537840/pulse.log)

markyang, 2018-11-01:
This issue can't be reproduced on UP2.

STEP:
1.aplay -Dhw:0,0 audio.wav
2.sudo ./xrun_injection.sh

Test env:
hardware version: UP2
sof-master: 43da169
soft-master: 4ac9248
kernel debug/v4.14-backport: 94f7a53e
tplg: sof-apl-pcm512x.tplg

lgirdwood, 2018-11-01:
@ZhendanYang can you attach a capture of the noise as a smal wav file and some screen shots of audacity of the noise at zoom in and zoom out levels (play a sine wave)

ZhendanYang, 2018-11-02:
I used aplay and paplay to playback the same sine wave, only paplay has noise issue.

zql@zql-MB-T:~$ aplay -D hw:0,0 -c2 -r 48000 -f s16_le sweep.wav -vv -i
Playing WAVE 'sweep.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Stereo
Hardware PCM card 0 'sof-bytcr-rt5651' device 0 subdevice 0

zql@zql-MB-T:~$ paplay sweep.wav 

I recorded one sine wave for each:
test sine wave: 
[image]
[image]

aplay test result: 
[image]
[image]

paplay test result: 
[image]
[image]

Here are the three sine wave files：
[wave.zip](https://github.com/thesofproject/sof/files/2540909/wave.zip)

lgirdwood, 2018-11-02:
@ZhendanYang thanks, good info - btw its best to use a constant frequency sine wave in the future, that way it's easy to work out duration of any glitches.

Can you confirm that PA was enabled during the aplay test (i.e. PA was running during aplay usage). I need to rule out if this was paplay only issue.

Fwiw looks like PA (via paplay) is using period of 84ms and buffer of 340ms
```
I: [alsa-sink-Port5 (*)] alsa-util.c: Trying to disable ALSA period wakeups, using timers only
D: [alsa-sink-Port5 (*)] alsa-util.c: Maximum hw buffer size is 340 ms
D: [alsa-sink-Port5 (*)] alsa-util.c: Set buffer size first (to 16320 samples), period size second (to 4032 samples).
I: [alsa-sink-Port5 (*)] alsa-util.c: ALSA period wakeups disabled
D: [alsa-sink-Port5 (*)] alsa-sink.c: hwbuf_unused=0
D: [alsa-sink-Port5 (*)] alsa-sink.c: setting avail_min=15439
I: [alsa-sink-Port5 (*)] alsa-sink.c: Time scheduling watermark is 18.38ms
I: [alsa-sink-Port5 (*)] alsa-sink.c: Resumed successfully...
I: [alsa-sink-Port5 (*)] alsa-sink.c: Starting playback.
```
but it is rewinding
```
D: [alsa-sink-Port5 (*)] alsa-sink.c: hwbuf_unused=0
D: [alsa-sink-Port5 (*)] alsa-sink.c: setting avail_min=15439
D: [alsa-sink-Port5 (*)] alsa-sink.c: Requested to rewind 65280 bytes.
D: [alsa-sink-Port5 (*)] alsa-sink.c: Limited to 65024 bytes.
D: [alsa-sink-Port5 (*)] alsa-sink.c: before: 16256
D: [alsa-sink-Port5 (*)] alsa-sink.c: after: 16256
D: [alsa-sink-Port5 (*)] alsa-sink.c: Rewound 65024 bytes.
D: [alsa-sink-Port5 (*)] sink.c: Processing rewind...
```

This could be a race between PA writing data and DSP reading during the rewind.

lgirdwood, 2018-11-05:
@ZhendanYang since it's paplay only issue (and nobody uses paplay directly) I've moved to P3 as it may be paplay trying to rewind too close to the pointers.

keqiaozhang, 2018-11-06:
@ZhendanYang @lgirdwood 
I think we can test some Youtube videos, see if we can reproduce this issue.

ZhendanYang, 2018-11-06:
@keqiaozhang 
Yes, this issue can reproduce when test with Youtube videos.

lgirdwood, 2018-11-06:
@ZhendanYang can you confirm pulseaudio running when playing youtube ?

ZhendanYang, 2018-11-07:
@lgirdwood If pulseaudio is disabled, then paplay can't work and youtube videos are all silent.

lgirdwood, 2018-11-07:
@ZhendanYang ok, so you are confirming that pulseaudio is running whilst using youtube is playing ? (btw, web browsers can still play youtube media with pusleaudio disabled).

If yes,  this issue is more to do with pulseaudio and certain client configurations playing audio. I've changed the title and  can you attempt to reprodue with aplay + pulseaudio (but use aplay command line switch to use same buffer/period sizes as paplay)

ZhendanYang, 2018-11-08:
@lgirdwood I'm confirming that pulseaudio is running whilst using youtube is playing.
I use  `paplay --rate=48000 --format=s16le --channels=2 music.wav` 
and  `aplay -D hw:0,0 -c 2 -r 48000 -f s16_le music.wav` for test.
I think they both use the same buffer/period sizes:
```
[ 9744.264520] sof-audio sof-audio: pcm: open stream 0 dir 0
[ 9744.264538] sof-audio sof-audio: period min 192 max 16384 bytes
[ 9744.264549] sof-audio sof-audio: period count 2 max 16
[ 9744.264559] sof-audio sof-audio: buffer max 65536 bytes
[ 9744.265026] sof-audio sof-audio: rate_min: 48000 rate_max: 48000
[ 9744.265039] sof-audio sof-audio: channels_min: 2 channels_max: 2
[ 9744.265057] sof-audio sof-audio: rate_min: 48000 rate_max: 48000
[ 9744.265067] sof-audio sof-audio: channels_min: 2 channels_max: 2
[ 9744.265089] sof-audio sof-audio: rate_min: 48000 rate_max: 48000
[ 9744.265099] sof-audio sof-audio: channels_min: 2 channels_max: 2
[ 9744.265115] sof-audio sof-audio: pcm: hw params stream 0 dir 0
```
I can't reproduce noise issue using aplay. Actually, except for paplay .wav file, system sounds also have such kind of noise, for example, the sound that I type 'Tab' key to automatically complete a word.

Jiangxinx, 2018-11-08:
@lgirdwood I retested pulseaudio on both apl and byt, got the result:
1.Paplay still have noise with the kernel config (not use DPIB):
CONFIG_SND_SOC_SOF_DEBUG_FORCE_IPC_POSITION=y
2.Can't reproduce noise with the kernel config (use DPIB by default):
#CONFIG_SND_SOC_SOF_DEBUG_FORCE_IPC_POSITION
3.Aplay can't reproduce noise with both DPIB config kernel and non-DPIB config kernel.

Step:
1.paplay audio.wav

Env:
hardware version: UP2
sof-master: 019eadeb
soft-master: 937032f
kernel topic/sof-dev: af44aa9b
tplg: sof-apl-pcm512x.tplg

Logger is blocked by #551

lgirdwood, 2018-11-08:
@Jiangxinx good investigative work ! DPIB mode should be enabled by default  in kernel config (please check). Difference with DPIB position readback is accuracy and latency compared to IPC/SRAM position readback (where position reported can be up to 1ms old) probably causing PA to render to close to the DSP pointer. I will close this since it's not a problem with DPIB mode and aplay works fine, it just appears to be mostly related to paplay.

closed 2018-11-08 by lgirdwood

mengdonglin, 2018-11-09:
@keyonjie fyi who enables DPIB mode :-)

However, Baytrail has no DPIB support and so always using IPC position update. So I expect MinnowBoard (BYT) can still suffer from this problem occasionally.