thesofproject/sof #523: error trace will be cleared after a few seconds

keqiaozhang, 2018-11-01:
Now sof-logger is back to work now, but I found that error trace will be cleared in memory window after few seconds.

Please refer to below steps:
chronos@localhost ~ $ sudo /usr/local/bin/sof-logger -l /root/sof-apl.ldc
 CORE  LEVEL  COMP_ID        TIMESTAMP            DELTA                FILE_NAME        CONTENT
    0      1  unknown    103910.364583    103910.367188          dma-trace.c:417        Number of dropped logs: 1
    0      1      IPC    103974.062500        63.697918            apl-ipc.c:82         Pen
chronos@localhost ~ $ sudo /usr/local/bin/sof-logger -l /root/sof-apl.ldc
 CORE  LEVEL  COMP_ID        TIMESTAMP            DELTA                FILE_NAME        CONTENT
    0      1  unknown    103910.364583    103910.367188          dma-trace.c:417        Number of dropped logs: 1
    0      1      IPC    103974.062500        63.697918            apl-ipc.c:82         Pen
chronos@localhost ~ $ sudo /usr/local/bin/sof-logger -l /root/sof-apl.ldc
 CORE  LEVEL  COMP_ID        TIMESTAMP            DELTA                FILE_NAME        CONTENT

At the third run, the trace log is cleared.

Tested with:
soft-master:4ac9248f
sof-master:43da16
kernel-debug/cros-v4.14-rebase: 90eee4

xiulipan, 2018-11-01:
@keqiaozhang 
Could you also paste the dmesg as well. To see if there are any debugfs error message in dmesg.

lgirdwood, 2018-11-01:
@xiulipan this is not coming DMA so it's possible something is over writting the window. Can you manually dump the window contents. If there is data then it's a tool bug if there is no data then a FW bug.

markyang, 2018-11-02:
**Summary:**
  This issue can be reproduced on UP2 too.
  no dmesg error.
_sudo sof-logger -l ./sof-apl.ldc-master-gcc-43da1696_
 CORE  LEVEL  COMP_ID        TIMESTAMP            DELTA                FILE_NAME        CONTENT
    0      1   VOLUME   4605507.083333   4605507.000000             volume.c:463        xro
    0      1     PIPE   4605522.239583        15.156250           pipeline.c:903        ePU
    0      1  unknown   4605524.843750         2.604167           pipeline.c:904        value 4
    0      1     PIPE   4605527.760417         2.916667           pipeline.c:1198       pxr
_sudo sof-logger -l ./sof-apl.ldc-master-gcc-43da1696_
 CORE  LEVEL  COMP_ID        TIMESTAMP            DELTA                FILE_NAME        CONTENT
_sudo sof-logger -l ./sof-apl.ldc-master-gcc-43da1696_
 CORE  LEVEL  COMP_ID        TIMESTAMP            DELTA                FILE_NAME        CONTENT

_sudo hexdump -C /sys/kernel/debug/sof/etrace_
00000000  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00002000

**Test steps:**
1. aplay -Dhw:0,0 -r 48000 -c 2 -f s16_le -i -vv ~/Music/18_WAV_48000Hz_stereo_16bit.wav
2. sudo ./xrun_injection.sh
3. sudo sof-logger -l ./sof-apl.ldc-master-gcc-43da1696 # repeated several times

**Test env:**
sof master: 43da1696
sof tool: 937032f
kernel sof-dev: a5ddc3d7
tplg: sof-apl-pcm512x.tplg-937032f

**Logs:**
[dmesg-apl.txt](https://github.com/thesofproject/sof/files/2541420/dmesg-apl.txt)

lgirdwood, 2018-11-02:
@markyang please use the markdown for pasting logs. 

@keyonjie  @ranj063 This is caused by runtime PM. Everytime we reboot the DSP after D3 -> D0 transition FW will clear the SRAM (otherwise SRAM can contain garbage). Lets make this a future enhancement to preserve old error logs from previous boots.

xiulipan, 2018-11-02:
@lgirdwood 
In 37877f2b04dc8a31a8a08c5774c36fae74198984 from https://github.com/thesofproject/sof/pull/325
FW will skip the clear up for the memory window 3.

I think there are any other issues for this situation.

xiulipan, 2018-11-02:
@lgirdwood @keyonjie @ranj063 
Maybe we need to check the trace_init function.
```c
+static int ticount;
 void trace_init(struct sof *sof)
 {
        dma_trace_init_early(sof);

        trace = rzalloc(RZONE_SYS | RZONE_FLAG_UNCACHED, SOF_MEM_CAPS_RAM,
                        sizeof(*trace));
        trace->enable = 1;
        trace->pos = 0;
        spinlock_init(&trace->lock);

        bzero((void *)MAILBOX_TRACE_BASE, MAILBOX_TRACE_SIZE);
        dcache_writeback_invalidate_region((void *)MAILBOX_TRACE_BASE,
                                           MAILBOX_TRACE_SIZE);
+
+       trace_error(0, "trace_init %d", ticount++);
 }
```
The log is here:
```bash
pxl@pxl-UP2:~$ sudo ./sof-logger -l sof-apl.ldc
 CORE  LEVEL  COMP_ID        TIMESTAMP            DELTA                FILE_NAME        CONTENT
    0      1  unknown     61560.052083     61560.050781              trace.c:732        trace_init 0
pxl@pxl-UP2:~$ sudo ./sof-logger -l sof-apl.ldc
 CORE  LEVEL  COMP_ID        TIMESTAMP            DELTA                FILE_NAME        CONTENT
    0      1  unknown     61560.052083     61560.050781              trace.c:732        trace_init 0
pxl@pxl-UP2:~$ sudo ./sof-logger -l sof-apl.ldc
 CORE  LEVEL  COMP_ID        TIMESTAMP            DELTA                FILE_NAME        CONTENT
    0      1  unknown     63247.291667     63247.292969              trace.c:732        trace_init 0
pxl@pxl-UP2:~$ sudo ./sof-logger -l sof-apl.ldc
 CORE  LEVEL  COMP_ID        TIMESTAMP            DELTA                FILE_NAME        CONTENT
    0      1  unknown     63247.291667     63247.292969              trace.c:732        trace_init 0
pxl@pxl-UP2:~$ sudo ./sof-logger -l sof-apl.ldc
 CORE  LEVEL  COMP_ID        TIMESTAMP            DELTA                FILE_NAME        CONTENT
    0      1  unknown     65336.666667     65336.667969              trace.c:732        trace_init 0
pxl@pxl-UP2:~$ sudo ./sof-logger -l sof-apl.ldc
 CORE  LEVEL  COMP_ID        TIMESTAMP            DELTA                FILE_NAME        CONTENT
    0      1  unknown     64844.479167     64844.480469              trace.c:732        trace_init 0
```
From TIMESTAMP we can see that actually each time the `trace_init 0` log is generate for each resume.
And almost the whole DSP is robooted.
We may need some method to store the **trace->pos** in suspend and restore them when resume, if the data in memory window of SRAM is not lost or changed after entring D3

xiulipan, 2018-11-04:
@lgirdwood 
But if we used Linux trace framwork, how could we get this error trace with sof-logger.
Maybe add some kernel side mem copy for these memory window is more simple? We can append error trace or replace other regs maps to the kernel side memory.

xiulipan, 2018-11-06:
@lgirdwood 
And also maybe useful for dma-trace failure.
OK, we can workaround this by disable PM, when we really need to debug something like that.

keyonjie, 2018-11-06:
we are working on fix dma trace been cleared issue, but for etrace cleared issue, no plan yet.