thesofproject/sof #526: verbose tracing doesn't refresh

keqiaozhang, 2018-11-02:
sof-logger  -t doesn't refresh under one condition, when runtime is suspended, run sof-logger first and then do audio playback.

How to reproduce:
1. enable verbose in trace.h
2. make sure runtime is suspended
 cat /sys/bus/pci/devices/0000\:00\:0e.0/power/runtime_status
3. open the trace log on one terminal
  sof-logger -l sof-apl.ldc -t 
4. run audio in another terminal
speaker-test -Dhw:0,1 -c 2 -t wav
5. check the trace log output, no refresh.

But if we run audio first then open trace log, no such issue, trace log will continue to update.

There is no obvious log in kernel dmesg. logger output shows as below:
CORE  LEVEL  COMP_ID        TIMESTAMP            DELTA                FILE_NAME        CONTENT
    0      2     WAIT  12693428.697917  12693429.000000 ../../src/include/sof/wait.h:76         WFX
    0      2     PIPE  12693432.447917         3.750000           schedule.c:368        sch
    0      2     WAIT  12693436.666667         4.218750 ../../src/include/sof/wait.h:73         WFE
    0      2     WAIT  12694428.697917       992.031250 ../../src/include/sof/wait.h:76         WFX
    0      2     PIPE  12694432.604167         3.906250           schedule.c:368        sch
    0      2     WAIT  12694436.875000         4.270833 ../../src/include/sof/wait.h:73         WFE
    0      2     WAIT  12695428.697917       991.822937 ../../src/include/sof/wait.h:76         WFX
    0      2     PIPE  12695432.447917         3.750000           schedule.c:368        sch
    0      2     WAIT  12695436.822917         4.375000 ../../src/include/sof/wait.h:73         WFE
    0      2     WAIT  12696428.697917       991.875000 ../../src/include/sof/wait.h:76         WFX
[logger-t.log](https://github.com/thesofproject/sof/files/2540796/logger-t.log)

lgirdwood, 2018-11-02:
@keyonjie could be related to your recent fix for debufFS PM runtime ?

lgirdwood, 2018-11-02:
@keqiaozhang  please use markdown when pasting logs. See above changes I've made.

keqiaozhang, 2018-11-02:
@lgirdwood  thanks, Learned.

xiulipan, 2018-11-02:
@lgirdwood 
I think **NO**. https://github.com/thesofproject/linux/pull/237 is not merged yet.
And another reason is dma-trace used another file_ops as other debugfs.

https://github.com/thesofproject/linux/blob/a5ddc3d78b47b4d3e77ac371b6f016f3a6a9278c/sound/soc/sof/trace.c#L117
`	.read = sof_dfsentry_trace_read,`

https://github.com/thesofproject/linux/blob/a5ddc3d78b47b4d3e77ac371b6f016f3a6a9278c/sound/soc/sof/debug.c#L92
`	.read = sof_dfsentry_read,`

So maybe we should goto `sof_dfsentry_trace_read` function to have a look.

mengdonglin, 2018-11-08:
@keyonjie Is any of your recent PR will fix this issue?

keyonjie, 2018-11-08:
@mengdonglin yes, PR https://github.com/thesofproject/linux/pull/248 should fix this.

keyonjie, 2018-11-20:
thesofproject/linux#287 merged to sof-dev, @ranj063 please backport it to cros branch then we can verify and close this issue on GLK.

keqiaozhang, 2018-11-27:
Confirmed that this issue is fixed.
Tested with:
kernel-cros-sof-v4.14-rebase: f8f388ef
sof-master: 443233
tplg-master: ad3f3e4

closed 2018-11-27 by keqiaozhang