thesofproject/sof #531: clk: Null dereference reported by Coverity

plbossart, 2018-11-02:
Coverity reports an Explicit null dereference in clock_set_freq()

path:
const struct freq_table *freq_table = NULL;
switch (clock) {
default: break;
}
idx = clock_get_freq(freq_table, freq_table_size, hz);

The default case should be an error?

tlauda, 2018-11-21:
@plbossart Can you rerun Coverity to confirm it's fixed with #598?

michalgrodzicki, 2019-01-28:
@plbossart please verify and/or close it.

plbossart, 2019-01-28:
On 1/28/19 2:32 AM, michalgrodzicki wrote:
>
> @plbossart <https://github.com/plbossart> please verify and/or close it.
>
Sorry, I've asked CI to support Coverity for a while now, and I am not 
going to keep doing this verification if it can be automated.

michalgrodzicki, 2019-01-28:
@plbossart - who promissed you add Coverity to the CI?
I'm askingbecause this person should be owner of the issue

michalgrodzicki, 2019-01-31:
Hi @xiulipan, did you have a chance to check issue by the Coverity scan?

tlauda, 2019-03-05:
@xiulipan Any updates?

tlauda, 2019-06-06:
@xiulipan Fixed with #598.

closed 2019-06-06 by tlauda