thesofproject/sof #533: dw-dma: Using uninitialized value (Coverity)

plbossart, 2018-11-02:
static uint64_t dw_dma_work(void *data, uint64_t delay)
1063{
1064        struct dma_id *dma_id = (struct dma_id *)data;
1065        struct dma *dma = dma_id->dma;
1066        struct dma_pdata *p = dma_get_drvdata(dma);
    	1. var_decl: Declaring variable next without initializer.
1067        struct dma_sg_elem next;
1068        int i = dma_id->channel;
1069
1070        tracev_dma("wrk");
1071
1072        /* skip if channel is not running */
    	2. Condition p->chan[i].status != 5, taking true branch.
1073        if (p->chan[i].status != COMP_STATE_ACTIVE) {
1074                trace_dma_error("eDs");
    	3. Jumping to label out.
1075                goto out;
1076        }
1077
1078        dw_dma_process_block(&p->chan[i], &next);
1079
1080out:
    	
CID 324979 (#1 of 1): Uninitialized scalar variable (UNINIT)
4. uninit_use: Using uninitialized value next.size.
1081        return next.size == DMA_RELOAD_END ? 0 : p->chan[i].timer_delay;

lgirdwood, 2018-11-09:
@libinyang please send PR with above coverity output in commit message.

libinyang, 2018-11-12:
> @libinyang please send PR with above coverity output in commit message.

Yes, I will. After checking with peers here, I found that there is no coverity in Shanghai. I can't do the coverity test on my patch. So I will send the patch directly.

libinyang, 2018-11-12:
patch is submitted:
https://github.com/thesofproject/sof/pull/565

Please help review.

libinyang, 2018-11-16:
Suppose this bug can be closed?

mengdonglin, 2018-11-28:
@plbossart Closing this issue. Please reopen if you still see the bug. Thanks!

closed 2018-11-28 by mengdonglin