thesofproject/sof #535: eq_fir: Array compared against 0 (Coverity)

plbossart, 2018-11-02:
/* Copy new config, find size from header */
   	
CID 324977 (#1 of 1): Array compared against 0 (NO_EFFECT)
array_null: Comparing an array to null is not useful: (*cdata).data->data, since the test will always evaluate as true.
   	Was (*cdata).data->data formerly declared as a pointer?
563                if (!cdata->data->data) {
564                        trace_eq_error("edn");
565                        return -EINVAL;
566                }

plbossart, 2018-11-05:
@singalsu not the PR doesn't make the warning go away. I don't quite get why we use these variable length array, what's wrong with old-fashioned pointers?

singalsu, 2019-02-05:
This issue was fixed by PR #582.

closed 2019-02-05 by singalsu