thesofproject/sof #536: Validate IPC input.

lgirdwood, 2018-11-02:
IPC input needs to be validated. i.e. negative stream_ids and other out of bounds variables are currently accepted.

marcinx-zielinski, 2019-05-09:
Do you mean stream ids like in this function from _handler.c_?
`static int ipc_stream_pcm_params(uint32_t stream)`
A struct field named stream_id exists only in hda-dma.c as a declaration and appears to be unused, so please clarify.

lgirdwood, 2019-05-09:
@marcinx-zielinski I mean anything that can be validated must be validated.

jajanusz, 2019-06-06:
This issue is about sanity checks in FW code. Not about checking what we can remove from the code, or about fuzzer app that is going to tests if invalid ipcs gonna break FW.

marcinx-zielinski, 2019-06-06:
I know. I just referred to the example given by lgirdwood. We can't do sanity checks on dead code, that would be pointless. I understand that the task is just to put the sanity checks everywhere it is possible and refers to working code.

lgirdwood, 2019-06-06:
@marcinx-zielinski sanity checks should be for any data passed in externally in from host. handler.c can check the headers (since they are common), but the feature specific IPC should be checked at the next level (in the feature specific code). We can't check everything, but obvious limits, types, sizes can be validated.

lgirdwood, 2022-02-11:
@cujomalainey it should be since it's IPC data and coming from the host or fuzzer.
@ranj063 do you see any gaps here with fuzzer ? are we fuzzing all IPC messages ?

cujomalainey, 2022-02-11:
There is still the gap of fuzzing all the way down to the component. We need to remove the whole dll system in the testbench to make that work. It will be a pain because then we have to have a linker script for the host build so we can dynamically register components like we do on a real build. This would also allow us to use xt-run on the testbench since it doesn't support dynamic libs either.

lgirdwood, 2023-07-04:
> With the new fuzzer @andyross added I think this is done

Ack, will close this and we can respond to fuzzer issues on case by case basis.

closed 2023-07-04 by lgirdwood