thesofproject/sof #539: DSP panic after system booted for topology load issue on up2

ZhendanYang, 2018-11-05:
Summary:
DSP panic happened after system booted for topology load issue. This issue is separated from #443 .

```
[    5.736618] sof-audio sof-audio: tplg: pipeline id 8 comp 47 scheduling comp id 46
[    5.736622] sof-audio sof-audio: pipeline PIPELINE.8.SSP3.IN: deadline 1000 pri 0 mips 5000 core 0 frames 48
[    5.736637] sof-audio sof-audio: ipc tx: 0x30100000
[    5.736743] sof-audio sof-audio: error : DSP panic!
[    5.736752] sof-audio sof-audio: panic: dsp_oops_offset 788480 offset 788480
[    5.736756] sof-audio sof-audio: status: fw entered - code 00000005
[    5.736778] sof-audio sof-audio: error: runtime exception
[    5.736780] sof-audio sof-audio: error: trace point 00004000
[    5.736782] sof-audio sof-audio: error: panic happen at :0
[    5.736785] sof-audio sof-audio: error: DSP Firmware Oops
[    5.736788] sof-audio sof-audio: error: Exception Cause: LoadProhibitedCause, A load referenced a page mapped with an attribute that does not permit loads
[    5.736791] sof-audio sof-audio: EXCCAUSE 0x0000001c EXCVADDR 0xfffffff8 PS       0x00060020 SAR     0x0000001d
[    5.736794] sof-audio sof-audio: EPC1     0xbe015a93 EPC2     0xbe0118f6 EPC3     0xbe013a48 EPC4    0x00000000
[    5.736796] sof-audio sof-audio: EPC5     0x00000000 EPC6     0x00000000 EPC7     0x00000000 DEPC    0x00000000
[    5.736799] sof-audio sof-audio: EPS2     0x00060720 EPS3     0x00060722 EPS4     0x00000000 EPS5    0x00000000
[    5.736802] sof-audio sof-audio: EPS6     0x00000000 EPS7     0x00000000 INTENABL 0x000101d0 INTERRU 0x00000222
[    5.736804] sof-audio sof-audio: stack dump from 0xbe07fd20
[    5.736807] sof-audio sof-audio: 0xbe07fd20: 0xbe015a93 0x00060d30 0x0000001d 0x00000180
[    5.736810] sof-audio sof-audio: 0xbe07fd24: 0xbe04db80 0x0000002f 0x9e050368 0x9e050368
[    5.736812] sof-audio sof-audio: 0xbe07fd28: 0x0000001c 0xbe015f70 0xbe015a90 0xbe015aae
[    5.736815] sof-audio sof-audio: 0xbe07fd2c: 0x00000002 0x00060d20 0xbe042048 0x20004eb8
[    5.736818] sof-audio sof-audio: 0xbe07fd30: 0x00000000 0x001d72fc 0x00000000 0xbe05b400
[    5.736821] sof-audio sof-audio: 0xbe07fd34: 0x00000008 0xbe07fdc0 0x07530000 0x00000000
[    5.736824] sof-audio sof-audio: 0xbe07fd38: 0xffffffff 0xbe014117 0xbe0141ac 0xbe0141b9
[    5.736826] sof-audio sof-audio: 0xbe07fd3c: 0x00000000 0x001d72fc 0x00000000 0x2000126c
[    5.736829] sof-audio sof-audio: error: waking up any trace sleepers
[    5.778036] random: crng init done
[    5.778041] random: 7 urandom warning(s) missed due to ratelimiting
[    6.048106] sof-audio sof-audio: error: ipc timed out for 0x30100000 size 0x30
```
```
be015a93: 7c1852 l16ui a5, a8, 248
case COMP_TYPE_COMPONENT:
if (icd->cd->comp.id == id)
```

Test env:
sof-master: aeb18ee
sof tool: 2d7a933
kernel sof-dev: 93853cb
tplg: sof-apl-nocodec.tplg

Log:
rmbox and rmbox -t are both empty.
[dmesg-apl.log](https://github.com/thesofproject/sof/files/2546785/dmesg-apl.log)

Another Exception Cause on UP2:

```
[ 6.132460] sof-audio sof-audio: ipc tx: 0x80010000
[ 6.132735] sof-audio sof-audio: error : DSP panic!
[ 6.132742] sof-audio sof-audio: panic: dsp_oops_offset 788480 offset 788480
[ 6.132745] sof-audio sof-audio: status: fw entered - code 00000005
[ 6.132766] sof-audio sof-audio: error: runtime exception
[ 6.132769] sof-audio sof-audio: error: trace point 00004000
[ 6.132771] sof-audio sof-audio: error: panic happen at :0
[ 6.132774] sof-audio sof-audio: error: DSP Firmware Oops
[ 6.132776] sof-audio sof-audio: error: Exception Cause: LoadStorePIFDataErrorCause, Synchronous PIF data error during LoadStore access
[ 6.132779] sof-audio sof-audio: EXCCAUSE 0x0000000d EXCVADDR 0x0000039c PS 0x00060020 SAR 0x00000001
[ 6.132782] sof-audio sof-audio: EPC1 0xbe0166d7 EPC2 0xbe014f4e EPC3 0xbe013d48 EPC4 0x00000000
[ 6.132785] sof-audio sof-audio: EPC5 0x00000000 EPC6 0x00000000 EPC7 0x00000000 DEPC 0x00000000
[ 6.132787] sof-audio sof-audio: EPS2 0x00060d20 EPS3 0x00060722 EPS4 0x00000000 EPS5 0x00000000
[ 6.132790] sof-audio sof-audio: EPS6 0x00000000 EPS7 0x00000000 INTENABL 0x000101d0 INTERRU 0x00000222
```
                     /* make sure we only config DAI comps */
                     switch (icd->cd->comp.type) {
```
be0166d4: 3f29a2 l32i a10, a9, 252
be0166d7: 272a92 l32i a9, a10, 156
be0166da: 022926 beqi a9, 2, be0166e0 <ipc_comp_dai_config+0x44>
be0166dd: db4966 bnei a9, 4, be0166bc <ipc_comp_dai_config+0x20>
```

                             /*
                              * set config if comp dai_index matches
                              * config dai_index.
                              */
                             if (dai->dai_index == config->dai_index &&
Test env:
sof-master: 1024e94
soft-master: c0a881a
kernel sof-dev: 93853c
tplg: sof-apl-nocodec.tplg

libinyang, 2018-11-05:
It seems I found why the dcache flush randomly changes the memory incorrectly. I will make another patch for #527.

However, I do think this patch is useful to improve the performance and should be merged. What do you guys think? Thanks ;-)

libinyang, 2018-11-09:
@ZhendanYang 
This bug should be fixed with the latest SOF

markyang, 2018-11-12:
**Summary:**
  This issue cannot be reproduced on UP2 any more. 

**Test env:**
sof master: 5838f717
sof tool: 2cd668c
kernel sof-dev: 9b81e164
tplg: sof-apl-nocodec.tplg-2cd668c

closed 2018-11-12 by lgirdwood