thesofproject/sof #544: handle sof_ipc_host_buffer with non-0 offset

keyonjie, 2018-11-06:
On SKL- platforms, we are using sof_ipc_host_buffer for host buffer definition:
/* stream ring info */
struct sof_ipc_host_buffer {
	uint32_t phy_addr;
	uint32_t pages;
	uint32_t size;
	uint32_t offset;
} __attribute__((packed));

Here, the offset is created aimed to make the DMA start copying from/to non-0 offset of the buffer.
Previously, we all set offset=0 in host side, but now we are running into non-0 offset requirement support for runtime suspend/resume cases where we may requirement the DMA resume copying from the previous offset(suspend point), so, we have requirement on FW capability to handle this non-0 offset now.

The basic thinking is that to add this handling into ipc_parse_page_descriptors(), for non-0 offset, we need to organize elem_array[] like this:

offset_page = offset / page_size;
page_offset = offset % page_size;

elem_array[0].dest = page[offset_page] + page_offset;
elem_array[0].size = page_size - page_offset;

elem_array[1].dest = page[offset_page + 1];
elem_array[1].size = page_size;
...
elem_array[m].dest = page[pages - 1];
elem_array[m].size = page_size;

elem_array[m+1].dest = page[0];
elem_array[m+1].size = page_size;
...

elem_array[m+n].dest = page[n-1];
elem_array[m+n].size = page_size;
...
elem_array[pages-1].dest = page[offset_page - 1];
elem_array[pages-1].size = page_size;

elem_array[pages].dest = page[offset_page];
elem_array[pages].size = page_offset ;

with the elem_array[] set correctly, we should be able to handle these non-0 offset cases, these are useful for e.g. DMA trace resume, stream playback/capture resume.

tlauda, 2019-06-06:
@keyonjie @lgirdwood SKL+ platforms don't use this buffer at all. The right starting data should be put at the beginning of BDL on the host side, so the Host DMA Gateway gets the right samples. Gateway starts from the beginning after every reset and pointer cannot be explicitly controlled by the DSP.
Can we close this?

jajanusz, 2019-06-10:
We don't plan implementing this. Feel free to reopen if topic comes back.

closed 2019-06-10 by jajanusz