thesofproject/sof #550: enable memory debug will cause firmware panic

keqiaozhang, 2018-11-08:
```
[    6.621074] sof-audio sof-audio: Firmware download successful, booting...
[    6.621077] sof-audio sof-audio: firmware boot complete
[    6.621128] sof-audio sof-audio: generating page table for ffffa56dc2ee9000 size 0x10000 pages 16
[    6.621130] sof-audio sof-audio: dma_trace_pages: 16
[    6.621168] sof-audio sof-audio: period_bytes:0x0
[    6.621169] sof-audio sof-audio: periods:1
[    6.621178] sof-audio sof-audio: stream_tag: 1
[    6.621196] sof-audio sof-audio: ipc tx: 0x90010000
[    6.621289] sof-audio sof-audio: error : DSP panic!
[    6.621294] sof-audio sof-audio: panic: dsp_oops_offset 788480 offset 788480
[    6.621300] sof-audio sof-audio: status: fw entered - code 00000005
[    6.621447] sof-audio sof-audio: error: runtime exception
[    6.621450] sof-audio sof-audio: error: trace point 00004000
[    6.621453] sof-audio sof-audio: error: panic happen at :0
[    6.621457] sof-audio sof-audio: error: DSP Firmware Oops
[    6.621461] sof-audio sof-audio: EXCCAUSE 0x0000001d EXCVADDR 0x296beb6b PS       0x00060025 SAR     0x0000001d
[    6.621464] sof-audio sof-audio: EPC1     0xbe013b90 EPC2     0xbe01192e EPC3     0xbe014840 EPC4    0x00000000
[    6.621468] sof-audio sof-audio: EPC5     0x00000000 EPC6     0x00000000 EPC7     0x00000000 DEPC    0x00000000
[    6.621471] sof-audio sof-audio: EPS2     0x00060720 EPS3     0x00060722 EPS4     0x00000000 EPS5    0x00000000
[    6.621474] sof-audio sof-audio: EPS6     0x00000000 EPS7     0x00000000 INTENABL 0x000001d0 INTERRU 0x00000222
[    6.621477] sof-audio sof-audio: stack dump from 0xbe07fbd0
[    6.621481] sof-audio sof-audio: 0xbe07fbd0: 0xbe013b90 0x00060f35 0x0000001d 0x00000000
[    6.621484] sof-audio sof-audio: 0xbe07fbd4: 0x296beb6b 0xffffffff 0xffffffff 0x2000041c
[    6.621488] sof-audio sof-audio: 0xbe07fbd8: 0x0000001d 0x00000000 0xbe0281f5 0xbe0281fb
[    6.621491] sof-audio sof-audio: 0xbe07fbdc: 0x00000000 0x00000000 0x00000000 0x00000000
[    6.621494] sof-audio sof-audio: 0xbe07fbe0: 0x00000000 0x00000000 0x00000000 0x00000000
[    6.621497] sof-audio sof-audio: 0xbe07fbe4: 0x00000000 0x00000000 0x00000000 0x00000000
[    6.621500] sof-audio sof-audio: 0xbe07fbe8: 0x00ffffff 0x001a9f24 0x00000000 0x2000041c
[    6.621504] sof-audio sof-audio: 0xbe07fbec: 0x00000000 0x00000000 0x00000000 0x00000000
[    6.928041] sof-audio sof-audio: error: ipc timed out for 0x90010000 size 0x1c
[    6.928055] sof-audio sof-audio: status: fw entered - code 00000005
[    6.928202] sof-audio sof-audio: error: runtime exception
[    6.928206] sof-audio sof-audio: error: trace point 00004000
[    6.928208] sof-audio sof-audio: error: panic happen at :0
[    6.928212] sof-audio sof-audio: error: DSP Firmware Oops
```

libinyang, 2018-11-08:
The bug is caused by the dcache. I will submit a patch later.

dabekjakub, 2018-11-16:
It is hard to debug since we can't recover exception memory as a binary - will try to work around it but if issue https://github.com/thesofproject/linux/issues/296 from kernel got fixed it would be a lot smoother.

michalgrodzicki, 2018-12-07:
@keqiaozhang @dabekjakub what is a status?

dabekjakub, 2018-12-07:
I am currently working on it. There were some problems along the way resulting in fix of a memory leak.
https://github.com/thesofproject/sof/pull/673
Currently i am resolving issue with filling pattern initially on whole memory map. It crashes on all platforms.

lgirdwood, 2018-12-19:
@keqiaozhang can you recheck this as #738 is merged.

keqiaozhang, 2018-12-24:
@markyang 
Could you help to check this issue on APL?

markyang, 2018-12-24:
**Summary:**
  This issue cannot be reproduced on APL with PCM512x.

**dmesg:**
```
[    3.970872] sof-audio sof-audio: Firmware info: version 1:1:0-27c09
[    3.970874] sof-audio sof-audio: Firmware: ABI 3:0:0 Kernel ABI 3:0:0
[    3.970979] sof-audio sof-audio: found ext header type 1 size 0xb8
[    3.971013] sof-audio sof-audio:  mailbox upstream 0x81000 - size 0x1000
[    3.971014] sof-audio sof-audio:  mailbox downstream 0xa0000 - size 0x2000
[    3.971016] sof-audio sof-audio:  stream region 0xc1000 - size 0x1000
[    3.971017] sof-audio sof-audio: booting DSP firmware completed
[    3.971019] sof-audio sof-audio: ipc rx done: 0x70000000: FW_READY
[    3.972625] sof-audio sof-audio: Firmware download successful, booting...
[    3.972629] sof-audio sof-audio: firmware boot complete
[    3.972636] sof-audio sof-audio: Debug PCIR: 00000000 at  00000048
[    3.972639] sof-audio sof-audio: Debug PCIW: 00000002 at  00000048
[    3.972647] sof-audio sof-audio: Debug PCIR: 00000004 at  00000044
[    3.972652] sof-audio sof-audio: Debug PCIW: 00000000 at  00000044
[    3.972698] sof-audio sof-audio: generating page table for 00000000be4a2e23 size 0x10000 pages 16
[    3.972700] sof-audio sof-audio: dma_trace_pages: 16
[    3.972727] sof-audio sof-audio: period_bytes:0x0
[    3.972728] sof-audio sof-audio: periods:1
[    3.972734] sof-audio sof-audio: stream_tag: 1
[    3.972755] sof-audio sof-audio: ipc tx: 0x90010000: GLB_TRACE_MSG
[    3.973640] sof-audio sof-audio: ipc tx succeeded: 0x90010000: GLB_TRACE_MSG
[    3.973749] sof-audio sof-audio: created machine bxt-pcm512x
```

**Test steps:**
enable the memory debug in sof:
```
--- a/src/lib/alloc.c
+++ b/src/lib/alloc.c
@@ -40,7 +40,7 @@
 #include <stdint.h>
 /* debug to set memory value on every allocation */
-#define DEBUG_BLOCK_FREE 0
+#define DEBUG_BLOCK_FREE 1
```
2. build and update sof
3. reboot device
4. check dmesg

**Test env:**
sof master: 27c09a94
kernel sof-dev: 695380ff
tplg: sof-apl-pcm512x.tplg

closed 2018-12-24 by markyang