thesofproject/sof #551: sof-logger doesn

keqiaozhang, 2018-11-08:


akloniex, 2018-11-08:
@lgirdwood @bkokoszx @keqiaozhang
Issue is caused by misalignment between FW and sof-logger.
Expected to be solved as soon as https://github.com/thesofproject/soft/pull/118 is merged and sof-logger rebuilt.

mengdonglin, 2018-11-08:
@akloniex @lrgirdwo May we request any PR/commit with ABI change add keyword "ABI" in the subject to highlight the risk?  And this PR should mention the soft PR in the body. Ideally, the SOFT update should be backward compatible and be merged at first before the SOF PR.

ZhendanYang, 2018-11-09:
sof-logger doesn't work even without -t on CNL:
```
test@test-CannonLake:~$ sudo ./sof-logger -l sof-cnl.ldc-11.9
 CORE  LEVEL  COMP_ID        TIMESTAMP            DELTA                FILE_NAME	CONTENT
Error: Invalid filename length. 
test@test-CannonLake:~$ sudo ./sof-logger -l sof-cnl.ldc-11.9 -t
 CORE  LEVEL  COMP_ID        TIMESTAMP            DELTA                FILE_NAME	CONTENT
Error: Invalid filename length. 
```

mengdonglin, 2018-11-09:
@bkokoszx  Are all the PR low necessary to fix this issue? Anything missing?

SOFT: https://github.com/thesofproject/soft/pull/118
FW part: https://github.com/thesofproject/sof/pull/530
Linux part: https://github.com/thesofproject/linux/pull/222

bkokoszx, 2018-11-09:
@mengdonglin 
Also FW PR I've made yesterday: https://github.com/thesofproject/sof/pull/548

bkokoszx, 2018-11-09:
@keqiaozhang 
Could you verify if issue still reproduces?

keqiaozhang, 2018-11-13:
@bkokoszx 
Confirmed that this issue is fixed on UP^2.
For Yorp,  there's one blocker issue with top sof-master :https://github.com/thesofproject/linux/issues/277
so there's no trace log available, but the "Error: Invalid filename length." is gone. 
@markyang 
please help to confirm this issue on other platform. Thanks~

markyang, 2018-11-13:
@bkokoszx @keqiaozhang 
This issue has been fixed on BYT, APL and CNL.

markyang, 2018-11-13:
There's no trace log available on BYT because of a blocker issue thesofproject/linux#233

keqiaozhang, 2018-11-13:
Thanks @markyang for your confirmation, this issue can be closed now.

closed 2018-11-13 by keqiaozhang