thesofproject/sof #552: White noise when doing playback/record via I2S on ICL

DarrenxWang, 2018-11-08:
Summary:
White noise when use i2s playback/record on firware branch icl-001-drop-stable. 

It seems to be a branch-specific firmware issue for icl-001-drop-stable.
Because with the same kernel & topology, this issue cannot be reproduced on firmware master branch (commit 33e4b99351) plus the patch below 0001-DMIC-Enable-power-for-DMIC-for-ICL-platform.patch

Test steps:
aplay -D hw:0,0 Music/48California_Dreaming.wav -d 20   & arecord -Dhw:0,0 -f dat -c 2 test.wav -d 20

Test env:
Sof- icl-001-drop-stable: 40c054ab3087466

Soft-Soft- master: 937032fb5f20 + 1 Patch 
[0001-topology-sample-code-for-icl-topology.patch.txt](https://github.com/thesofproject/sof/files/2560788/0001-topology-sample-code-for-icl-topology.patch.txt)

Linux kernel: topic/sof-dev: af44aa9b + 3 Kernel patches
[0001-ASoC-Intel-hdac_hdmi-add-Icelake-support.patch.txt](https://github.com/thesofproject/sof/files/2560786/0001-ASoC-Intel-hdac_hdmi-add-Icelake-support.patch.txt)
[0002-ASoC-intel-add-icelake-dummy-codec-match-table.patch.txt](https://github.com/thesofproject/sof/files/2560789/0002-ASoC-intel-add-icelake-dummy-codec-match-table.patch.txt)
[0003-ASoC-Intel-add-ICL-machine-driver.patch.txt](https://github.com/thesofproject/sof/files/2560790/0003-ASoC-Intel-add-ICL-machine-driver.patch.txt)

Firmware patch：
[0001-DMIC-Enable-power-for-DMIC-for-ICL-platform.patch.txt](https://github.com/thesofproject/sof/files/2560777/0001-DMIC-Enable-power-for-DMIC-for-ICL-platform.patch.txt)

Log&Wav:
There is no obvious error in kmesg.
[logger.txt](https://github.com/thesofproject/sof/files/2560771/logger.txt)
[wav.zip](https://github.com/thesofproject/sof/files/2560767/wav.zip)

mrajwa, 2018-11-08:
The only significant difference between the commit that is fine 33e4b99 and the icl-001-drop-stable is this commit 4fa9a7b386. @singalsu, could you please check if you see reported noise issue caused by your commit changes?

singalsu, 2018-11-08:
@mrajwa I don't have ICL. Do you know which bits -> bits conversion is the suspected volume controller doing?

singalsu, 2018-11-08:
I made just PR #555. That's an obvious error, sorry for that. If the playback / capture pipelines in  ICL did s24 -> s16 sample format conversion in volume that should impact.

DarrenxWang, 2018-11-09:
There is white noise when doing 16 bit playback/record by I2S. 
There is no white noise when doing 24/32 bit playback/record by I2S.

singalsu, 2018-11-09:
@DarrenxWang I don't have device to test this conversion. Can you please test if PR #555 fixes this.

bardliao, 2018-11-09:
@mrajwa @singalsu This issue is fixed by PR #555 . Thanks

singalsu, 2018-11-09:
@bardliao Excellent, thanks!

singalsu, 2018-11-09:
Closing this issue.

closed 2018-11-09 by singalsu