thesofproject/sof #553: debugability: macros breaking multicore builds

mwierzbix, 2018-11-08:
Building multicore (for instance with `--with-arch=xtensa-smp `), we get following error mesasge
`sof section `.text' will not fit in region `sof_text'`

mwierzbix, 2018-11-08:
It would maybe be good if Travis had both single- and mult- core builds?

lgirdwood, 2018-11-08:
@mwierzbix already given this task to @xiulipan , but this makes it official.

mwierzbix, 2018-11-08:
I was just about to PR a workaround, and with both @mmaka1 and @akloniex to investigate.

lgirdwood, 2018-11-08:
I guess you will be able to fix first since it's evening time now for @xiulipan

mwierzbix, 2018-11-08:
Temporary fix would be to add `0x800` to `SOF_TEST_SIZE` to allow other developers to work while we work this issue out.

And if we don't manage, then maybe @xiulipan will :)

lgirdwood, 2018-11-08:
Sorry, I was meaning have @xiulipan fix Travis CI for SMP builds, not for memory sizes

mwierzbix, 2018-11-08:
Oh, well then he would discover that they fail because of this :P

mwierzbix, 2018-11-09:
Status update: workaround is already merged to master, @akloniex has a fix PRed.

closed 2018-11-09 by mwierzbix