thesofproject/sof #560: cAVS SSP MCLK handling refactor

plbossart, 2018-11-09:
The current MCLK handling code is duplicated multiple times, when the only difference is the value of the MCLK. We should get this information from a platform definition instead of hard-coding a define and using ifdefs

see https://github.com/thesofproject/sof/blob/805bcb10729bb7ce117d42942259ca8002429192/src/drivers/intel/cavs/ssp.c#L324

lgirdwood, 2018-11-09:
@mmaka1 I guess you were already thinking about de-duplicating this ?

lbetlej, 2019-03-27:
@bkokoszx take a look at ../cavs/ssp.c for MCLK handling. This should be `CAVS_VERSION` dependent.

jajanusz, 2019-03-27:
@lbetlej 
Imho it should be more like @plbossart wrote. Platform generic code that just takes values from platform/ip(for example cavs) defs.

bkokoszx, 2019-04-08:
@lgirdwood @plbossart 
I think this issue can be closed since #1197 has been merged.

closed 2019-04-08 by tlauda