thesofproject/sof #563: Topology component load failed

markyang, 2018-11-12:
**Summary:**
  Topology component load failed on UP2. 
  sof-logger is empty. sof-logger -t log is attached below.

**dmesg:**
```
[    6.270275] sof-audio sof-audio: ipc tx: 0x30030000: GLB_TPLG_MSG: COMP_CONNECT
[    6.274068] sof-audio sof-audio: ipc tx succeeded: 0x30030000: GLB_TPLG_MSG: COMP_CONNECT
[    6.274105] sof-audio sof-audio: sink HDA2.OUT control none source BUF4.1
[    6.274117] sof-audio sof-audio: ipc tx: 0x30030000: GLB_TPLG_MSG: COMP_CONNECT
[    6.275307] sof-audio sof-audio: ipc tx succeeded: 0x30030000: GLB_TPLG_MSG: COMP_CONNECT
[    6.275392] sof-audio sof-audio: tplg: config SSP5 fmt 0x4001 mclk 24576000 bclk 3072000 fclk 48000 width (24)32 slots 2 mclk id 0
[    6.275412] sof-audio sof-audio: ipc tx: 0x80010000: GLB_DAI_MSG: CONFIG
[    6.275562] sof-audio sof-audio: ipc tx succeeded: 0x80010000: GLB_DAI_MSG: CONFIG
[    6.275571] sof-audio sof-audio: ASoC: physical link iDisp1 (id 1) not exist
[    6.275580] sof-audio sof-audio: error: tplg component load failed -22
[    6.275600] sof-audio sof-audio: error: failed to load DSP topology -22
[    6.275604] sof-audio sof-audio: ASoC: failed to probe component -22
[    6.275621] bxt-pcm512x bxt-pcm512x: ASoC: failed to instantiate card -22
[    6.275627] bxt-pcm512x bxt-pcm512x: Topology forgot to remove link Port5?
[    6.275629] bxt-pcm512x bxt-pcm512x: Topology forgot to remove link HDMI1?
```

**Test steps:**
1. aplay -l
aplay: device_list:270: no soundcards found...

**Test env:**
sof master: 5838f717
sof tool: 2cd668c
kernel sof-dev: 9b81e164
tplg: sof-apl-pcm512x.tplg-2cd668c

**Log:**
[dmesg-apl.log](https://github.com/thesofproject/sof/files/2571280/dmesg-apl.log)
[sof-logger-t.log](https://github.com/thesofproject/sof/files/2571281/sof-logger-t.log)

lgirdwood, 2018-11-12:
@RanderWang are we missing iDISP BE DAI link or something in the Kconfig ?

xiulipan, 2018-11-13:
@markyang 
It seems the kconfig is miss some HDA HDMI options

markyang, 2018-11-13:
**Summary:**
  This issue cannot be reproduced on UP2 any more.

**Test env:**
sof master: 73475e3f
sof tool: 2cd668c
kernel sof-dev: 165b34de
tplg: sof-apl-pcm512x.tplg-2cd668c

closed 2018-11-13 by markyang