thesofproject/sof #571: ipc: occasionally see a dropped msg from host on UP^2

lgirdwood, 2018-11-13:
I occasionally see a dropped message from the host in the DSP logs on APL.
```
    0      2          IPC            131637.291667        85.520836            apl-ipc.c:109 	ipc: processing cmd 0x30030000
    0      2          IPC            131640.937500         3.645833            handler.c:933 	ipc_glb_tplg_comp_connect()
    0      2         PIPE 0.0        131681.562500        40.625000           pipeline.c:318 	pipeline_comp_connect()
    0      2          IPC            131762.500000        80.937500            apl-ipc.c:109 	ipc: processing cmd 0x30030000
    0      2          IPC            131766.145833         3.645833            handler.c:933 	ipc_glb_tplg_comp_connect()
    0      2         PIPE 384.0      131858.125000        91.979164           pipeline.c:341 	pipeline_buffer_connect()
    0      2          IPC            131865.000000         6.875000            apl-ipc.c:167 	ipc: sending msg 0x90020000
    0      1          IPC            131933.645833        68.645836            apl-ipc.c:80  	ipc: dropping msg 0x30030000
    0      2          IPC            131940.572917         6.927083            apl-ipc.c:109 	ipc: processing cmd 0x30030000
    0      2          IPC            131944.427083         3.854167            handler.c:933 	ipc_glb_tplg_comp_connect()
    0      2         PIPE 0.0        131988.645833        44.218750           pipeline.c:318 	pipeline_comp_connect()
    0      2          IPC            132056.979167        68.333336            apl-ipc.c:109 	ipc: processing cmd 0x30030000
    0      2          IPC            132060.833333         3.854167            handler.c:933 	ipc_glb_tplg_comp_connect()
    0      2         PIPE 768.0      132105.416667        44.583332           pipeline.c:341 	pipeline_buffer_connect()
```

There is no error log in the host side as it looks like the message is only processed correctly as it shares the same message type as the next or previous message. @ZhendanYang can someone from QA try and reproduce. @RanderWang fyi

xiulipan, 2018-11-14:
```
    0      2          IPC            131865.000000         6.875000            apl-ipc.c:167 	ipc: sending msg 0x90020000
    0      1          IPC            131933.645833        68.645836            apl-ipc.c:80  	ipc: dropping msg 0x30030000
```

@lgirdwood 
This is no issue, please notice the 0x90020000 above.
```
/* test code to check working IRQ */
static void irq_handler(void *arg)
{

	tracev_ipc("ipc: irq dipct 0x%x dipcie 0x%x", dipct, dipcie);

	/* new message from host */
	if (dipct & IPC_DIPCT_BUSY) {
...

		/* TODO: place message in Q and process later */
		/* It's not Q ATM, may overwrite */
		if (_ipc->host_pending) {
			trace_ipc_error("ipc: dropping msg 0x%x", msg);
			trace_ipc_error(" dipct 0x%x dipcie 0x%x dipcctl 0x%x",
					dipct, dipcie, ipc_read(IPC_DIPCCTL));
		} else {
			_ipc->host_msg = msg;
			_ipc->host_pending = 1;
			ipc_schedule_process(_ipc);
		}
	}

	/* reply message(done) from host */
	if (dipcie & IPC_DIPCIE_DONE) {
...
	}

}
```

We will get another IRQ from the DSP->HOST IPC **0x90020000** _reply message(done) from host_.
But the  **0x30030000** is not finished, and still in the queue.

We have used queue to avoid conflict HOST->DSP IPC

lgirdwood, 2018-11-14:
@xiulipan there is no Q incoming today on FW side - all messages should be serialised using HW flow control - check with @RanderWang for details.

xiulipan, 2018-11-14:
@lgirdwood 
Sorry not a **queue**, but a scheduled task.
The error message is triggered by a DSP->HOST message  **0x90020000**'s **reply** when the last HOST->DSP message **0x30030000** is not finished and `_ipc->host_pending` still be 1

lgirdwood, 2018-11-14:
@xiulipan so are you saying the trace message from DSP -> HOST is causing this host to send a new topology message ? Looks like a kernel bug then.

lgirdwood, 2018-11-14:
two more
```
 0      2       BUFFER            120777.916667        19.739584             buffer.c:53  	new
    0      2          IPC            120896.822917       118.906250            apl-ipc.c:111 	ipc: msg rx -> 0x30010000
    0      2          IPC            120901.718750         4.895833            handler.c:829 	ipc: pipe 10 comp 58 -> new (type 2)
    0      2          DAI            120922.968750        21.250000                dai.c:177 	dai_new()
    0      1          IPC            120937.343750        14.375000            apl-ipc.c:80  	ipc: dropping msg 0x30010000
    0      1          IPC            120942.864583         5.520833            apl-ipc.c:82  	 dipct 0xb0010000 dipcie 0x40000000 dipcctl 0x2
    0      2          DAI            120951.979167         9.114583                dai.c:86  	dai-get 0xbe047444 sref 2
    0      2          DMA            120955.937500         3.958333                dma.c:116 	dma-get 0xbe02e958 sref 10
    0      2          IPC            121061.510417       105.572914            apl-ipc.c:111 	ipc: msg rx -> 0x30100000
```

and

```
  0      2         PIPE 384.0      125798.750000        22.395834           pipeline.c:341 	pipeline_buffer_connect()
    0      2          IPC            125879.114583        80.364586            apl-ipc.c:111 	ipc: msg rx -> 0x30030000
    0      2          IPC            125883.593750         4.479167            handler.c:920 	ipc: comp sink 32, source 31  -> connect
    0      2         PIPE 0.0        125906.406250        22.812500           pipeline.c:318 	pipeline_comp_connect()
    0      1          IPC            125986.041667        79.635414            apl-ipc.c:80  	ipc: dropping msg 0x30030000
    0      1          IPC            125991.354167         5.312500            apl-ipc.c:82  	 dipct 0xb0030000 dipcie 0x40000000 dipcctl 0x2
    0      2          IPC            125998.854167         7.500000            apl-ipc.c:111 	ipc: msg rx -> 0x30030000
    0      2          IPC            126002.760417         3.906250            handler.c:920 	ipc: comp sink 31, source 33  -> connect
    0      2         PIPE 768.0      126025.677083        22.916666           pipeline.c:341 	pipeline_buffer_connect()
```

xiulipan, 2018-11-15:
@lgirdwood

I mean `static void irq_handler(void *arg) ` will enter in two situation.
If there are only HOST->DSP msg, no such issues will happen
But we have DSP->HOST reply msg will also trigger `irq_handler` function, if the HOST->DSP msg is not finished, it will look like we drop it. But in fact, we are **handling** it right now.

lgirdwood, 2018-11-15:
@xiulipan yes I think we always handle it (since I never see topology errors)  but we are erroneously reporting an error.  The Mask should also be checked i.e.
```diff
diff --git a/src/ipc/apl-ipc.c b/src/ipc/apl-ipc.c
index 79eb8da5..12d0fa8a 100644
--- a/src/ipc/apl-ipc.c
+++ b/src/ipc/apl-ipc.c
@@ -59,18 +59,20 @@ static void irq_handler(void *arg)
 {
        uint32_t dipct;
        uint32_t dipcie;
+       uint32_t dipcctl;
        uint32_t msg = 0;
 
        dipct = ipc_read(IPC_DIPCT);
        dipcie = ipc_read(IPC_DIPCIE);
+       dipcctl = ipc_read(IPC_DIPCCTL);
 
        tracev_ipc("ipc: irq dipct 0x%x dipcie 0x%x", dipct, dipcie);
 
        /* new message from host */
-       if (dipct & IPC_DIPCT_BUSY) {
+       if (dipct & IPC_DIPCT_BUSY && dipcctl & IPC_DIPCCTL_IPCTBIE) {
 
                /* mask Busy interrupt */
-               ipc_write(IPC_DIPCCTL, ipc_read(IPC_DIPCCTL) & ~IPC_DIPCCTL_IPCTBIE);
+               ipc_write(IPC_DIPCCTL, dipcctl & ~IPC_DIPCCTL_IPCTBIE);
 
                msg = dipct & IPC_DIPCT_MSG_MASK;
```

xiulipan, 2018-11-15:
@lgirdwood 
I think this patch should solve the problem.

libinyang, 2018-11-16:
@xiulipan 

Maybe I'm wrong as I'm not familiar with the hw setting. Please correct me:
with 'if (dipct & IPC_DIPCT_BUSY && dipcctl & IPC_DIPCCTL_IPCTBIE)',
we just skip the hw interrupt and don't logout the dropping message. But actually, we still drop the IPC without handling the new IPC? Actually we do accept a new IPC as dipct is IPC_DIPCT_BUSY. And in your code, you skip this as IPC_DIPCCTL_IPCTBIE is not set?

xiulipan, 2018-11-16:
@libinyang 
notice `IPC_DIPCCTL_IPCTBIE`, when we are handling an IPC, this enable bit will be unmask.
That `IPC_DIPCT_BUSY ` will no longer trigger IRQ here, but it is still set and not cleaned here.
There have `IPC_DIPCIE_DONE` BIT that can also trigger this IRQ. So check `IPC_DIPCCTL_IPCTBIE` to avoid make sure we handle the right IRQ triggered by the right BIT.

libinyang, 2018-11-16:
> @libinyang
> notice `IPC_DIPCCTL_IPCTBIE`, when we are handling an IPC, this enable bit will be unmask.
> That `IPC_DIPCT_BUSY ` will no longer trigger IRQ here, but it is still set and not cleaned [...]

Do you mean that the second IRQ is triggered by IPC_DIPCIE_DONE?

xiulipan, 2018-11-16:
@libinyang 
Yep, the one with dropping message is triggered by `IPC_DIPCIE_DONE`

mengdonglin, 2018-11-22:
@lgirdwood  Closing this issue since the fixing is merged in latest https://github.com/thesofproject/sof/blob/master/src/ipc/apl-ipc.c

mengdonglin, 2018-11-22:
@lgirdwood please reopen this issue if you still see it. Thanks!

closed 2018-11-22 by mengdonglin