thesofproject/sof #576: [APL] Issue with loading sof-apl-nocodec.tplg

smakows, 2018-11-14:
Commits
SOF 73475e3fbb3e29ce9f0c26e317ae8191f9827688 branch **master**, project **thesofproject**
Linux 165b34de9fdfdb80bc75352fa11ae362c0ba17f6 branch **topic/sof-dev**, project **thesofproject**
SOFT 2cd668ccfaf4af99b8ce3000a1e8faaa9a796b5b branch **master**, project **thesofproject**

[image]

Logs from dmesg and kernel build .config file attached

[dmesg.log](https://github.com/thesofproject/sof/files/2580220/dmesg.log)
[apl.config.txt](https://github.com/thesofproject/sof/files/2580225/apl.config.txt)

bardliao, 2018-11-15:
@smakows I can reproduce your issue on the same commit id as yours.
But the issue is gone if I use the latest code:
kernel: sof-dev 21947cd7724218af2b5f248dbd05f0620039003c
soft: master 5894ed4cc5266f2ca859b1a59c2c82848c2f0713
sof: master 43120c923fed72a2c71dd6294097d2f6378dd74c

Could you check it on your side?
Thanks.

smakows, 2018-11-15:
> @smakows I can reproduce your issue on the same commit id as yours.
> But the issue is gone if I use the latest code:
> kernel: sof-dev 21947cd7724218af2b5f248dbd05f0620039003c
> soft: master 5894ed4cc5266f2ca859b1a59c2c82848c2f0713
> sof: master [43120c9](https://github.com/thesofproject/sof/commit/43120c923fed72a2c71dd6294097d2f6378dd74c)
> 
> Could you check it on your side?
> Thanks.

On this configuration the issue is no longer reproduced.

closed 2018-11-15 by smakows